raspberrypi/openocd #6: Flashing through openocd doesn't work as expected

sde1000, 2021-01-23:
Flashing through gdb connected to openocd using the `load` command _does_ work as expected.  But flashing using the `program` command in openocd doesn't:

```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -c "program test.elf"
Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-22-17:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10002aac
** Programming Finished **
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
^Cshutdown command invoked
```

Trying `flash banks`:

```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -c "flash banks"
Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-22-17:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
#0 : rp2040.flash (rp2040_flash) at 0x10000000, size 0x00200000, buswidth 32, chipwidth 1

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
^Cshutdown command invoked
```

sde1000, 2021-01-23:
With `--debug` on both those commands:

```
openocd --debug -f interface/picoprobe.cfg -f target/rp2040.cfg -c "program test.elf" 2>&1 | grep -v "Add 8 idle cycles"
Open On-Chip Debugger 0.10.0+dev-g14c0d0d-dirty (2021-01-22-17:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 2 options.c:63 configuration_output_handler(): 
Debug: 15 2 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 2 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 2 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 18 2 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 2 configuration.c:42 add_script_search_dir(): adding /home/steve/pico/openocd/tcl
Debug: 20 2 configuration.c:42 add_script_search_dir(): adding /home/steve/.config/openocd
Debug: 21 2 configuration.c:42 add_script_search_dir(): adding /home/steve/.openocd
Debug: 22 2 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 23 2 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 24 2 configuration.c:97 find_file(): found /home/steve/pico/openocd/tcl/interface/picoprobe.cfg
Debug: 25 2 command.c:146 script_debug(): command - adapter driver picoprobe
Info : 27 2 transport.c:117 allow_transports(): only one transport option; autoselect 'swd'
Debug: 28 2 command.c:352 register_command_handler(): registering 'swd'...
Debug: 29 2 command.c:146 script_debug(): command - transport select swd
Warn : 30 2 transport.c:296 jim_transport_select(): Transport "swd" was already selected
Debug: 31 2 command.c:146 script_debug(): command - adapter speed 5000
Debug: 33 2 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 34 2 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 35 2 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 36 2 options.c:63 configuration_output_handler(): adapter speed: 5000 kHz
User : 37 2 options.c:63 configuration_output_handler(): 
Debug: 38 2 configuration.c:97 find_file(): found /home/steve/pico/openocd/tcl/target/rp2040.cfg
Debug: 39 2 configuration.c:97 find_file(): found /home/steve/pico/openocd/tcl/target/swj-dp.tcl
Debug: 40 2 command.c:146 script_debug(): command - transport select
Debug: 41 2 configuration.c:97 find_file(): found /home/steve/pico/openocd/tcl/mem_helper.tcl
Debug: 42 2 command.c:146 script_debug(): command - add_usage_text mrw address
Debug: 44 2 command.c:1123 help_add_command(): added 'mrw' help text
Debug: 45 2 command.c:146 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 47 2 command.c:1136 help_add_command(): added 'mrw' help text
Debug: 48 2 command.c:146 script_debug(): command - add_usage_text mrh address
Debug: 50 2 command.c:1123 help_add_command(): added 'mrh' help text
Debug: 51 2 command.c:146 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 53 2 command.c:1136 help_add_command(): added 'mrh' help text
Debug: 54 2 command.c:146 script_debug(): command - add_usage_text mrb address
Debug: 56 2 command.c:1123 help_add_command(): added 'mrb' help text
Debug: 57 2 command.c:146 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 59 2 command.c:1136 help_add_command(): added 'mrb' help text
Debug: 60 2 command.c:146 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 62 2 command.c:1123 help_add_command(): added 'mmw' help text
Debug: 63 2 command.c:146 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 65 2 command.c:1136 help_add_command(): added 'mmw' help text
Debug: 66 2 command.c:146 script_debug(): command - transport select
Debug: 67 2 command.c:146 script_debug(): command - transport select
Debug: 68 2 command.c:146 script_debug(): command - swd newdap rp2040.core0 cpu -dp-id 0x01002927 -instance-id 0
Debug: 69 2 tcl.c:604 jim_newtap_cmd(): Creating New Tap, Chip: rp2040.core0, Tap: cpu, Dotted: rp2040.core0.cpu, 4 params
Debug: 70 2 tcl.c:618 jim_newtap_cmd(): Processing option: -dp-id
Debug: 71 2 tcl.c:618 jim_newtap_cmd(): Processing option: -instance-id
Debug: 72 2 core.c:1484 jtag_tap_init(): Created Tap: rp2040.core0.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 73 2 command.c:146 script_debug(): command - transport select
Debug: 74 2 command.c:146 script_debug(): command - transport select
Debug: 75 2 command.c:146 script_debug(): command - swd newdap rp2040.core1 cpu -dp-id 0x01002927 -instance-id 1
Debug: 76 2 tcl.c:604 jim_newtap_cmd(): Creating New Tap, Chip: rp2040.core1, Tap: cpu, Dotted: rp2040.core1.cpu, 4 params
Debug: 77 2 tcl.c:618 jim_newtap_cmd(): Processing option: -dp-id
Debug: 78 2 tcl.c:618 jim_newtap_cmd(): Processing option: -instance-id
Debug: 79 2 core.c:1484 jtag_tap_init(): Created Tap: rp2040.core1.cpu @ abs position 1, irlen 0, capture: 0x0 mask: 0x0
Debug: 80 2 command.c:146 script_debug(): command - dap create rp2040.core0.dap -chain-position rp2040.core0.cpu
Debug: 81 3 command.c:352 register_command_handler(): registering 'rp2040.core0.dap'...
Debug: 82 3 command.c:352 register_command_handler(): registering 'rp2040.core0.dap'...
[... cut ...]
Debug: 485 38 command.c:352 register_command_handler(): registering 'flash'...
Debug: 486 38 command.c:146 script_debug(): command - nand init
Debug: 490 40 tcl.c:498 handle_nand_init_command(): Initializing NAND devices...
Debug: 491 40 command.c:146 script_debug(): command - pld init
Debug: 495 42 pld.c:206 handle_pld_init_command(): Initializing PLDs...
Info : 496 42 gdb_server.c:3503 gdb_target_start(): starting gdb server for rp2040.core0 on 3333
Info : 497 42 server.c:310 add_service(): Listening on port 3333 for gdb connections
^C
```

Wren6991, 2021-01-23:
Thank you for the logs. I don't see the flash driver being probed at all (it's [this file](src/flash/nor/rp2040.c)) so at first glance this is either a configuration issue, or there is some command that gdb is passing to the gdb server, and is not being supplied on the command line, to probe the flash driver.

Wren6991, 2021-01-25:
Thanks, I can reproduce this. Seems like this is our bug. Our hopefully-named `auto_probe` handler is never called, but it's fine when gdb explicitly probes us. Looking through the openocd code now to figure out exactly where the problem is.

Wren6991, 2021-01-25:
```
Thread 1 "openocd" hit Breakpoint 2, get_flash_bank_by_addr (target=0x5555559a1a80, addr=268435456, 
    check=false, result_bank=0x7fffffffd2b8) at src/flash/nor/core.c:278
278	{
(gdb) n
282		for (c = flash_banks; c; c = c->next) {
(gdb) 
283			if (c->target != target)
(gdb) p *target
$8 = {type = 0x555555959330, cmd_name = 0x55555599e830 "rp2b0.core1", target_number = 1, 
  tap = 0x555555997340, coreid = 1, defer_examine = false, examined = true, running_alg = false, 
  event_action = 0x555555993a80, reset_halt = 1, working_area = 0, working_area_virt_spec = false, 
  working_area_virt = 0, working_area_phys_spec = true, working_area_phys = 536936448, 
  working_area_size = 65536, backup_working_area = 0, working_areas = 0x0, 
  debug_reason = DBG_REASON_DBGRQ, endianness = TARGET_LITTLE_ENDIAN, state = TARGET_HALTED, 
  reg_cache = 0x5555559798e0, breakpoints = 0x0, watchpoints = 0x0, trace_info = 0x5555559a1bb0, 
  dbgmsg = 0x0, dbg_msg_enabled = 0, arch_info = 0x5555559a1c58, private_config = 0x5555559a1be0, 
  next = 0x0, verbose_halt_msg = true, halt_issued = false, halt_issued_time = 1611597314944, 
  dbgbase_set = false, dbgbase = 0, has_dap = true, dap_configured = true, tap_configured = false, 
  rtos = 0x5555559a18e0, rtos_auto_detect = false, backoff = {times = 0, count = 0}, smp = 1, 
  head = 0x55555599e810, gdb_service = 0x5555559a5950, fileio_info = 0x5555559a8160, 
  gdb_port_override = 0x0, semihosting = 0x555555999f00}
(gdb) p *c->target
$9 = {type = 0x55555595df80, cmd_name = 0x555555996fb0 "rp2b0.core0", target_number = 0, 
  tap = 0x555555997170, coreid = 0, defer_examine = false, examined = true, running_alg = false, 
  event_action = 0x555555993990, reset_halt = 1, working_area = 0, working_area_virt_spec = false, 
  working_area_virt = 0, working_area_phys_spec = true, working_area_phys = 536936448, 
  working_area_size = 65536, backup_working_area = 0, working_areas = 0x0, 
  debug_reason = DBG_REASON_DBGRQ, endianness = TARGET_LITTLE_ENDIAN, state = TARGET_HALTED, 
  reg_cache = 0x555555993ad0, breakpoints = 0x0, watchpoints = 0x0, trace_info = 0x55555599a8a0, 
  dbgmsg = 0x0, dbg_msg_enabled = 0, arch_info = 0x55555599a948, private_config = 0x55555599a8d0, 
  next = 0x5555559a1a80, verbose_halt_msg = true, halt_issued = false, 
  halt_issued_time = 1611597314889, dbgbase_set = false, dbgbase = 0, has_dap = true, 
  dap_configured = true, tap_configured = false, rtos = 0x55555599a600, rtos_auto_detect = false, 
  backoff = {times = 0, count = 0}, smp = 1, head = 0x55555599e810, gdb_service = 0x5555559a5950, 
  fileio_info = 0x5555559a5cd0, gdb_port_override = 0x5555559a7ee0 "3333", semihosting = 0x55555599a0f0}
(gdb) 

```

Ha, it's checking whether it can autoprobe any matching banks on core 1, and the bank is associated with core 0. The explicit probe by gdb works. That's why you get this in your log:

```
** Programming Started **
Warn : no flash bank found for address 0x10000000
Warn : no flash bank found for address 0x10002aac
** Programming Finished **
```

Need to find out why this is, but have some idea of the problem now.

Wren6991, 2021-01-25:
Just to confirm, does this work for you as a temporary workaround?

```
openocd -f blah -c "targets rp2b0.core0; program pico-examples/build/blink/blink.elf"
```

i.e. explicitly selecting the target associated with the bank? There's a bit of an abstraction failure here, so not sure yet what the cleanest patch would be.

sde1000, 2021-01-25:
If I use that exact argument to `targets` it doesn't work, but helpfully it says:
```
Target: rp2b0.core0 is unknown, try one of:

    TargetName         Type       Endian TapName            State       
--  ------------------ ---------- ------ ------------------ ------------
 0  rp2040.core0       cortex_m   little rp2040.core0.cpu   unknown
 1* rp2040.core1       cortex_m   little rp2040.core1.cpu   unknown
```

If I use `rp2040.core0` then it does work.

Wren6991, 2021-01-25:
Ah sorry I was using an older config file I had lying around for a different (non-Pico) board with FTDI on-board. Okay thank you for confirming.

Wren6991, 2021-02-16:
Fixed by #12

closed 2021-02-16 by Wren6991