raspberrypi/openocd #7: Fails to compile on MacOS using the latest libftdi

ep1cman, 2021-01-23:
I tried to compile this today for an M1 MacBook but ran into the following error while compiling:
```
src/jtag/drivers/presto.c:163:6: error: 'ftdi_usb_purge_buffers' is deprecated
      [-Werror,-Wdeprecated-declarations]
        if (ftdi_usb_purge_buffers(&presto->ftdic) < 0) {
```

 Installing libftdi via:
```
brew install libftdi
```
installs version 1.5.0 which seems to be incompatible with the version of openOCD this was forked from. This was a known issue back in October (https://www.mail-archive.com/openocd-devel@lists.sourceforge.net/msg13025.html), but I am not sure if it has been resolved upstream.

It seems disabling FTDI support with `./configure --disable-ftdi` does not resolve the issue. 

Workarounds are:
- Installing an older 1.4 release of libftdi
- Disabling depreciation errors with `-Wno-deprecated` added to CFLAGS

ep1cman, 2021-01-23:
For those wanting to use brew to install 1.4 you can use the following commands:

```
brew tap-new --no-git local/local
brew extract --version='1.4' libftdi local/homebrew-local
```

liamfraser, 2021-01-25:
The issue you are describing is in the presto driver so try building with `--enable-presto`?

ep1cman, 2021-01-25:
Ah I hadn't clocked that in the path. It seems the same issue exists in the openjtag driver. After running
```
./configure --disable-presto --disable-openjtag
```
I was able to successfully build using libftdi 1.5

aallan, 2021-01-25:
@liamfraser Does this need a update in the documentation, with a [NOTE] block for building on macOS?

idcrook, 2021-04-06:
Could this also be related to the `--disable-werror` in the picoprobe Build OpenOCD -> Mac section in Appendix A in current revision of the `getting-started-with-pico.pdf`?  I noticed that there after I built using the above `disable`-s workaround.

To build the `picoprobe` openocd on my **M1 Mac**, however, I had to introduce one additional tweak to get around the following error:

```
src/target/arm_disassembler.c:30:10: fatal error: 'capstone/capstone.h' file not found
```
so the error is header-related, but I had the `capstone` package installed in Homebrew. What is wrong?

```console
$ brew list capstone
/opt/homebrew/Cellar/capstone/4.0.2/bin/cstool
/opt/homebrew/Cellar/capstone/4.0.2/include/capstone/ (14 files)
/opt/homebrew/Cellar/capstone/4.0.2/lib/libcapstone.4.dylib
/opt/homebrew/Cellar/capstone/4.0.2/lib/pkgconfig/capstone.pc
/opt/homebrew/Cellar/capstone/4.0.2/lib/ (2 other files)
```

the headers are available in `include`

```console
$ pkg-config --cflags capstone
-I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone
```

ah, it is not including the normal top-level `include` search path in its compile command (`/opt/homebrew/include`) and the `src` file has '**`capstone`**`/capstone.h'`.  

Adding to CFLAGS for capstone (the particular envariable found in `./configure --help`) and the following configure invocation led to a successful build (and without needing the `--disable-werror` configure flag, I may add)

```shell
CAPSTONE_CFLAGS="-I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone -I/opt/homebrew/include" ./configure \
  --enable-picoprobe --disable-presto --disable-openjtag
```

my full "recipe"

```shell
brew install open-ocd --only-dependencies
brew install automake
brew install autoconf pkg-config libtool texinfo wget gcc

cd ~/projects/pico
git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1 --no-single-branch
cd openocd

export PATH="/opt/homebrew/opt/texinfo/bin:/usr/local/opt/texinfo/bin:$PATH"
./bootstrap
CAPSTONE_CFLAGS="-I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone -I/opt/homebrew/include" ./configure  \
  --enable-picoprobe --disable-presto --disable-openjtag
```

Note: I used the dependencies for the `open-ocd` homebrew formula, so I assume that it includes anything needed to build. And, I am not sure that `gcc` was needed. `gcc` is mentioned in Appendix A for Mac, but configure was finding the xcode toolchains I had installed on my system in runs prior to installing `gcc`.

idcrook, 2021-04-06:
Since it is kinda buried in my above reply, I'll repeat here:

I didn't need the `--disable-werror` configure flag referenced in Appendix A for building `picoprobe` OpenOCD on Mac.

I _did_ need an additional setting for `configure` to get past a fatal build error on my M1 Mac / Apple Silicon Homebrew system.

idcrook, 2021-04-06:
I've whittled to a slightly shorter version of working `configure` for openocd + picoprobe on macOS. the other steps are same as above "recipe".  tested on an M1 Mac.

```shell
CAPSTONE_CFLAGS="-I$(brew --prefix)/include" \
  ./configure --prefix="$(brew --prefix)"  \
  --enable-picoprobe --disable-presto --disable-openjtag
```

I've also added the `--prefix` to `configure` so that `make install` will match your Homebrew (whether using Intel or Apple silicon setup). The assumed prefix is `/usr/local` but on Apple Silicon Homebrew, this is instead `/opt/homebrew`

Some related observations w.r.t. this

- `CAPSTONE_CFLAGS` does not seem to need `"$(pkg-config capstone)"` portion to compile successfully
- It's not clear to me why setting  `--includedir=/opt/homebrew/include` (or even just the correct `--prefix`) in `configure` does not work in this case. The include paths for the `capstone` build should imply `PREFIX/include` according to `./configure --help`. My guess would be that `capstone` autoconf setup is not quite right in the openocd sources.

lkoepsel, 2021-08-23:
Thanks, idcrook. I followed your instructions, including changing "CAPSTONE_CFLAGS" on Apr 6 and everything appears to compile correctly. I'm on Big Sur M1 Mini.

omijh, 2022-04-27:
idcrook, Thanks it worked for me but i needed the disable-werror or the build was failing
`CAPSTONE_CFLAGS="-I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone -I/opt/homebrew/include" ./configure  \
  --enable-picoprobe --disable-presto --disable-openjtag --disable-werror`

mcuee, 2022-07-17:
I guess this can be closed now.

karlzheng, 2022-07-27:
[error_info.txt](https://github.com/raspberrypi/openocd/files/9197997/error_info.txt)
Hi All, I compiled it on Macbook pro M1,  MacOS 12.31, error as below and attached shown:
`libtool: link: gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -g -O2 -o src/openocd src/main.o  src/.libs/libopenocd.a -L/usr/local/Cellar/libusb/1.0.22/lib -lusb-1.0 -lm ./jimtcl/libjim.a
ld: warning: ignoring file /usr/local/Cellar/libusb/1.0.22/lib/libusb-1.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64
Undefined symbols for architecture arm64:
  "_libusb_alloc_transfer", referenced from:
      _stlink_usb_xfer_rw in libopenocd.a(libocdjtagdrivers_la-stlink_usb.o)
      _mpsse_flush in libopenocd.a(libocdjtagdrivers_la-mpsse.o)
`

karlzheng, 2022-07-27:
Are there someone can offer some suggestions? 
Thanks!

lurch, 2022-07-28:
`ld: warning: ignoring file /usr/local/Cellar/libusb/1.0.22/lib/libusb-1.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64`

This seems to imply that you've somehow got the x86_64 version of libusb installed, rather than the arm64 version?

karlzheng, 2022-07-31:
> `ld: warning: ignoring file /usr/local/Cellar/libusb/1.0.22/lib/libusb-1.0.dylib, building for macOS-arm64 but attempting to link with file built for macOS-x86_64`
> 
> This seems to imply that you've somehow got the x86_64 version of libusb installed, rather than the arm64 version?

How to fix it? ( I install the libusb by brew command)

lurch, 2022-07-31:
I dunno, I don't have an M1 Mac; so unless somebody else reading this issue has any suggestions, maybe you'll need to ask for help at https://forums.raspberrypi.com/viewforum.php?f=143 ? (or maybe on the Brew forums?)  :shrug:

mcuee, 2022-07-31:
Firstly install homebrew properly by following the installation instruction.

```
mcuee@mcuees-Mac-mini pico % which brew
/opt/homebrew/bin/brew
```

Secondly, install the dependancies.
> ==> Dependencies
> Build: pkg-config ✔
> Required: capstone ✔, hidapi ✔, libftdi ✔, libusb ✔, libusb-compat ✔

mcuee, 2022-07-31:
BTW, if you have both ARM64 and x86_64 Homebrew installed, you may want to have tricks like this.

```
mcuee@mcuees-Mac-mini ~ % cat .zprofile
if [ "$(sysctl -n sysctl.proc_translated)" = "1" ]; then
    eval $(/usr/local/Homebrew/bin/brew shellenv)
else
    eval $(/opt/homebrew/bin/brew shellenv)
fi
```

karlzheng, 2022-07-31:
> if [ "$(sysctl -n sysctl.proc_translated)" = "1" ]; then
>     eval $(/usr/local/Homebrew/bin/brew shellenv)
> else
>     eval $(/opt/homebrew/bin/brew shellenv)
> fi

Great !! It's working for me 👍 💯

hmoazed, 2022-08-24:
Thank you idcrook! Their recipe worked on my M1 Max MacBook but I had to disable werror by adding `--disable-werror` option to the configure command otherwise the build would fail:
`src/flash/nor/dsp5680xx_flash.c:48:11: error: variable 'offset' set but not used [-Werror,-Wunused-but-set-variable]
        uint32_t offset = HFM_FLASH_BASE_ADDR;
                 ^
1 error generated.`

mcuee, 2022-08-24:
Yes, `--disable-werror` is necessary now.

closed 2023-07-07 by P33M