raspberrypi/openocd #8: "PAGE_SIZE redefined error" during compilation

nyh-workshop, 2021-01-25:
Hello rpi team,

I'm compiling this RPi's OpenOCD to use the FT2232 on the Pico using Windows 10 (with Cygwin).

However during the compilation, it gives this message:

> src/flash/nor/rp2040.c:150: error: "PAGE_SIZE" redefined [-Werror]
>   150 | #define PAGE_SIZE 256
>       |
> In file included from /usr/lib/gcc/x86_64-pc-cygwin/10/include/limits.h:195,
>                  from /usr/lib/gcc/x86_64-pc-cygwin/10/include/syslimits.h:7,
>                  from /usr/lib/gcc/x86_64-pc-cygwin/10/include/limits.h:34,
>                  from /usr/include/sys/param.h:12,
>                  from ./src/helper/system.h:75,
>                  from ./config.h:350,
>                  from src/flash/nor/rp2040.c:2:
> /usr/include/limits.h:223: note: this is the location of the previous definition
>   223 | #define PAGE_SIZE PAGESIZE
>       |
> 

Should we have include guards for the PAGE_SIZE? This occurs in commit 7c961195b91d921ccd5b71f178f9c4f53e7af921.

kilograham, 2021-01-25:
That is a define name collision... Quick fix for now just #undef it above line 150

nyh-workshop, 2021-01-26:
Hello there, it compiles! Thanks for the assistance. :)

closed 2021-01-26 by nyh-workshop