raspberrypi/openocd #9: Jlink support

happyme531, 2021-01-25:
I have a jlink ob with swd port available but it seems not working with the RP2040. I don't understand why, and there is some debug log.
```
openocd "-c" "gdb_port 50000" "-f" "interface/jlink.cfg" -c "transport select swd" -c "adapter_khz 2000" "-f" "target/rp2040.cfg" --debug
Open On-Chip Debugger 0.10.0+dev-g7c96119-dirty (2021-01-25-20:18)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 1 options.c:63 configuration_output_handler(): 
Debug: 15 1 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 1 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 1 options.c:246 add_default_dirs(): exepath=/usr/bin
Debug: 18 1 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/user/.config/openocd
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/user/.openocd
Debug: 21 1 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/site
Debug: 22 1 configuration.c:42 add_script_search_dir(): adding /usr/bin/../share/openocd/scripts
Debug: 23 1 command.c:146 script_debug(): command - gdb_port 50000
Debug: 25 1 configuration.c:97 find_file(): found /usr/bin/../share/openocd/scripts/interface/jlink.cfg
Debug: 26 1 command.c:146 script_debug(): command - adapter driver jlink
Debug: 28 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 29 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 30 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 31 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 32 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 33 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 34 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 35 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 36 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 37 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 38 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 39 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 40 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 41 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 42 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 43 1 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 44 1 command.c:146 script_debug(): command - transport select swd
Debug: 45 1 command.c:352 register_command_handler(): registering 'swd'...
User : 46 1 options.c:63 configuration_output_handler(): swdUser : 47 1 options.c:63 configuration_output_handler(): 
Debug: 48 1 command.c:146 script_debug(): command - echo DEPRECATED! use 'adapter speed' not 'adapter_khz'
User : 50 1 command.c:767 jim_echo(): DEPRECATED! use 'adapter speed' not 'adapter_khz'
Debug: 51 1 command.c:146 script_debug(): command - adapter speed 2000
Debug: 53 1 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 54 1 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 55 1 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 56 1 options.c:63 configuration_output_handler(): adapter speed: 2000 kHz
User : 57 1 options.c:63 configuration_output_handler(): 
Debug: 58 1 configuration.c:97 find_file(): found /usr/bin/../share/openocd/scripts/target/rp2040.cfg
Debug: 59 1 configuration.c:97 find_file(): found /usr/bin/../share/openocd/scripts/target/swj-dp.tcl
Debug: 60 1 command.c:146 script_debug(): command - transport select
Debug: 61 1 configuration.c:97 find_file(): found /usr/bin/../share/openocd/scripts/mem_helper.tcl
Debug: 62 1 command.c:146 script_debug(): command - add_usage_text mrw address
Debug: 64 1 command.c:1123 help_add_command(): added 'mrw' help text
Debug: 65 1 command.c:146 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 67 1 command.c:1136 help_add_command(): added 'mrw' help text
Debug: 68 1 command.c:146 script_debug(): command - add_usage_text mrh address
Debug: 70 1 command.c:1123 help_add_command(): added 'mrh' help text
Debug: 71 1 command.c:146 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 73 1 command.c:1136 help_add_command(): added 'mrh' help text
Debug: 74 1 command.c:146 script_debug(): command - add_usage_text mrb address
Debug: 76 1 command.c:1123 help_add_command(): added 'mrb' help text
Debug: 77 1 command.c:146 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 79 1 command.c:1136 help_add_command(): added 'mrb' help text
Debug: 80 1 command.c:146 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 82 1 command.c:1123 help_add_command(): added 'mmw' help text
Debug: 83 1 command.c:146 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 85 1 command.c:1136 help_add_command(): added 'mmw' help text
Debug: 86 1 command.c:146 script_debug(): command - swd newdap rp2040.core0 cpu -dp-id 0x01002927 -instance-id 0
Debug: 87 1 tcl.c:604 jim_newtap_cmd(): Creating New Tap, Chip: rp2040.core0, Tap: cpu, Dotted: rp2040.core0.cpu, 4 params
Debug: 88 1 tcl.c:618 jim_newtap_cmd(): Processing option: -dp-id
[... 284 lines cut ...]
Debug: 469 19 command.c:626 run_command(): Command 'init' failed with error code -4
User : 470 19 command.c:692 command_run_line(): 
Debug: 471 19 target.c:1991 target_free_all_working_areas_restore(): freeing all working areas
Debug: 472 19 target.c:1991 target_free_all_working_areas_restore(): freeing all working areas
Error: 473 19 jlink.c:2153 jlink_swd_switch_seq(): Sequence 3 not supported.
Error: 474 19 jlink.c:2153 jlink_swd_switch_seq(): Sequence 4 not supported.
Debug: 475 19 jlink.c:2138 jlink_swd_switch_seq(): SWD line reset
Debug: 476 19 jlink.c:2167 jlink_swd_run_queue(): Executing 5 queued transactions.
Debug: 486 19 jlink.c:2191 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 487 19 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 0, retrying...
Debug: 488 19 jlink.c:2138 jlink_swd_switch_seq(): SWD line reset
Debug: 489 19 jlink.c:2167 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 499 20 jlink.c:2191 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 500 20 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 0, retrying...
Debug: 501 20 jlink.c:2138 jlink_swd_switch_seq(): SWD line reset
Debug: 502 20 jlink.c:2167 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 512 20 jlink.c:2191 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 513 20 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 0
Debug: 518 21 jlink.c:526 jaylink_log_handler(): Closing device (bus:address = 003:027).
Debug: 519 21 jlink.c:526 jaylink_log_handler(): Device closed successfully.
Debug: 520 21 jlink.c:526 jaylink_log_handler(): Device destroyed (bus:address = 003:027).
```
Also please provide some binary packages, as not everyone are familar with compiling the software.

If only this chip can be supported by SEGGER itself，we will be able to use the offical jlink software and it will simplify many steps.

JamesH65, 2021-01-25:
> 
> Also please provide some binary packages, as not everyone are familar with compiling the software.
> 
> If only this chip can be supported by SEGGER itself，we will be able to use the offical jlink software and it will simplify many steps.

What do you mean by binary packages? For which part of the work flow?

SEGGER  will need to support their own device. Note that you can use a Pico to debug with SWD to another Pico, so at most that is $5 extra expense if you cannot get the SEGGER working.

happyme531, 2021-01-25:
> What do you mean by binary packages? For which part of the work flow?
>
> SEGGER will need to support their own device. Note that you can use a Pico to debug with SWD to another Pico, so at most that [...]

I mean the openocd itself, especially for windows. This is helpful for further integration (maybe platformio or something like NOOBS for pico)

There are some areas where you can get a jlink ob (absolutely clone) for 3$ and it is probably the most common debug tool for developers.  
So it is pretty important!

kilograham, 2021-01-25:
Yup; we didn't include JLink changes for SWD multidrop because we didn't have anything to test them on, and figured that could be done after we up-streamed our changes. Watch this space, and we'll post something that _might_ work shortly

kilograham, 2021-01-25:
try the rp2040_jlink branch (as I say untested)

Hexxeh, 2021-01-25:
I tried the branch mentioned, the "Sequence <n> not supported errors" disappeared but I'm still unable to successfully run openocd. Output: 

```sudo openocd "-c" "gdb_port 50000" "-f" "interface/jlink.cfg" -c "transport select swd" -c "adapter_khz 2000" "-f" "target/rp2040.cfg" --debug
Open On-Chip Debugger 0.10.0+dev-g7c961195b-dirty (2021-01-25-22:12)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 0 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 0 options.c:63 configuration_output_handler(): 
Debug: 15 0 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 0 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 0 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 18 0 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 0 configuration.c:42 add_script_search_dir(): adding /root/.config/openocd
Debug: 20 0 configuration.c:42 add_script_search_dir(): adding /root/.openocd
Debug: 21 0 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 22 0 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 23 0 command.c:146 script_debug(): command - gdb_port 50000
Debug: 25 0 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/interface/jlink.cfg
Debug: 26 0 command.c:146 script_debug(): command - adapter driver jlink
Debug: 28 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 29 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 30 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 31 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 32 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 33 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 34 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 35 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 36 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 37 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 38 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 39 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 40 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 41 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 42 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 43 0 command.c:352 register_command_handler(): registering 'jlink'...
Debug: 44 0 command.c:146 script_debug(): command - transport select swd
Debug: 45 0 command.c:352 register_command_handler(): registering 'swd'...
User : 46 0 options.c:63 configuration_output_handler(): swdUser : 47 0 options.c:63 configuration_output_handler(): 
Debug: 48 0 command.c:146 script_debug(): command - echo DEPRECATED! use 'adapter speed' not 'adapter_khz'
User : 50 0 command.c:767 jim_echo(): DEPRECATED! use 'adapter speed' not 'adapter_khz'
Debug: 51 0 command.c:146 script_debug(): command - adapter speed 2000
Debug: 53 0 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 54 0 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 55 0 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
User : 56 0 options.c:63 configuration_output_handler(): adapter speed: 2000 kHz
User : 57 0 options.c:63 configuration_output_handler(): 
Debug: 58 1 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/target/rp2040.cfg
Debug: 59 1 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/target/swj-dp.tcl
Debug: 60 1 command.c:146 script_debug(): command - transport select
Debug: 61 1 configuration.c:97 find_file(): found /usr/local/bin/../share/openocd/scripts/mem_helper.tcl
Debug: 62 1 command.c:146 script_debug(): command - add_usage_text mrw address
Debug: 64 1 command.c:1123 help_add_command(): added 'mrw' help text
Debug: 65 1 command.c:146 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 67 1 command.c:1136 help_add_command(): added 'mrw' help text
Debug: 68 1 command.c:146 script_debug(): command - add_usage_text mrh address
Debug: 70 1 command.c:1123 help_add_command(): added 'mrh' help text
Debug: 71 1 command.c:146 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 73 1 command.c:1136 help_add_command(): added 'mrh' help text
Debug: 74 1 command.c:146 script_debug(): command - add_usage_text mrb address
Debug: 76 1 command.c:1123 help_add_command(): added 'mrb' help text
Debug: 77 1 command.c:146 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 79 1 command.c:1136 help_add_command(): added 'mrb' help text
Debug: 80 1 command.c:146 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 82 1 command.c:1123 help_add_command(): added 'mmw' help text
Debug: 83 1 command.c:146 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 85 1 command.c:1136 help_add_command(): added 'mmw' help text
Debug: 86 1 command.c:146 script_debug(): command - transport select
Debug: 87 1 command.c:146 script_debug(): command - transport select
[... cut ...]
Debug: 519 100 jlink.c:2205 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 520 100 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 0
Debug: 526 100 jlink.c:526 jaylink_log_handler(): Last read operation left 16 bytes in the buffer.
Debug: 528 100 jlink.c:526 jaylink_log_handler(): Closing device (bus:address = 003:015).
Debug: 529 100 jlink.c:526 jaylink_log_handler(): Device closed successfully.
Debug: 530 100 jlink.c:526 jaylink_log_handler(): Device destroyed (bus:address = 003:015).```

happyme531, 2021-01-26:
Jlink is working for me on branch rp2040_jlink.
```
 ../src/openocd "-c" "gdb_port 50000" "-f" "interface/jlink.cfg" -c "transport select swd" -c "adapter_khz 2000" "-f" "target/rp2040.cfg"
Open On-Chip Debugger 0.10.0+dev-g7e5ea18-dirty (2021-01-26-13:37)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
swd
DEPRECATED! use 'adapter speed' not 'adapter_khz'
adapter speed: 2000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Link OB-STM32F072-CortexM compiled Jan  7 2019 14:09:37
Info : Hardware version: 1.00
Info : VTarget = 3.300 V
Info : clock speed 2000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 50000
Info : Listening on port 50000 for gdb connections
Info : accepting 'gdb' connection on tcp/50000
```

Hexxeh, 2021-01-26:
I suspect given it works for you and not me, it's a problem with my setup somehow. I attached a logic analyser on the SWD pins and saw no activity when running openocd, so some problem with my cables or JLink Base likely.

ep1cman, 2021-01-26:
May be worth checking the hardware revision of your J-Link, they don't all support multi-drop SWD: https://wiki.segger.com/Software_and_Hardware_Features_Overview

ep1cman, 2021-01-26:
I can confirm the `rp2040_jlink` branch works for me using a J-Link EDU 11.0, I've attached a photo of how I wired it up (The pinout diagrams of the J-Link are from the perspective of the socket so you have to flip it)

[image]

ryanplusplus, 2021-01-26:
I'm able to debug using the jlink branch. However, I'm not able to program with it.

Here's my openocd script for programming:

```
source [find interface/jlink.cfg]
transport select swd
adapter speed 1000
source [find target/rp2040.cfg]
program build/debug/target.hex verify reset exit
```

which fails with:

```
Open On-Chip Debugger 0.10.0+dev-g7e5ea18-dirty (2021-01-26-07:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : J-Link V10 compiled Oct  5 2020 14:56:16
Info : Hardware version: 10.10
Info : VTarget = 3.293 V
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
Warn : no flash bank found for address 0x10000000
** Programming Finished **
** Verify Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x21000000 pc: 0x10002efa msp: 0x20041fa0
Error: checksum mismatch - attempting binary compare
** Verify Failed **
shutdown command invoked
```

with `--debug`:

```
Open On-Chip Debugger 0.10.0+dev-g7e5ea18-dirty (2021-01-26-07:54)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
User : 13 1 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 1 options.c:63 configuration_output_handler(): 
Debug: 15 1 options.c:244 add_default_dirs(): bindir=/home/ryan/git/pico-sdk-tiny/tools/openocd/build/bin
Debug: 16 1 options.c:245 add_default_dirs(): pkgdatadir=/home/ryan/git/pico-sdk-tiny/tools/openocd/build/share/openocd
Debug: 17 1 options.c:246 add_default_dirs(): exepath=/home/ryan/git/pico-sdk-tiny/tools/openocd/build/bin
Debug: 18 1 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 1 configuration.c:42 add_script_search_dir(): adding /home/ryan/.config/openocd
Debug: 20 1 configuration.c:42 add_script_search_dir(): adding /home/ryan/.openocd
Debug: 21 1 configuration.c:42 add_script_search_dir(): adding /home/ryan/git/pico-sdk-tiny/tools/openocd/build/bin/../share/openocd/site
Debug: 22 1 configuration.c:42 add_script_search_dir(): adding /home/ryan/git/pico-sdk-tiny/tools/openocd/build/bin/../share/openocd/scripts
Debug: 23 1 configuration.c:97 find_file(): found tools/upload-debug.cfg
Debug: 24 1 configuration.c:97 find_file(): found /home/ryan/git/pico-sdk-tiny/tools/openocd/build/bin/../share/openocd/scripts/interface/jlink.cfg
Debug: 25 1 command.c:146 script_debug(): command - adapter driver jlink
Debug: 27 1 command.c:352 register_command_handler(): registering 'jlink'...
[... 1077 lines cut ...]
User : 4357 670 command.c:692 command_run_line(): 
Debug: 4358 670 target.c:1991 target_free_all_working_areas_restore(): freeing all working areas
Debug: 4359 670 target.c:1991 target_free_all_working_areas_restore(): freeing all working areas
Debug: 4360 670 target.c:1759 print_wa_layout():    0x20010000-0x2001ffff (65536 bytes)
Debug: 4361 671 jlink.c:2140 jlink_swd_switch_seq(): SWD line reset
Debug: 4362 671 jlink.c:2179 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 4372 671 jlink.c:2179 jlink_swd_run_queue(): Executing 0 queued transactions.
Debug: 4382 671 jlink.c:2160 jlink_swd_switch_seq(): SWD-to-DORMANT
Debug: 4383 671 jlink.c:2140 jlink_swd_switch_seq(): SWD line reset
Debug: 4384 671 jlink.c:2179 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 4394 672 jlink.c:2205 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 4395 672 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 1, retrying...
Debug: 4396 672 jlink.c:2140 jlink_swd_switch_seq(): SWD line reset
Debug: 4397 672 jlink.c:2179 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 4407 672 jlink.c:2205 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 4408 672 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 1, retrying...
Debug: 4409 672 jlink.c:2140 jlink_swd_switch_seq(): SWD line reset
Debug: 4410 672 jlink.c:2179 jlink_swd_run_queue(): Executing 4 queued transactions.
Debug: 4420 673 jlink.c:2205 jlink_swd_run_queue(): SWD ack not OK: 7 JUNK
Debug: 4421 673 adi_v5_swd.c:655 swd_multidrop_select_target(): Failed to select core 1
Debug: 4427 673 jlink.c:526 jaylink_log_handler(): Last read operation left 16 bytes in the buffer.
Debug: 4429 673 jlink.c:526 jaylink_log_handler(): Closing device (bus:address = 001:079).
Debug: 4430 673 jlink.c:526 jaylink_log_handler(): Device closed successfully.
Debug: 4431 673 jlink.c:526 jaylink_log_handler(): Device destroyed (bus:address = 001:079).
```

Hexxeh, 2021-01-26:
@ep1cman thanks for the photo! I was missing the VTref pin connection. I hadn't soldered those headers on yet, only the SWD ones, not realising it was needed.

When looking it up though, I noted that it says it's used a reference for the clock/IO levels. Are those at 3.3v like the rest of the IO? If so, why is that not using the 3.3v out as a reference? I guess it'll work because it's still within the allowable range for a "high" signal?

ep1cman, 2021-01-26:
You are totally right, I wrongly assumed what Vsys was thinking it was 3.3v, I should have read the data sheet better. VTref should indeed be connected to 3.3v to set the J-Link signal voltages.

conkerkh, 2021-01-28:
I can program raspberry just fine but it will always fail on verification at different addresses, it looks like some problem with reading from flash. Also I noticed that reading from 0x10000000 doesn't always work, I assume it requires properly configure XIP_SSI first?

JamesH65, 2021-01-28:
Often that can result from the wiring. You need to ensure that the SWD wires are as short as possible, with nothing in between (for example, I used to have mine set up via a breadboard, but that was unreliable - straight wires from the Pi to the Pico worked fine)

HeadBoffin, 2021-01-29:
I have my J-Link on macOS 10.14 using this openocd talking to my Pico AND I've arrived at a place where gdb works but I'll have to do it again to figure out if it was luck or skill.

conkerkh, 2021-01-30:
I tried different wiring including soldering directly to the board and I'm still getting verification failed on "monitor program" command (see attached log). I think there is a problem with flash loader in regards to jlink, otherwise debugging and everything else works fine which further confirms its not a wiring. Anyway at 400khz its very unlikely that 5cm wires would be a problem...

`target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
** Programming Started **
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
Writing 40960 bytes starting at 0x0
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
** Programming Finished **
** Verify Started **
target was in unknown state when halt was requested
SWD DPIDR 0x0bc12477
SWD DLPIDR 0x00000001
error executing cortex_m crc algorithm
SWD DPIDR 0x0bc12477
SWD DLPIDR 0x00000001
Failed to read memory at 0x100000b0
embedded:startup.tcl:530: Error: ** Verify Failed **
in procedure 'program' 
in procedure 'program_error' called at file "embedded:startup.tcl", line 591
at file "embedded:startup.tcl", line 530`

conkerkh, 2021-01-30:
I belive the problem comes from wrong implementation of WAIT response in jlink driver. When WAIT instead of ACK is received at DP/AP transcation transfer is dropped instead of retried. I don't know where error handling should be implemented in jlink driver or higher level drivers, but that seems to be the case. From ADI specification one can read:

**WAIT response to a DPACC or APACC access**

A WAIT response indicates that the previous transaction has not completed. Normally, after receiving a WAIT response the host retries the DPACC or APACC access.

Note

The previous transaction might be either a DP or an AP access. DP accesses are stalled, by returning WAIT, until any previous AP transaction has completed.

Normally, if software detects a WAIT response, it retries the same transfer, which enables the protocol to process data as quickly as possible. However, in case an abort mechanism is implemented, if the software has retried a transfer several times, and has waited long enough for a slow interconnect and memory system to respond, it might write to the ABORT register to cancel the operation. This action signals that the active AP must terminate the transfer that it is attempting, to permit access to other parts of the debug system. An AP might not be able to terminate a transfer on its SoC interface. However, on receiving an abort request, the AP must free its interface to the DP.

No request is generated at the Update-DR state, and the shifted-in data is discarded. The captured value of ReadResult[31:0] is UNKNOWN.

rei-vilo, 2021-09-17:
Thank you for the very useful tips: I can now use the J-Link to debug against the RP2040.

rkta, 2022-02-14:
For people like me wondering where the that jlink branch is: seems like the changes are included in branch rp2040 and the jlink branch was just deleted. I can use branch rp2040 with a J-Link.

mcuee, 2022-07-17:
So this issue can be closed.

closed 2023-07-07 by P33M