raspberrypi/openocd #11: FT2232H support requested

nyh-workshop, 2021-01-26:
Hello RPi team again,

I have attempted to see if FT2232H actually supports the Pico.

Unfortunately, all it said was this after using the config file for rp2040 and the adafruit's FT2232:

```
 $  openocd -f tcl/interface/ftdi/ft232h-module-swd.cfg -f tcl/target/rp2040.cfg
Open On-Chip Debugger 0.10.0+dev-g7c96119-dirty (2021-01-25-21:56)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
adapter speed: 8 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 8 kHz
Info : DAP init failed
```

I have inserted the 470 ohm resistor between ADBUS1 and ADBUS2 (from the swd-resistor-hack.cfg), but DAP init failed.

The USB drivers are being switched using Zadig to WinUSB.

Maybe I might have wired it wrongly, or it isn't supported right now?

aallan, 2021-01-26:
I think this is the correct wiring SWD debug using FT2232, but I haven't checked it for acuracy.

[image]

kilograham, 2021-01-26:
But yes, it should work

aallan, 2021-01-26:
Thanks @kilograham

aallan, 2021-01-26:
Thinking about it, I think the SWD GND pin should also be wired. Doesn't SWD need its own GND?

dp111, 2021-01-26:
It may work without its own ground, but please use its own ground

dp111, 2021-01-26:
The same with the UART use Pin3 as well of Pico

jimbojr, 2021-01-26:
Yes use adjacent GNDs for UART and SWD for best signal integrity. Multiple GND pins are useful - who knew... :-)

nyh-workshop, 2021-01-27:
Hello team, I've tested it again. I have to jiggle the wires and reconnect again. The diagram by @aallan is working fine. 

```
 $  openocd -f tcl/interface/ftdi/ft232h-module-swd.cfg -f tcl/target/rp2040.cfg
Open On-Chip Debugger 0.10.0+dev-g7c96119-dirty (2021-01-25-21:56)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : FTDI SWD mode enabled
adapter speed: 8 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_NOT_SUPPORTED
Info : clock speed 8 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

closed 2021-01-27 by nyh-workshop