raspberrypi/openocd #13: add support for stlink

EthercatT, 2021-01-26:
I am trying to use an stlink to program my rp2040

openocd -f interface/stlink.cfg -f target/rp2040.cfg  

And end up with this issue:
/usr/local/bin/../share/openocd/scripts/target/swj-dp.tcl:30: Error: Unknown param: -dp-id, try one of: -irlen, -irmask, -ircapture, -enable, -disable, -expected-id, or -ignore-version
in procedure 'script'

HeadBoffin, 2021-01-29:
I fully appreciate you can use a Pico to do this, but many of us have lots of random ST-Link's - I've an original and a draw full of the cheap clones off eBay. As my IoT devices have long test times, it would be useful to use these ST-Links for monitoring but I can just double up on Pico's if they can be bought in pairs as an alternative.

ghollingworth, 2021-01-29:
I'm guessing the STLink doesn't support multidrop SWD...

That would need to be fixed upstream first

EthercatT, 2021-01-30:
I belive it does on the latest firmware, there's a couple of dual core STM32 parts.

GMMan, 2021-02-03:
I believe ST-Link V3 supports multi-drop, but not necessarily earlier versions. In any case someone has to reverse engineer ST-Link's protocol for it since it's a high level adapter, not one where you could twiddle the lines yourself.

maxgerhardt, 2021-07-15:
I'm wondering what happened after https://community.st.com/s/question/0D53W00000YVOfM/does-stlink-support-ad-v2multidrop-swd .. No confirmation from ST yet.

mcuee, 2022-07-17:
I remember that ST-Link only supports STM32/STM8 MCUs from ST by license.

P33M, 2023-07-07:
If you want stlink swd support, then ask ST to merge their OpenOCD patches upstream - https://github.com/STMicroelectronics/OpenOCD
We won't be doing it here.

closed 2023-07-07 by P33M

UweBonnes, 2024-05-02:
https://github.com/STMicroelectronics/OpenOCD does not implement multidrop for STlinks.