raspberrypi/openocd #20: Segmentation fault with libusb 1.0.24 and msys2 / Windows

gaeeronimo, 2021-02-10:
It seems the picoprobe USB code has problems with libusb-1.0.24. The executable can be started, but will segfault when trying to talk to the picoprobe with libusb. When using an older libusb-1.0.23 openocd works normally. A similar problem also occurs for picotool.

I'm running on Windows 10 Build 19042.746 and did setup msys2 from scratch following the [docs](https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf).

lurch, 2021-02-10:
Duplicate of https://github.com/raspberrypi/picoprobe/issues/3 ?

gaeeronimo, 2021-02-10:
Yes, looks like it's the same. As openocd segfaulted, I searched for issues here. Would have saved me some time also looking in picoprobe 😢

closed 2021-02-11 by aallan