raspberrypi/openocd #22: openocd fails to start when using VSCode

hamslabs, 2021-03-08:
There is no udev rule for the board so this fails due to permissions.

My workaround was to add this to contrib/60-openocd.rules and copy it to /etc/udev/rules.d
```
#Raspberry Pi Pico
ATTRS{idVendor}=="2e8a", ATTRS{idProduct}=="0004", MODE="660", GROUP="plugdev", TAG+="uaccess"
```

It might be a good idea to also do this during install or document how to do it manually.

aallan, 2021-03-08:
Which platform are you running this on, Raspberry Pi OS?

hamslabs, 2021-03-08:
I'm running on Mint Linux 20 Ulyana. If I run openocd (not as root) from the command line it also fails and I see this output
```
$ openocd -c "gdb_port 50000" -s "/home/ghampton/mysrc/RP2040/src/hellotest" -f interface/picoprobe.cfg -f target/rp2040.cfg 
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-03-01-20:19)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: libusb_open() failed with LIBUSB_ERROR_ACCESS
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```

lurch, 2021-03-08:
Duplicate of #5 ?

aallan, 2021-03-08:
> Duplicate of #5 ?

Ah! Yup.

closed 2021-03-08 by aallan