raspberrypi/openocd #23: Linux compile error

iandownie07, 2021-03-18:
I'm trying to use openocd on my installation for the pico and get the following error:

pen On-Chip Debugger 0.10.0+dev-gf8e14ec-dirty (2021-03-17-19:30)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (bcm2835gpio)
The following debug adapters are available:
1: ftdi
2: usb_blaster
3: ft232r
4: presto
5: usbprog
6: openjtag
7: jlink
8: vsllink
9: rlink
10: ulink
11: arm-jtag-ew
12: hla
13: osbdm
14: opendous
15: sysfsgpio
16: aice
17: xds110
18: st-link

Any ideas?

alex4o, 2021-03-31:
This is not a bug, you just forgot to add `--enable-bcm2835gpio`, you should always check what flags can be enabled by doing `./configure --help`.

mcuee, 2022-07-17:
> This is not a bug, you just forgot to add `--enable-bcm2835gpio` to your `./configure` command, you should always check what flags can be enabled by doing `./configure --help`.

So this can be closed now.

closed 2023-07-07 by P33M

CoastalAmusementsFrank, 2024-02-07:
What do you mean by add "-enable-bcm2835gpio to your ./configure"? What configure command are you talking about?