raspberrypi/openocd #24: ftdi, sysfsgpio and bcm2835gpio not enabled after configurations step

tpbadger, 2021-04-02:
After running config command:

`./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio`

The configuration summary is missing ftdi sysfsgpio and bcm2835gpio. 

[image]

When running openocd it errors with 

`Error: The specified debug interface was not found (bcm2835gpio)`

Not sure if I have missed something or this is an issue others are having?

tpbadger, 2021-04-02:
Ignore this! Correct me if I am wrong but is the bcm2835gpio interface for the raspberry pi to debug using SWD?

closed 2021-04-02 by tpbadger

lurch, 2021-04-02:
Yes, https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf says "If you are building on a Raspberry Pi you can also pass --enable-sysfsgpio --enable-bcm2835gpio to allow bitbanging SWD via the GPIO pins."

sanjayadhikari007, 2021-08-28:
I am also facing same issue. Any workaround ?

lurch, 2021-08-31:
@sanjayadhikari007 Please open a new issue, providing as much information as possible. (e.g. what instructions you're following, what commands you've run, which OS you're using, what errors you get, etc.)

tcs-ulli, 2021-10-04:
Iam running in the same thing now, is there any new issue ? What was wrong with this issue ? I see the same information i also would add.

tcs-ulli, 2021-10-05:
What information is missing ?
we both rebuild openocd with 

./bootstrap
./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio

then no bcm2835gpio appears in the overview 

and after 

make;;sudo make install

openocd tells that it dont knows an programmer device named bcm2835gpio...

Please tell what information you miss. Please define "enough information"
For me it was enougth information to reproduce the issue, i have no clue what you wish more

lurch, 2021-10-05:
@tcs-ulli Apologies if my previously reply seemed curt, but a lot of users comment "I'm having the same problem" but then never follow up with extra information when requested :roll_eyes: 

Thank you for the information you've provided so far. Can you please also tell us:
Which branch of openocd did you clone? What OS are you using? What hardware are you running on?

Also, which instructions are you following? Those in https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf or some third-party website?