raspberrypi/openocd #25: Flash error using openocd

alexandruradovici, 2021-04-15:
I am trying to flash a binary with the following layout:

```
MEMORY
{
  /* boot from Flash */
  rom (rx)  : ORIGIN = 0x10000000, LENGTH = 128K
  prog (rx) : ORIGIN = 0x10020000, LENGTH = 1920K
  ram (rwx) : ORIGIN = 0x20004000, LENGTH = 256K
}

MPU_MIN_ALIGN = 8K;
PAGE_SIZE = 4K;
```

When trying to flash, I get the following openocd output:

```
$ openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg 
Open On-Chip Debugger 0.10.0+dev-geb22ace-dirty (2021-02-23-15:05)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 1001 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x0000012a msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
Error: Whole bank access must start at beginning of bank.
Error: flash_erase returned -904
```

Flashing works fine if I remove the `prog` line from the layout file or put the address somewhere in RAM (> 0x2000000).

kilograham, 2021-04-15:
This appears to be a bug in `flash/nor/rp2040.c` which assumes one flash_erase followed by one flash_write; however in the case of discontinuous chunks, openocd does flash_erase, flash_erase, flash_write, flash_write.

alexandruradovici, 2021-04-28:
Hi all, Is there any progress on this issue? Not being able to flash this is a small blocker for us. Thank you.

alexandruradovici, 2021-05-03:
@kilograham I am trying to debug this, but I am not sure what to look at. What is the assumption that `flash/nor/rp2040.c` makes? I not able to infer any connection between the erase and write functions, but most likely I am missing something.

kilograham, 2021-05-03:
the flash erase here calls two methods to prepare for access to the flash before erasing

https://github.com/raspberrypi/openocd/blob/f8e14ec97b0d98c8e2ffdd08ab5ff9537f1c9a63/src/flash/nor/rp2040.c#L236

and the write restores things back to normal XIP mode here

https://github.com/raspberrypi/openocd/blob/f8e14ec97b0d98c8e2ffdd08ab5ff9537f1c9a63/src/flash/nor/rp2040.c#L211

so erase->erase->write leaves things in XIP mode so the final write fails.

It should work fine, if both the write methods do the prepare at the beginning and the restore at the end.

lurch, 2021-05-03:
> https://github.com/raspberrypi/openocd/blob/f8e14ec97b0d98c8e2ffdd08ab5ff9537f1c9a63/src/flash/nor/rp2040.c#L211

There's two identical `LOG_ERROR("RP2040 write: failed to flush flash cache");` lines there - I guess the second one ought to be something like `LOG_ERROR("RP2040 write: failed to enter flash XIP mode");` ?

alexandruradovici, 2021-05-04:
Thank you, I fixed it, I'll submit a PR with it.

PR #29 (2021-06-17): Exit and enter XIP for erase and write
This PR fixes #25 by exiting and entering XIP mode separately for erase and write.

This will allow writing discontinuous flash regions.
commit e32e11ea (2021-06-17): Exit and enter XIP for erase and write (#29)

* Exit and enter XIP for erase and write

This fixes #25

* Update rp2040.c

* exit and enter xip mode are now separate functions

* delete extra debug and code style update

* update typo
closed 2021-06-17 by Wren6991