raspberrypi/openocd #26: Update jimtcl

EvanTheB, 2021-04-17:
Jimtcl has a new home: https://github.com/msteveb/jimtcl

And recently fixed the whitespace compilation issue: https://github.com/msteveb/jimtcl/commit/bcd4434161897da5203cdfaef8b2227e15d82c45

lurch, 2021-04-17:
Yes, the whitespace issue got fixed thanks to me reporting it :smiley: 

I've been planning to bump the version of `jimtcl` used in this repo, but before I can do that I need to check that openocd, VSCode and picoprobe all still work properly, and unfortunately I've not had time to do that yet... :slightly_frowning_face:

lurch, 2021-04-17:
(@aallan please feel free to assign this issue to me)

lurch, 2024-08-24:
@EvanTheB I assume that this issue is now redundant? (gosh, 3 years old!)