raspberrypi/openocd #28: Upstream to the official openocd repo

perexg, 2021-04-26:
It would be really nice to see this fork upstreamed to the official openocd repository, so Linux distributions can pick it easily.

aallan, 2021-04-26:
> It would be really nice to see this fork upstreamed to the official openocd repository, so Linux distributions can pick it easily.

We would love to have this upstreamed. But it's with the upstream maintainers at this point, they need to accept the PRs.

closed 2021-04-26 by aallan

perexg, 2021-04-26:
Perhaps, you may update the upstream status here? (Links to PR/Patches?) Thank you.

cz172638, 2021-04-26:
try #openocd @ freenode

cz172638, 2021-04-26:
and mailing list

lurch, 2021-04-26:
I just did a bit of searching around and found http://openocd.zylin.com/#/q/rp2040

perexg, 2021-04-26:
The picoprobe support is missing in the above PRs... I think that we're missing a plan for the merges. But I'm glad to see that there's work-in-progress for this.