raspberrypi/openocd #30: Compile error MacOS: capstone.h not found

berikv, 2021-05-07:
Capstone is installed via homebrew:

```
> brew info capstone
capstone: stable 4.0.2 (bottled), HEAD
```

After
```
./configure --enable-picoprobe --disable-werror
make -j4
```

The compiler errors with
```
libtool: compile:  gcc -DHAVE_CONFIG_H -I. -I./src -I./src -I./src/helper -DPKGDATADIR=\"/usr/local/share/openocd\" -DBINDIR=\"/usr/local/bin\" -I./jimtcl -I./jimtcl -I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -g -O2 -MT src/target/libtarget_la-armv4_5_mmu.lo -MD -MP -MF src/target/.deps/libtarget_la-armv4_5_mmu.Tpo -c src/target/armv4_5_mmu.c -o src/target/libtarget_la-armv4_5_mmu.o
src/target/arm_disassembler.c:30:10: fatal error: 'capstone/capstone.h' file not found
#include <capstone/capstone.h>
         ^~~~~~~~~~~~~~~~~~~~~
1 error generated.
```

The Capstone headers are installed in `/opt/homebrew/Cellar/capstone/4.0.2/include`
```
> ls /opt/homebrew/Cellar/capstone/4.0.2/include/capstone/
arm.h        capstone.h   m680x.h      mips.h       ppc.h        systemz.h    x86.h
arm64.h      evm.h        m68k.h       platform.h   sparc.h      tms320c64x.h xcore.h
```

A similar issue has occurred here https://github.com/WerWolv/ImHex/issues/145

berikv, 2021-05-07:
Compilation succeeded by changing `configure.ac`:
`PKG_CHECK_MODULES([CAPSTONE], [capstone], [have_capstone=no], [have_capstone=no])`

This disables capstone support obviously.

berikv, 2021-05-07:
Oh and I tried to install capstone's latest by running `brew install capstone --HEAD`.
However I did that in an attempt to fix this issue. I'll try and install the stable version and update the install location.

berikv, 2021-05-07:
```
> brew install capstone
==> Downloading https://ghcr.io/v2/homebrew/core/capstone/manifests/4.0.2
Already downloaded: /Users/berik/Library/Caches/Homebrew/downloads/d8caa83b6b16573878067f49634ea8c0e94c617b48dd9516b31f0cea46b40cb9--capstone-4.0.2.bottle_manifest.json
==> Downloading https://ghcr.io/v2/homebrew/core/capstone/blobs/sha256:5c2d67aeb32a36c76d1918ec10de347971b
Already downloaded: /Users/berik/Library/Caches/Homebrew/downloads/c5fc5208d5bee197968fe85540ffd586fc95e7334fce497b9e385073222c0230--capstone--4.0.2.arm64_big_sur.bottle.tar.gz
==> Pouring capstone--4.0.2.arm64_big_sur.bottle.tar.gz
🍺  /opt/homebrew/Cellar/capstone/4.0.2: 25 files, 14.8MB

> brew info capstone
capstone: stable 4.0.2 (bottled), HEAD
Multi-platform, multi-architecture disassembly framework
https://www.capstone-engine.org/
/opt/homebrew/Cellar/capstone/4.0.2 (25 files, 14.8MB) *
  Poured from bottle on 2021-05-07 at 17:24:35
From: https://github.com/Homebrew/homebrew-core/blob/HEAD/Formula/capstone.rb
License: BSD-3-Clause
==> Options
--HEAD
	Install HEAD version
==> Analytics
install: 1,290 (30 days), 3,786 (90 days), 9,525 (365 days)
install-on-request: 363 (30 days), 1,299 (90 days), 6,530 (365 days)
build-error: 0 (30 days)

> brew list capstone
/opt/homebrew/Cellar/capstone/4.0.2/bin/cstool
/opt/homebrew/Cellar/capstone/4.0.2/include/capstone/ (14 files)
/opt/homebrew/Cellar/capstone/4.0.2/lib/libcapstone.4.dylib
/opt/homebrew/Cellar/capstone/4.0.2/lib/pkgconfig/capstone.pc
/opt/homebrew/Cellar/capstone/4.0.2/lib/ (2 other files)
```

berikv, 2021-05-07:
From https://docs.brew.sh/FAQ

> The default prefix is /usr/local for macOS on Intel, /opt/homebrew for macOS on ARM, and /home/linuxbrew/.linuxbrew for Linux.

lurch, 2021-05-10:
Looking at your first message again... you have the `capstone.h` file at:
```
/opt/homebrew/Cellar/capstone/4.0.2/include/capstone/capstone.h
```
and the error message says:
```
src/target/arm_disassembler.c:30:10: fatal error: 'capstone/capstone.h' file not found
#include <capstone/capstone.h>
```
and the compiler command-line included:
```
-I/opt/homebrew/Cellar/capstone/4.0.2/include/capstone
```

so _maybe_ the compiler was looking for `/opt/homebrew/Cellar/capstone/4.0.2/include/capstone/capstone/capstone.h`, and the error would be fixed if you can figure out how to change the compiler command-line to use
```
-I/opt/homebrew/Cellar/capstone/4.0.2/include
```
instead? :shrug:  (this is just a guess!)

lurch, 2021-05-10:
I don't have much familiarity with OSX, but I bought myself an i3 MacMini a couple of weeks ago so that I could help with testing Mac stuff. It's still running Catalina, I've not upgraded it to Big Sur.
I just ran through the Mac section of Appendix A of https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf on my MacMini for the first time, and it all built fine. So then I did a `make clean; brew install capstone; ./configure --enable-picoprobe --disable-werror; make -j4` and it again built fine. So I guess perhaps the problem is with Homebrew (or capstone? or openocd?) on M1 Macs? :shrug: 

FYI:
```
% ls /usr/local/Cellar/capstone/4.0.2/include/capstone
arm.h		capstone.h	m680x.h		mips.h		ppc.h		systemz.h	x86.h
arm64.h		evm.h		m68k.h		platform.h	sparc.h		tms320c64x.h	xcore.h
```
and if I scroll back through the terminal I can see that some of the compiler-commands include
```
-I/usr/local/Cellar/capstone/4.0.2/include/capstone
```
so perhaps the "guess" in my previous message was in fact wrong? Sorry.

Kristine1975, 2022-02-04:
The issue you referenced has been solved. Maybe that solution works here, too: https://github.com/WerWolv/ImHex/issues/145#issuecomment-766843267

(@lurch it looks like you were right on the money with https://github.com/raspberrypi/openocd/issues/30#issuecomment-837460809)

mcuee, 2022-07-17:
I guess this issue can be closed as well.

Crest, 2023-04-04:
The problem is that pkg-config --cflags returns CFLAGS already suffixed with "/capstone", but the code attempts to `#include <capstone/capstone.h>`. Removing the prefix like this allows openocd to build:
```
./configure --disable-werror CAPSTONE_CFLAGS="$(pkg-config capstone --cflags | sed s/.capstone\$//)"
```

P33M, 2023-07-07:
Any changes proposed here should be submitted upstream.

closed 2023-07-07 by P33M