raspberrypi/openocd #31: Openocd fails to compile for rp2040

embeddedpenguin, 2021-05-19:
Following the steps from the rpi org guide: https://datasheets.raspberrypi.org/pico/getting-started-with-pico.pdf#page=19&zoom=100,153,0

```
$ cd ~/pico
$ sudo apt install automake autoconf build-essential texinfo libtool libftdi-dev libusb-1.0-0-
dev
$ git clone https://github.com/raspberrypi/openocd.git --recursive --branch rp2040 --depth=1
$ cd openocd
$ ./bootstrap
$ ./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio
$ make -j4
$ sudo make install
```

Compilation fails with these errors/warnings regarding presto (I guess)
```
src/jtag/drivers/openjtag.c: In function ‘openjtag_init_standard’:
src/jtag/drivers/openjtag.c:439:9: error: ‘ftdi_usb_purge_buffers’ is deprecated [-Werror=deprecated-declarations]
  439 |         if (ftdi_usb_purge_buffers(&ftdic) < 0) {
      |         ^~

```

```
src/jtag/drivers/presto.c: In function ‘presto_open_libftdi’:
src/jtag/drivers/presto.c:163:9: error: ‘ftdi_usb_purge_buffers’ is deprecated [-Werror=deprecated-declarations]
  163 |         if (ftdi_usb_purge_buffers(&presto->ftdic) < 0) {
      |         ^~
In file included from src/jtag/drivers/presto.c:37:
/usr/include/libftdi1/ftdi.h:566:20: note: declared here
  566 |     int DEPRECATED(ftdi_usb_purge_buffers(struct ftdi_context *ftdi));
      |                    ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/libftdi1/ftdi.h:247:55: note: in definition of macro ‘DEPRECATED’
  247 | #define DEPRECATED(func) __attribute__ ((deprecated)) func
      |                                                       ^~~~
src/jtag/drivers/presto.c:177:17: error: ‘ftdi_usb_purge_buffers’ is deprecated [-Werror=deprecated-declarations]
  177 |                 if (ftdi_usb_purge_buffers(&presto->ftdic) < 0)
      |                 ^~
In file included from src/jtag/drivers/presto.c:37:
/usr/include/libftdi1/ftdi.h:566:20: note: declared here
  566 |     int DEPRECATED(ftdi_usb_purge_buffers(struct ftdi_context *ftdi));
      |                    ^~~~~~~~~~~~~~~~~~~~~~
/usr/include/libftdi1/ftdi.h:247:55: note: in definition of macro ‘DEPRECATED’
  247 | #define DEPRECATED(func) __attribute__ ((deprecated)) fun
```

I'm on arch linux (5.12.5-arch-1)

embeddedpenguin, 2021-05-19:
I'm actually getting dozens more errors after fixing those manually. I'm thinking something is up with my setup.

embeddedpenguin, 2021-05-19:
Openocd fails to compile all on its own now. Weird.

embeddedpenguin, 2021-05-20:
Gotcha. Sorry about that. I posted here because it compiled, then it didn't. I figured a git pull broke everything... It is actually on my end.

closed 2021-05-20 by embeddedpenguin

cibonato, 2022-06-27:
Issue closed, however I'd like to add a comment. A quite similar problem happened with me during ``./bootstrap``. It was solved installed ``pkg-config``.

It made possible to install ``openocd`` on a Raspberry Pi 3 running DietPI v8.5.1 (it's a Debian 11 like system).

lurch, 2022-06-28:
Thanks for the info, and I'm glad you managed to resolve your problem. However the instructions in our official documentation are only for (the latest version of) Raspberry Pi OS, and other OSes may have different packages installed. We don't provide instructions for every OS variation as that'd make the instructions too complicated.