raspberrypi/openocd #33: Uncompilable with GCC 11

zsigmondszilveszter, 2021-05-27:
Hello,

The actual state of the code from this repository doesn't compile with GCC 11 on Arch Linux (probably the distribution doesn't count). I even tried to manually solve the compile errors to a certain point, but I after a while get tired fixing the code which is already patched on the upstream repository. FYI, the upstream OpenOCD compiles well with GCC 11. Do you have any plans to merge the picoprobe branch into the upstream OpenOCD repository or merge the upstream OpenOCD into the Picoprobe branch in this repository?

thierer, 2021-05-30:
What features do you need? Only today I modified the [official PKGBUILD](https://github.com/archlinux/svntogit-community/tree/packages/openocd/trunk) to patch the RP2040 support from the rp2040 branch into the 0.11.0 openocd release version. I could make that available on github, but I haven't included the support for 1. gpio bitbanging and 2. the picoprobe adapter, because I use neither.

zsigmondszilveszter, 2021-05-31:
I need the Picoprobe branch from this repo, which is not merged into the upstream repo if I am right, but neither the upstream repository is merged into the Picoprobe branch. I am not an expert in OpenOCD, I don't know what packages could I skip without braking the functionalities I need. I checked and can assure, that the code where the compilation errors come is fixed in the upstream repo. The GPIO bitbanging part is not affected right now, because it works only on Raspberry Pies and there is no GCC 11 yet (at least with Raspberry Pi OS). I managed to compile the Picoprobe branch with an older Arch release with GCC 10 and copied over the source directory containing the compiled binaries. I can use it from the place but cannot `make install` because it detects the toolchain version differences and start over the build part.

thierer, 2021-05-31:
Try running `configure --disable-werror`. If that doesn't help, try an `autoreconf --no-recursive` before `configure`.

zsigmondszilveszter, 2021-06-02:
Yes the `--disable-werror` helps and it compiles well with it. But OpenOCD is not usable from outside of its source directory anyway,  `interface/picoprobe.cfg` cannot be found from outside.

thierer, 2021-06-02:
It should be, if you installed it via `make install`. Did you do that? (I thought that was your plan). If so, did you specify a `--prefix` for `configure`? If not, it got probably installed to `/usr/local/`. What does `type -p openocd` say?

zsigmondszilveszter, 2021-06-02:
Yes I installed with `make install`. I didn't specified a prefix and you are right it was installed under `/usr/local/bin/`. 
```
szilveszter@szilvArchDesktop ~> which openocd
/usr/local/bin/openocd
szilveszter@szilvArchDesktop ~>
```
The `type -p openocd` gives the following response:
```
szilveszter@szilvArchDesktop ~> type -p openocd
-
szilveszter@szilvArchDesktop ~>
```
I built the picoprobe branch from this repository with the following commands:
```
szilveszter@szilvArchDesktop ~> git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1 --no-single-branch
szilveszter@szilvArchDesktop ~> cd openocd
szilveszter@szilvArchDesktop ~> ./bootstrap
szilveszter@szilvArchDesktop ~> ./configure --enable-picoprobe --disable-presto --disable-openjtag --disable-werror
szilveszter@szilvArchDesktop ~> make -j4
szilveszter@szilvArchDesktop ~> sudo make install
```

```
szilveszter@szilvArchDesktop ~> openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.10.0+dev-geb22ace (2021-05-27-12:43)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
embedded:startup.tcl:26: Error: Can't find interface/picoprobe.cfg
in procedure 'script' 
at file "embedded:startup.tcl", line 26
szilveszter@szilvArchDesktop ~> [1]
```
Meanwhile I found out that this is happening only with Fish, but not with Bash. I am wondering what could be wrong with my Fish environment?

thierer, 2021-06-02:
> Meanwhile I found out that this is happening only with Fish, but not with Bash. I am wondering what could be wrong with my Fish environment?

You mean it's working with bash? I have never used fish, I can't comment on that.

Does `/usr/local/share/openocd/scripts/interface/picoprobe.cfg` exist?
If it does, what does `openocd -d` say? It should list the directories where openocd is searching for its scripts. (The `add_script_search_dir()` lines). In your case, it should probably include `/usr/local/bin/../share/openocd/scripts`.

Btw. you don't need the `-s tcl` option when openocd is installed. It's only used to add the `tcl` subdirectory of the repository to the search path for scripts when running the command from there (although I don't think it would hurt; I think it only **adds** to the search path and doesn't replace it).

zsigmondszilveszter, 2021-06-03:
> You mean it's working with bash? I have never used fish, I can't comment on that.
Yes it is working with Bash and now with Fish also. It was my bad. I created an alias to my previously compiled elsewhere binary and didn't realize that I am still using that version under Fish.
Now everything works as supposed. To anybody who runs into this problem, the key to my problem was that `--disable-werror` compile flag. It is not a final solution, but rather a temporary fix, tell the compiler not to handle warnings as errors, until the code is fixed.
@thierer thank you for your help and sorry for wasting your time. I appreciate a lot.

kilograham, 2021-06-04:
i re-based rp2040 onto v0.11.0 from upstream to make rp2040-v0.11.0-test.

Merging upstream is largely under the control of openocd itself- the core SWD and cortex M SMP bits have had PRs open for several years. It will happen at some point, but pushing it along hasn't been highest on our use of time priority list

P33M, 2023-07-07:
There is now very little downstream code remaining.

closed 2023-07-07 by P33M