raspberrypi/openocd #34: Support for SEGGER RTT

lgj99, 2021-05-28:
Everything works fine, except for RTT (openocd manual; section 15.7) commands issued via telnet. Issuing any of these commands gives "invalid command name: rtt".  I also noticed that the folder rtt is not present in the src folder, and it is present in the openocd master on sourceforge.

So, was RTT removed (assuming that it was) for a reason, or was lack of support for RTT unintentional.

Thanks and Thanks for all of your efforts making the rp2040 and RPi Pico easy to work with at the SDK level.

--lance

thierer, 2021-06-02:
That's because RTT support was apparently added after the code was forked that is the base for the rp2040 support in this repository.

If you like, you can try [the rpi_pico branch in my openocd repository](https://github.com/thierer/openocd/tree/rpi_pico), where I added the rp2040 support from here (minus SWD multidrop support for the bitbang protocol and the "picoprobe" interface) to the latest openocd release version 0.11.0.

Please note that this doesn't imply that RTT will actually work with the rp2040. If it needs special support it probably does not.

thierer, 2021-06-04:
FYI @kilograham [just announced](https://github.com/raspberrypi/openocd/issues/33#issuecomment-854895725) the [rp2040-v0.11.0-test](https://github.com/raspberrypi/openocd/tree/rp2040-v0.11.0-test) branch, so you should use that.

To avoid confusion, I'll set my repository private again.

kilograham, 2021-06-04:
note i didn't look at this issue specifically - it is entirely possible the RTT openocd stuff is not aware of how to talk SWD multidrop (which is required)

lgj99, 2021-06-05:
Thanks everyone.  I'll dig a little further based on your comments.

peter9477, 2021-06-11:
For what it's worth, I just manually merged the RTT stuff from rp2040-v0.11.0-test into the picoprobe branch, and was able to run it successfully against an nRF52 (Nordic) chip running the SEGGER_RTT stuff, so it does work.  I believe the only little hitch aside from a straight merge was needing to deal with an extra argument that was added to the `add_service()` routine in the 0.11.0 changes, but wasn't in the picoprobe (based on 0.10.0) branch. (I handled that by dropping the extra argument from several calls to add_service, guessing it wasn't critical.)

Unrelated: to get nrf52.cfg to work with the picoprobe I had to fix the handling of -expected-id in the swj_dap command. Will probably add a separate issue to cover that as I'm not familiar enough with openocd or the code base to be certain I handled it the best way.

josuah, 2022-12-15:
Related: https://github.com/raspberrypi/openocd/issues/56
Related: https://github.com/raspberrypi/picoprobe/pull/31

If the upstream OpenOCD can be used with the picoprobe (only reason for this fork), then no need to push support for RTT on this fork.

closed 2023-07-07 by P33M