raspberrypi/openocd #37: Error running interface picoprobe and target nrf52

alej0varas, 2021-06-20:
I've compiled openocd from github  branch picoprobe. When I execute `openocd --file picoprobe.cfg --file nrf52.cfg` I see the following error:

```
../share/openocd/scripts/target/swj-dp.tcl:30: Error: Unknown param: 0x2ba01477, try one of: -irlen, -irmask, -ircapture, -enable, -disable, -expected-id, -ignore-version, -dp-id, or -instance-id
in procedure 'script' 
at file "embedded:startup.tcl", line 26
in procedure 'swj_newdap' called at file "nrf52.cfg", line 27
at file "../share/openocd/scripts/target/swj-dp.tcl", line 30
```

Both `.cfg` files are form the installed `scripts` folder.

lurch, 2021-06-20:
I suspect this _may_ be the wrong place to be asking for help about the nrf52 parts of openocd? :man_shrugging:

ionspin, 2021-07-06:
I've worked around this by editing nrf52.cfg and chaning line 27 like this 
```
#swj_newdap $_CHIPNAME cpu -expected-id $_CPUTAPID
swj_newdap $_CHIPNAME cpu -expected-id
```
Got this log when running with the changed file (which I **think** is a success):
```
▶ openocd -f ./picoprobe-pinetime.cfg -f target/nrf52.cfg
Open On-Chip Debugger 0.10.0+dev-g18b4c3584 (2021-07-05-16:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 8000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x2ba01477
Info : nrf52.cpu: hardware has 6 breakpoints, 4 watchpoints
Info : starting gdb server for nrf52.cpu on 3333
Info : Listening on port 3333 for gdb connections
```
The picoprobe-pinetime is same as picoprobe.cfg just with adapter speed set to 8000.

This is my first day working with openocd and swd, so what I tried here might be incredibly stupid and not work at all.

andrew-rogers, 2021-08-12:
@ionspin I can confirm your workaround (deleting the argument to -expected-id) works for me too. I am using the target/stm32l4x.cfg for my device and faced the same issue. However, the workaround needs to be undone if you use a different interface.

Many thanks.

andrew-rogers, 2021-08-12:
The problem seems to be in the src/jtag/tcl.c which I have changed as below in order to fix the bug.

```
diff --git a/src/jtag/tcl.c b/src/jtag/tcl.c
index fbb49d595..9ed590f27 100644
--- a/src/jtag/tcl.c
+++ b/src/jtag/tcl.c
@@ -625,6 +625,13 @@ static int jim_newtap_cmd(Jim_GetOptInfo *goi)
                                        instance_id_specified = true;
                                        e = jim_newtap_md_param(n, goi, pTap);
                                        break;
+                               case NTAP_OPT_EXPECTED_ID:
+                               case NTAP_OPT_IRLEN:
+                               case NTAP_OPT_IRMASK:
+                               case NTAP_OPT_IRCAPTURE:
+                                       /* dummy read to ignore the next argument */
+                                       Jim_GetOpt_Wide(goi, NULL);
+                                       break;
                                default:
                                        e = JIM_OK;
                                        break;
```

mcuee, 2022-07-17:
Again this is not relevant for this repo. So this should be closed as well.

P33M, 2022-11-14:
Closing - bug is in unmodified upstream code.

closed 2022-11-14 by P33M