raspberrypi/openocd #38: Windows compilation problems.

prime6809, 2021-06-21:
Building under MINGW / MSYS causes one of two problems :

1) Using the current latest version of libusb (1.0.24-3) causes a segmentation fault.
2) Using the previous version 1.0.23, causes the picoprobe not to be detected.

Building under cygwin which has an earlier version of libusb (1.0.20-1), produces a working executable.

Cheers.

Phill.

lurch, 2021-06-21:
Duplicate of https://github.com/raspberrypi/picoprobe/issues/3 ?

mcuee, 2022-07-13:
https://github.com/raspberrypi/picoprobe/issues/3#issuecomment-1132810855

The following libusb issue has been resolved in libusb release 1.0.25. Latest libusb version is 1.0.26. It is recommended that you use libusb-1.0.26 to build openocd.
* https://github.com/libusb/libusb/issues/954

But WinUSB driver is still recommended and not libusb0.sys.

mcuee, 2022-07-14:
So this can be closed now.

closed 2023-07-07 by P33M