raspberrypi/openocd #40: Segfault with -c "gdb_port disabled"

aurimasniekis, 2021-06-25:
I am have set up the OpenOCD Runner action inside CLion IDE, if you just run the target it will start OpenOCD with few commands to disable ports and program flash. 

This is example command created by IDE:

```sh
$ /usr/local/bin/openocd -s /usr/local/share/openocd/scripts -f /Volumes/Storage/Projects/RP2040/rp2040-project-template/openocd.cfg -c "tcl_port disabled" -c "gdb_port disabled" -c "tcl_port disabled" -c "program \"/Volumes/Storage/Projects/RP2040/rp2040-project-template/cmake-build-debug/rp2040_project.elf\"" -c reset -c shutdown
```

Running it causes segfault at this step:

```
Open On-Chip Debugger 0.10.0+dev-g71510a77a61c (2021-06-25-18:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
User : 13 0 options.c:63 configuration_output_handler(): debug_level: 3
User : 14 0 options.c:63 configuration_output_handler():
Debug: 15 0 options.c:244 add_default_dirs(): bindir=/usr/local/bin
Debug: 16 0 options.c:245 add_default_dirs(): pkgdatadir=/usr/local/share/openocd
Debug: 17 0 options.c:246 add_default_dirs(): exepath=/usr/local/bin
Debug: 18 0 options.c:247 add_default_dirs(): bin2data=../share/openocd
Debug: 19 0 configuration.c:42 add_script_search_dir(): adding /Users/aurimas/Library/Preferences/org.openocd
Debug: 20 0 configuration.c:42 add_script_search_dir(): adding /Users/aurimas/.config/openocd
Debug: 21 0 configuration.c:42 add_script_search_dir(): adding /Users/aurimas/.openocd
Debug: 22 0 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/site
Debug: 23 0 configuration.c:42 add_script_search_dir(): adding /usr/local/bin/../share/openocd/scripts
Debug: 24 1 configuration.c:97 find_file(): found /Volumes/Storage/Projects/RP2040/rp2040-project-template/openocd.cfg
Debug: 25 1 configuration.c:97 find_file(): found /usr/local/share/openocd/scripts/interface/cmsis-dap.cfg
Debug: 26 1 command.c:146 script_debug(): command - adapter driver cmsis-dap
Debug: 28 1 command.c:352 register_command_handler(): registering 'cmsis-dap'...
Debug: 29 1 command.c:352 register_command_handler(): registering 'cmsis-dap'...
Debug: 30 1 command.c:352 register_command_handler(): registering 'cmsis_dap_vid_pid'...
Debug: 31 1 command.c:352 register_command_handler(): registering 'cmsis_dap_serial'...
Debug: 32 1 command.c:146 script_debug(): command - transport select swd
Debug: 33 1 command.c:352 register_command_handler(): registering 'swd'...
Debug: 34 1 command.c:146 script_debug(): command - adapter speed 4000
Debug: 36 1 core.c:1822 jtag_config_khz(): handle jtag khz
Debug: 37 1 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 38 1 core.c:1785 adapter_khz_to_speed(): convert khz to interface specific speed value
Debug: 39 1 configuration.c:97 find_file(): found /usr/local/share/openocd/scripts/target/rp2040.cfg
Debug: 40 1 configuration.c:97 find_file(): found /usr/local/share/openocd/scripts/target/swj-dp.tcl
Debug: 41 1 command.c:146 script_debug(): command - transport select
Debug: 42 1 configuration.c:97 find_file(): found /usr/local/share/openocd/scripts/mem_helper.tcl
Debug: 43 1 command.c:146 script_debug(): command - add_usage_text mrw address
Debug: 45 1 command.c:1123 help_add_command(): added 'mrw' help text
Debug: 46 1 command.c:146 script_debug(): command - add_help_text mrw Returns value of word in memory.
Debug: 48 1 command.c:1136 help_add_command(): added 'mrw' help text
Debug: 49 1 command.c:146 script_debug(): command - add_usage_text mrh address
Debug: 51 1 command.c:1123 help_add_command(): added 'mrh' help text
Debug: 52 1 command.c:146 script_debug(): command - add_help_text mrh Returns value of halfword in memory.
Debug: 54 1 command.c:1136 help_add_command(): added 'mrh' help text
Debug: 55 16 command.c:146 script_debug(): command - add_usage_text mrb address
Debug: 57 16 command.c:1123 help_add_command(): added 'mrb' help text
Debug: 58 16 command.c:146 script_debug(): command - add_help_text mrb Returns value of byte in memory.
Debug: 60 16 command.c:1136 help_add_command(): added 'mrb' help text
Debug: 61 16 command.c:146 script_debug(): command - add_usage_text mmw address setbits clearbits
Debug: 63 16 command.c:1123 help_add_command(): added 'mmw' help text
Debug: 64 16 command.c:146 script_debug(): command - add_help_text mmw Modify word in memory. new_val = (old_val & ~clearbits) | setbits;
Debug: 66 16 command.c:1136 help_add_command(): added 'mmw' help text
Debug: 67 16 command.c:146 script_debug(): command - transport select
Debug: 68 16 command.c:146 script_debug(): command - transport select
Debug: 69 16 command.c:146 script_debug(): command - swd newdap rp2040.core0 cpu -dp-id 0x01002927 -instance-id 0
Debug: 70 16 tcl.c:605 jim_newtap_cmd(): Creating New Tap, Chip: rp2040.core0, Tap: cpu, Dotted: rp2040.core0.cpu, 4 params
Debug: 71 19 tcl.c:618 jim_newtap_cmd(): Processing option: -dp-id
Debug: 72 19 tcl.c:618 jim_newtap_cmd(): Processing option: -instance-id
Debug: 73 19 core.c:1488 jtag_tap_init(): Created Tap: rp2040.core0.cpu @ abs position 0, irlen 0, capture: 0x0 mask: 0x0
Debug: 74 19 command.c:146 script_debug(): command - transport select
Debug: 75 19 command.c:146 script_debug(): command - transport select
Debug: 76 19 command.c:146 script_debug(): command - swd newdap rp2040.core1 cpu -dp-id 0x01002927 -instance-id 1
Debug: 77 19 tcl.c:605 jim_newtap_cmd(): Creating New Tap, Chip: rp2040.core1, Tap: cpu, Dotted: rp2040.core1.cpu, 4 params
[... 600 lines cut ...]
Debug: 696 911 rp2040.c:95 rp2040_call_rom_func(): Calling ROM func IF with 0 arguments
Debug: 697 911 rp2040.c:96 rp2040_call_rom_func(): Calling on core "rp2040.core0"
Debug: 698 913 target.c:2408 target_read_u32(): address: 0x00000010, value: 0x0201754d
Debug: 699 915 target.c:2432 target_read_u16(): address: 0x00000014, value: 0x007a
Debug: 700 917 target.c:2432 target_read_u16(): address: 0x0000007a, value: 0x3350
Debug: 701 919 target.c:2432 target_read_u16(): address: 0x0000007e, value: 0x3352
Debug: 702 921 target.c:2432 target_read_u16(): address: 0x00000082, value: 0x334c
Debug: 703 923 target.c:2432 target_read_u16(): address: 0x00000086, value: 0x3354
Debug: 704 925 target.c:2432 target_read_u16(): address: 0x0000008a, value: 0x534d
Debug: 705 927 target.c:2432 target_read_u16(): address: 0x0000008e, value: 0x3453
Debug: 706 929 target.c:2432 target_read_u16(): address: 0x00000092, value: 0x434d
Debug: 707 931 target.c:2432 target_read_u16(): address: 0x00000096, value: 0x3443
Debug: 708 933 target.c:2432 target_read_u16(): address: 0x0000009a, value: 0x4255
Debug: 709 935 target.c:2432 target_read_u16(): address: 0x0000009e, value: 0x5444
Debug: 710 937 target.c:2432 target_read_u16(): address: 0x000000a2, value: 0x4544
Debug: 711 939 target.c:2432 target_read_u16(): address: 0x000000a6, value: 0x5657
Debug: 712 941 target.c:2432 target_read_u16(): address: 0x000000aa, value: 0x4649
Debug: 713 943 target.c:2432 target_read_u16(): address: 0x000000ac, value: 0x24a1
Debug: 714 943 rp2040.c:121 rp2040_call_rom_func(): Set r7 = 000024a1
Debug: 715 943 rp2040.c:121 rp2040_call_rom_func(): Set sp = 20010100
Debug: 716 943 target.c:1646 target_call_event_callbacks(): target event 3 (resume-start) for core rp2040.core0
[1]    56836 segmentation fault  /usr/local/bin/openocd -s /usr/local/share/openocd/scripts -f  -c  -c  -c  -c
```

Removing the `gdb_port disabled` solves the issue, but I can't really remove it as IDE adds it on the Run scheme.

aurimasniekis, 2021-06-25:
Also looks like if you pause the program, and resume it segfaults too. I picked up rp2040-core0.cfg file from upstream, modified it a bit and now it runs perfectly even with gdb  port disabled. 

Here is my config right now:

```
transport select swd

source [find target/swj-dp.tcl]

if { [info exists CHIPNAME] } {
	set _CHIPNAME $CHIPNAME
} else {
	set _CHIPNAME rp2040
}

if { [info exists WORKAREASIZE] } {
	set _WORKAREASIZE $WORKAREASIZE
} else {
	set _WORKAREASIZE 0x10000
}

if { [info exists CPUTAPID] } {
	set _CPUTAPID $CPUTAPID
} else {
	set _CPUTAPID 0x01002927
}

swj_newdap $_CHIPNAME.core0 cpu -dp-id $_CPUTAPID -instance-id 0
dap create $_CHIPNAME.dap -chain-position $_CHIPNAME.core0.cpu
set _TARGETNAME $_CHIPNAME.cpu
target create $_TARGETNAME cortex_m -dap $_CHIPNAME.dap
$_TARGETNAME configure -work-area-phys 0x20010000 -work-area-size $_WORKAREASIZE

set _FLASHNAME $_CHIPNAME.flash
set _FLASHSIZE 0x200000
set _FLASHBASE 0x10000000
flash bank $_FLASHNAME rp2040_flash $_FLASHBASE $_FLASHSIZE 1 32 $_TARGETNAME

# srst does not exist; use SYSRESETREQ to perform a soft reset
cortex_m reset_config sysresetreq
```

aurimasniekis, 2021-06-26:
Someone has solved the issue with gdb disable https://www.raspberrypi.org/forums/viewtopic.php?f=144&t=310446

bfloch, 2022-08-03:
I took the liberty to convert the solution into a PR #66

PR #66 (2023-01-20): Fixes flashing with picoprobe in clion via 'Run'
According to https://forums.raspberrypi.com/viewtopic.php?t=310446
clion does not have a gdb service in 'Run' mode vs. 'Debug Run'.
As such the following checks are necessary.

This allows clion to program via picoprobe when a program is 'Run'.
Big thanks to 'mike1024k' for this workaround.

Fixes #40
closed 2023-01-20 by P33M