raspberrypi/openocd #43: Configuration Failure On Ubuntu

Spottigus, 2021-09-30:
When I go to configure open OCD i get the following:

 ./configure --enable-picoprobe

checking for makeinfo... makeinfo
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a thread-safe mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.out
checking for suffix of executables...
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether we are using the GNU C compiler... yes
checking whether gcc accepts -g... yes
checking for gcc option to accept ISO C89... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
checking for gcc option to accept ISO C99... none needed
checking for ranlib... ranlib
./configure: line 4513: syntax error near unexpected token `0.23'
./configure: line 4513: `PKG_PROG_PKG_CONFIG(0.23)'

lurch, 2021-09-30:
a) Which branch of this repo are you building?
b) Which version of Ubuntu are you using?
c) Which compilation instructions are you following?

Broatcast, 2022-05-12:
I have the same problem running on Debian.

Branch: rp2040
Instructions: Original from Raspberry, [datasheets.raspberrypi.com/pico](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf) Point 5.1

lurch, 2022-05-12:
What version of Debian are you running? Is this on a Pi or a PC? Are all your other packages up to date? Do you have `pkg-config` installed?

Broatcast, 2022-05-12:
Its on a PC Debian 10.12
Linux hot-box 4.19.0-20-amd64 #1 SMP Debian 4.19.235-1 (2022-03-17) x86_64 GNU/Linux

All up to date, found the solution in another post. pkg-config missing was the problem at the end. Maybe the doc should be updated to this.

lurch, 2022-05-12:
As it notes in the PDF:
> They also assume you are using Raspberry Pi OS running on a Raspberry Pi 4, or an equivalent Debian-based Linux distribution running on another platform.

I'm afraid we only support Raspberry Pi OS, and not the myriad other Linux distributions, otherwise the instructions would become too long.

Broatcast, 2022-05-24:
Sry to bring this up again, but i get the point, sure supporting every Linux distribution will be a nightmare.

But as its mentioned:
> or an equivalent Debian-based Linux
shouldn't at least an last clean and fresh Debian installation be supportet in the current release?

lurch, 2022-05-24:
That's fair. Section 2.2 of the Getting-Started-Guide _does_ say:
"Ubuntu and Debian users might additionally need to also install libstdc++-arm-none-eabi-newlib."

So I'll add a similar note about needing pkg-config to Appendix A.