raspberrypi/openocd #45: Unable to upload elf file - picoprobe - rp2040

ghubcoder, 2021-11-06:
Hello, 

I'm following the guides [here](https://blog.smittytone.net/2021/02/05/how-to-debug-a-raspberry-pi-pico-with-a-mac-swd/) and [here](https://andrejacobs.org/electronics/develop-and-debug-raspberry-pi-pico-on-macos-big-sur/) (second one has more wires 🙂 ). This wasn't working via vscode however so I'm using the following from the cli:

```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg
```

I then connect via gdb and try to load an elf file:

```
arm-none-eabi-gdb sleep.elf
Type "apropos word" to search for commands related to "word"...
Reading symbols from sleep.elf...
(No debugging symbols found in sleep.elf)
(gdb) target extended-remote localhost:3333
Remote debugging using localhost:3333
0x000000ee in ?? ()
(gdb) load
Error erasing flash with vFlashErase packet
```

From the openocd side I see:

```
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
Error: SWD Read data parity mismatch
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Error: Failed to read memory at 0x0000007c
Error: Debug trampoline not found in RP2040 ROM.
Error: RP2040 exit xip: failed to connect internal flash
Error: failed erasing sectors 0 to 2
Error: flash_erase returned -4
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
```

This does seem to have actually worked at some point however as I can see the target pico device is running my simple program to switch on the led (after having the flash wiped). 

Any idea what I'm doing wrong?

lurch, 2021-11-06:
ping @liamfraser and @kilograham

ghubcoder, 2021-11-06:
I managed to get this working by having both pico's powered by separate usb cables. 

The swd ground pin on the target pico was then connected directly to a ground pin on the picoprobe pico....

Apologies for the noise!

closed 2021-11-06 by ghubcoder

lurch, 2021-11-06:
Yes, it's always important to have a common ground shared by all the parts of your circuit. Glad to hear that you managed to solve your problem :+1: