raspberrypi/openocd #46: Build fails on MacOS

struandw, 2021-12-05:
This is MacOS 11.5.2, M1 chip. Build log is:
`$ make -j4
cat src/helper/startup.tcl src/jtag/startup.tcl src/target/startup.tcl src/server/startup.tcl src/flash/startup.tcl | ./src/helper/bin2char.sh > src/startup_tcl.inc || { rm -f src/startup_tcl.inc; false; }
cp src/jtag/drivers/minidriver_imp.h src/jtag/minidriver_imp.h
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in jimtcl
        CC      jim-subcmd.o
        MKLDEXT _load-static-exts.c
        CC      jim-interactive.o
        CC      jim-format.o
        CC      jim.o
        CC      utf8.o
        CC      jimregexp.o
        CC      jimiocompat.o
        CC      linenoise.o
        CC      jim-tty.o
        CC      jim-aio.o
        CC      jim-array.o
        CC      jim-clock.o
        CC      jim-eventloop.o
        CC      jim-exec.o
        CC      jim-file.o
jim-exec.c:1046:15: warning: 'vfork' is deprecated: Use posix_spawn or fork [-Wdeprecated-declarations]
        pid = vfork();
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/unistd.h:604:1: note: 'vfork' has been explicitly marked deprecated here
__deprecated_msg("Use posix_spawn or fork")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:208:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
        CC      jim-history.o
1 warning generated.
        CC      jim-interp.o
        CC      jim-load.o
        CC      jim-namespace.o
        CC      jim-pack.o
        CC      jim-package.o
        CC      jim-posix.o
        CC      jim-readdir.o
        CC      jim-regexp.o
        CC      jim-signal.o
        CC      jim-syslog.o
        TCLEXT  _glob.c
        TCLEXT  _nshelper.c
        CC      glob.o
        TCLEXT  _oo.c
        CC      nshelper.o
        CC      oo.o
        TCLEXT  _stdlib.c
        TCLEXT  _tclcompat.c
        CC      stdlib.o
        CC      tclcompat.o
        TCLEXT  _tree.c
        CC      jimsh.o
        CC      tree.o
        TCLEXT  _initjimsh.c
        CC      initjimsh.o
        CC      _load-static-exts.o
        AR      libjim.a
        LINK    jimsh
Making all in src/jtag/drivers/libjaylink
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in libjaylink
/bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden -I/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0 -g -O2 -MT libjaylink_la-buffer.lo -MD -MP -MF .deps/libjaylink_la-buffer.Tpo -c -o libjaylink_la-buffer.lo `test -f 'buffer.c' || echo './'`buffer.c
/bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden -I/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0 -g -O2 -MT libjaylink_la-core.lo -MD -MP -MF .deps/libjaylink_la-core.Tpo -c -o libjaylink_la-core.lo `test -f 'core.c' || echo './'`core.c
/bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden -I/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0 -g -O2 -MT libjaylink_la-device.lo -MD -MP -MF .deps/libjaylink_la-device.Tpo -c -o libjaylink_la-device.lo `test -f 'device.c' || echo './'`device.c
/bin/sh ../libtool  --tag=CC --silent  --mode=compile gcc -DHAVE_CONFIG_H -I. -I..    -Wall -Wextra -Werror -fvisibility=hidden -I/opt/homebrew/Cellar/libusb/1.0.24/include/libusb-1.0 -g -O2 -MT libjaylink_la-discovery.lo -MD -MP -MF .deps/libjaylink_la-discovery.Tpo -c -o libjaylink_la-discovery.lo `test -f 'discovery.c' || echo './'`discovery.c
mv -f .deps/libjaylink_la-discovery.Tpo .deps/libjaylink_la-discovery.Plo
mv -f .deps/libjaylink_la-buffer.Tpo .deps/libjaylink_la-buffer.Plo
[... 88 lines cut ...]
        mv -f $depbase.Tpo $depbase.Po
./doc/openocd.texi:10195: warning: `.' or `,' must follow @xref, not `c'.
./doc/openocd.texi:10934: Unknown command `raggedright'.
./doc/openocd.texi:10939: Bad argument `raggedright' to `@end', using `table'.
./doc/openocd.texi:10939: @item found outside of an insertion block.
./doc/openocd.texi:10941: @item found outside of an insertion block.
./doc/openocd.texi:10943: @item found outside of an insertion block.
./doc/openocd.texi:10946: @item found outside of an insertion block.
./doc/openocd.texi:10948: @item found outside of an insertion block.
./doc/openocd.texi:10950: @item found outside of an insertion block.
./doc/openocd.texi:10952: @item found outside of an insertion block.
./doc/openocd.texi:10953: Unknown command `raggedright'.
./doc/openocd.texi:10957: Unmatched `@end'.
./doc/openocd.texi:10958: Unmatched `@end'.
makeinfo: Removing output file `doc/openocd.info' due to errors; use --force to preserve.
make[2]: *** [doc/openocd.info] Error 1
make[2]: *** Waiting for unfinished jobs....
mv -f src/jtag/drivers/.deps/libocdjtagdrivers_la-picoprobe.Tpo src/jtag/drivers/.deps/libocdjtagdrivers_la-picoprobe.Plo
mv -f src/jtag/drivers/.deps/libocdjtagdrivers_la-xds110.Tpo src/jtag/drivers/.deps/libocdjtagdrivers_la-xds110.Plo
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2`

`./bootstrap` and `./configure --enable-picoprobe --disable-werror` both run without error.

Expected behaviour: `make -j4` finishes cleanly, per the documentation.

Kristine1975, 2022-01-25:
I had that error, too. It means the installed version of `texinfo` is too old. This solved the problem for me:

1. Install a newer `texinfo`, e.g. using Homebrew: `brew install texinfo`.
2. Export the path to the updated `texinfo` like so: `export PATH="/usr/local/opt/texinfo/bin:$PATH"`.
3. Run the build by calling `make`.

aallan, 2022-01-25:
Just made a quick note of this, it'll be in the next docs update. Thanks.

closed 2022-01-25 by aallan

odaki, 2022-01-25:
For your reference.
When I installed texinfo in brew, it did not put symlink, so it was not used.
I solved this problem by using "brew link texinfo --force" to put symlink.
By the way, the installation binary path for Homebrew for M1 Mac is /opt/homebrew/bin.
Thank you.

dancesWithMachines, 2022-09-23:
> Just made a quick note of this, it'll be in the next docs update. Thanks.

I encountered the same issue, related to the wrong path beeing exported, and I spend too much time investigating what was going to be a trivial thing. Please include both paths in the docs, I shouldn't have waste time, looking through github issues for solution.

dancesWithMachines, 2022-09-24:
> I don't have a M1 Mac myself, but I guess we ought to expand the footnote here
I think this would be sufficient.

> to mention that a different path needs to be added to `PATH` on Apple Silicon Macs?
From my point of view, I think the both paths should be included. The question of how this will be done is of secondary importance.