raspberrypi/openocd #47: building on ubuntu 20.10

ignik, 2021-12-16:
1. Add to Makefile  '-Wno-stringop-overflow' for src/target/arm920t.c
< 	$(AM_V_CC)$(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) --mode=compile $(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(src_target_libtarget_la_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -MT src/target/libtarget_la-arm920t.lo -MD -MP -MF src/target/$(DEPDIR)/libtarget_la-arm920t.Tpo -c -o src/target/libtarget_la-arm920t.lo `test -f 'src/target/arm920t.c' || echo '$(srcdir)/'`src/target/arm920t.c
---
> 	$(AM_V_CC)$(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) --mode=compile $(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(src_target_libtarget_la_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -Wno-stringop-overflow -MT src/target/libtarget_la-arm920t.lo -MD -MP -MF src/target/$(DEPDIR)/libtarget_la-arm920t.Tpo -c -o src/target/libtarget_la-arm920t.lo `test -f 'src/target/arm920t.c' || echo '$(srcdir)/'`src/target/arm920t.c

2. Cut some tabs in ./src/flash/nor/numicro.c (avoid -Wmisleading-indentation)
1246c1246
< 			LOG_DEBUG("status: 0x%" PRIx32 "", status);
---
> 		LOG_DEBUG("status: 0x%" PRIx32 "", status);
1515c1515
< 				LOG_DEBUG("status: 0x%" PRIx32 "", status);
---
> 			LOG_DEBUG("status: 0x%" PRIx32 "", status);
1604c1604
< 					LOG_DEBUG("status: 0x%" PRIx32 "", status);
---
> 				LOG_DEBUG("status: 0x%" PRIx32 "", status);

lurch, 2021-12-16:
Sounds like these are possibly compiler-warnings in the [upstream OpenOCD](https://openocd.org/) code?

ignik, 2021-12-17:
Yes, this info for code authors. I did not study the problem meaningfully, the goal was to build a package for pico under the current ubuntu.

bgni, 2022-08-10:
Thanks! I ran in to this issue as well and your solution worked. For the lazy:
```
diff --git a/Makefile b/Makefile
index 9931e67..8a9dffa 100644
--- a/Makefile
+++ b/Makefile
@@ -16,6 +16,7 @@
 
 
 
+
 am__is_gnu_make = { \
   if test -z '$(MAKELEVEL)'; then \
     false; \
@@ -4590,8 +4591,9 @@ src/target/libtarget_la-arm9tdmi.lo: src/target/arm9tdmi.c
 #      $(AM_V_CC)source='src/target/arm9tdmi.c' object='src/target/libtarget_la-arm9tdmi.lo' libtool=yes \
 #      DEPDIR=$(DEPDIR) $(CCDEPMODE) $(depcomp) \
 #      $(AM_V_CC_no)$(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) --mode=compile $(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(src_target_libtarget_la_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -c -o src/target/libtarget_la-arm9tdmi.lo `test -f 'src/target/arm9tdmi.c' || echo '$(srcdir)/'`src/target/arm9tdmi.c
+
 src/target/libtarget_la-arm920t.lo: src/target/arm920t.c
-       $(AM_V_CC)$(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) --mode=compile $(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(src_target_libtarget_la_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -Wno-stringop-overflow -MT src/target/libtarget_la-arm920t.lo -MD -MP -MF src/target/$(DEPDIR)/libtarget_la-arm920t.Tpo -c -o src/target/libtarget_la-arm920t.lo `test -f 'src/target/arm920t.c' || echo '$(srcdir)/'`src/target/arm920t.c
+       $(AM_V_CC)$(LIBTOOL) $(AM_V_lt) --tag=CC $(AM_LIBTOOLFLAGS) $(LIBTOOLFLAGS) --mode=compile $(CC) $(DEFS) $(DEFAULT_INCLUDES) $(INCLUDES) $(src_target_libtarget_la_CPPFLAGS) $(CPPFLAGS) $(AM_CFLAGS) $(CFLAGS) -MT src/target/libtarget_la-arm920t.lo -MD -MP -MF src/target/$(DEPDIR)/libtarget_la-arm920t.Tpo -c -o src/target/libtarget_la-arm920t.lo `test -f 'src/target/arm920t.c' || echo '$(srcdir)/'`src/target/arm920t.c
        $(AM_V_at)$(am__mv) src/target/$(DEPDIR)/libtarget_la-arm920t.Tpo src/target/$(DEPDIR)/libtarget_la-arm920t.Plo
 #      $(AM_V_CC)source='src/target/arm920t.c' object='src/target/libtarget_la-arm920t.lo' libtool=yes \
 #      DEPDIR=$(DEPDIR) $(CCDEPMODE) $(depcomp) \
diff --git a/src/flash/nor/numicro.c b/src/flash/nor/numicro.c
index 1971daa..7609fa8 100644
--- a/src/flash/nor/numicro.c
+++ b/src/flash/nor/numicro.c
@@ -1243,7 +1243,7 @@ static uint32_t numicro_fmc_cmd(struct target *target, uint32_t cmd, uint32_t ad
                retval = target_read_u32(target, NUMICRO_FLASH_ISPTRG, &status);
                if (retval != ERROR_OK)
                        return retval;
-               LOG_DEBUG("status: 0x%" PRIx32 "", status);
+                       LOG_DEBUG("status: 0x%" PRIx32 "", status);
                if ((status & (ISPTRG_ISPGO)) == 0)
                        break;
                if (timeout-- <= 0) {
@@ -1512,7 +1512,7 @@ static int numicro_erase(struct flash_bank *bank, unsigned int first,
                        retval = target_read_u32(target, NUMICRO_FLASH_ISPTRG, &status);
                        if (retval != ERROR_OK)
                                return retval;
-                       LOG_DEBUG("status: 0x%" PRIx32 "", status);
+                               LOG_DEBUG("status: 0x%" PRIx32 "", status);
                        if (status == 0)
                                break;
                        if (timeout-- <= 0) {
@@ -1601,7 +1601,7 @@ static int numicro_write(struct flash_bank *bank, const uint8_t *buffer,
                                retval = target_read_u32(target, NUMICRO_FLASH_ISPTRG, &status);
                                if (retval != ERROR_OK)
                                        return retval;
-                               LOG_DEBUG("status: 0x%" PRIx32 "", status);
+                                       LOG_DEBUG("status: 0x%" PRIx32 "", status);
                                if (status == 0)
                                        break;
                                if (timeout-- <= 0) {

```

P33M, 2023-07-07:
Any changes proposed here should be submitted upstream.

closed 2023-07-07 by P33M