raspberrypi/openocd #53: MacOS build on Catalina 10.15 fails at libusb

kpishere, 2022-02-03:
On building the openocd, my build is failing at the following step.  I've downloaded the source for libusb and repaced all non x86_64 versions.  Still, I get this error.  Is there a copy of the library in a pre-aggregated one somewhere?  Not sure what to do.

```
gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -o src/openocd src/main.o  src/.libs/libopenocd.a -L/usr/local/lib /usr/local/lib/libusb.dylib -L/usr/local/Cellar/libftdi/1.5_2/lib -lftdi1 -L/usr/local/Cellar/hidapi/0.11.2/lib -lhidapi /usr/local/lib/libusb-1.0.dylib -lobjc -lm -L/usr/local/Cellar/capstone/4.0.2/lib -lcapstone ./jimtcl/libjim.a -pthread -v
Apple clang version 12.0.0 (clang-1200.0.32.29)
Target: x86_64-apple-darwin19.6.0
Thread model: posix
InstalledDir: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin
 "/Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/ld" -demangle -lto_library /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/libLTO.dylib -dynamic -arch x86_64 -platform_version macos 10.15.0 10.15.6 -syslibroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk -o src/openocd -L/usr/local/lib -L/usr/local/Cellar/libftdi/1.5_2/lib -L/usr/local/Cellar/hidapi/0.11.2/lib -L/usr/local/Cellar/capstone/4.0.2/lib -L/usr/local/lib src/main.o src/.libs/libopenocd.a /usr/local/lib/libusb.dylib -lftdi1 -lhidapi /usr/local/lib/libusb-1.0.dylib -lobjc -lm -lcapstone ./jimtcl/libjim.a -lSystem /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/lib/clang/12.0.0/lib/darwin/libclang_rt.osx.a
Undefined symbols for architecture x86_64:
  "_usb_bulk_read", referenced from:
      _armjtagew_usb_message in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_reset in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      _usbprog_reset in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      ...
  "_usb_bulk_write", referenced from:
      _armjtagew_usb_message in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _armjtagew_reset in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _ep1_generic_commandl in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      _usbprog_reset in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      ...
  "_usb_claim_interface", referenced from:
      _armjtagew_init in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_close", referenced from:
      _armjtagew_quit in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_quit in libopenocd.a(libocdjtagdrivers_la-rlink.o)
     (maybe you meant: _transport_usb_close)
  "_usb_device", referenced from:
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
     (maybe you meant: _versaloon_usb_device_handle)
  "_usb_find_busses", referenced from:
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
  "_usb_find_devices", referenced from:
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
  "_usb_get_busses", referenced from:
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
  "_usb_init", referenced from:
      _armjtagew_init in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_open", referenced from:
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
     (maybe you meant: _jtag_usb_open, _transport_usb_open )
  "_usb_release_interface", referenced from:
      _rlink_quit in libopenocd.a(libocdjtagdrivers_la-rlink.o)
  "_usb_set_altinterface", referenced from:
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_set_configuration", referenced from:
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_set_debug", referenced from:
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_strerror", referenced from:
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_reset in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
ld: symbol(s) not found for architecture x86_64
clang: error: linker command failed with exit code 1 (use -v to see invocation)

```

kpishere, 2022-02-03:
the check with `find /usr/local/Cellar/ -name *.a -exec lipo -i {} \; | grep -v x86_64`... 

```
Non-fat file: /usr/local/lib/libusb-1.0.dylib is architecture: x86_64
Architectures in the fat file: /usr/local/lib/libusbpp-0.1.4.dylib are: i386 ppc7400 x86_64 
fatal error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/lipo: can't figure out the architecture type of: /usr/local/lib/libusb.la
Non-fat file: /usr/local/lib/libusb-1.0.0.dylib is architecture: x86_64
Non-fat file: /usr/local/lib/libusbpp.dylib is architecture: x86_64
Non-fat file: /usr/local/lib/libusb-1.0.a is architecture: x86_64
fatal error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/lipo: can't figure out the architecture type of: /usr/local/lib/libusb-1.0.la
fatal error: /Applications/Xcode.app/Contents/Developer/Toolchains/XcodeDefault.xctoolchain/usr/bin/lipo: can't figure out the architecture type of: /usr/local/Cellar/stlink/1.7.0/include/stlink/libusb_settings.h
```

kpishere, 2022-02-03:
I try to build with gcc-11 and the switch -v does yield more information ..

```
 gcc-11 -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2 -o src/openocd src/main.o  src/.libs/libopenocd.a -L/usr/local/lib -lusb -L/usr/local/Cellar/libftdi/1.5_2/lib -lftdi1 -L/usr/local/Cellar/hidapi/0.11.2/lib -lhidapi  -lm -L/usr/local/Cellar/capstone/4.0.2/lib -lcapstone ./jimtcl/libjim.a -v
Using built-in specs.
COLLECT_GCC=gcc-11
COLLECT_LTO_WRAPPER=/usr/local/Cellar/gcc/11.2.0_3/bin/../libexec/gcc/x86_64-apple-darwin19/11/lto-wrapper
Target: x86_64-apple-darwin19
Configured with: ../configure --prefix=/usr/local/opt/gcc --libdir=/usr/local/opt/gcc/lib/gcc/11 --disable-nls --enable-checking=release --with-gcc-major-version-only --enable-languages=c,c++,objc,obj-c++,fortran,d --program-suffix=-11 --with-gmp=/usr/local/opt/gmp --with-mpfr=/usr/local/opt/mpfr --with-mpc=/usr/local/opt/libmpc --with-isl=/usr/local/opt/isl --with-zstd=/usr/local/opt/zstd --with-pkgversion='Homebrew GCC 11.2.0_3' --with-bugurl=https://github.com/Homebrew/homebrew-core/issues --enable-libphobos --build=x86_64-apple-darwin19 --with-system-zlib --disable-multilib --with-native-system-header-dir=/usr/include --with-sysroot=/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk
Thread model: posix
Supported LTO compression algorithms: zlib zstd
gcc version 11.2.0 (Homebrew GCC 11.2.0_3) 
COMPILER_PATH=/usr/local/Cellar/gcc/11.2.0_3/bin/../libexec/gcc/x86_64-apple-darwin19/11/:/usr/local/Cellar/gcc/11.2.0_3/bin/../libexec/gcc/
LIBRARY_PATH=/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11/:/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/:/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11/../../../
COLLECT_GCC_OPTIONS='-Wall' '-Wstrict-prototypes' '-Wformat-security' '-Wshadow' '-Wextra' '-Wno-unused-parameter' '-Wbad-function-cast' '-Wcast-align' '-Wredundant-decls' '-Wpointer-arith' '-Werror' '-g' '-O2' '-o' 'src/openocd' '-L/usr/local/lib' '-L/usr/local/Cellar/libftdi/1.5_2/lib' '-L/usr/local/Cellar/hidapi/0.11.2/lib' '-L/usr/local/Cellar/capstone/4.0.2/lib' '-v' '-mmacosx-version-min=10.15.0' '-asm_macosx_version_min=10.15' '-mtune=core2' '-dumpdir' 'src/openocd.'
 /usr/local/Cellar/gcc/11.2.0_3/bin/../libexec/gcc/x86_64-apple-darwin19/11/collect2 -syslibroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/ -dynamic -arch x86_64 -macosx_version_min 10.15.0 -weak_reference_mismatches non-weak -o src/openocd -L/usr/local/lib -L/usr/local/Cellar/libftdi/1.5_2/lib -L/usr/local/Cellar/hidapi/0.11.2/lib -L/usr/local/Cellar/capstone/4.0.2/lib -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11 -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11/../../.. src/main.o src/.libs/libopenocd.a -lusb -lftdi1 -lhidapi -lcapstone ./jimtcl/libjim.a -lSystem -lgcc_ext.10.5 -lgcc -lSystem -no_compact_unwind -v -idsym
collect2 version 11.2.0
/usr/bin/ld -syslibroot /Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/ -dynamic -arch x86_64 -macosx_version_min 10.15.0 -weak_reference_mismatches non-weak -o src/openocd -L/usr/local/lib -L/usr/local/Cellar/libftdi/1.5_2/lib -L/usr/local/Cellar/hidapi/0.11.2/lib -L/usr/local/Cellar/capstone/4.0.2/lib -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11 -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc -L/usr/local/Cellar/gcc/11.2.0_3/bin/../lib/gcc/11/gcc/x86_64-apple-darwin19/11/../../.. src/main.o src/.libs/libopenocd.a -lusb -lftdi1 -lhidapi -lcapstone ./jimtcl/libjim.a -lSystem -lgcc_ext.10.5 -lgcc -lSystem -no_compact_unwind -v
@(#)PROGRAM:ld  PROJECT:ld64-609.8
BUILD 15:07:46 Dec 18 2020
configured to support archs: armv6 armv7 armv7s arm64 arm64e arm64_32 i386 x86_64 x86_64h armv6m armv7k armv7m armv7em
Library search paths:
	/usr/local/lib
	/usr/local/Cellar/libftdi/1.5_2/lib
	/usr/local/Cellar/hidapi/0.11.2/lib
	/usr/local/Cellar/capstone/4.0.2/lib
	/usr/local/Cellar/gcc/11.2.0_3/lib/gcc/11/gcc/x86_64-apple-darwin19/11
	/usr/local/Cellar/gcc/11.2.0_3/lib/gcc/11/gcc
	/usr/local/Cellar/gcc/11.2.0_3/lib/gcc/11
	/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/usr/lib
Framework search paths:
	/Library/Developer/CommandLineTools/SDKs/MacOSX10.15.sdk/System/Library/Frameworks/
Undefined symbols for architecture x86_64:
  "_usb_bulk_read", referenced from:
      _armjtagew_usb_message in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_reset in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      _usbprog_reset in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
      ...
  "_usb_bulk_write", referenced from:
      _armjtagew_usb_message in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _armjtagew_reset in libopenocd.a(libocdjtagdrivers_la-arm-jtag-ew.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _ep1_generic_commandl in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
[... cut ...]
      _usbprog_init in libopenocd.a(libocdjtagdrivers_la-usbprog.o)
  "_usb_strerror", referenced from:
      _rlink_init in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_speed in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _rlink_reset in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _dtc_queue_run in libopenocd.a(libocdjtagdrivers_la-rlink.o)
      _jtag_usb_open in libopenocd.a(libocdjtagdrivers_la-usb_common.o)
ld: symbol(s) not found for architecture x86_64
collect2: error: ld returned 1 exit status
Mac-mini:openocd kevinpeck$ 
```

Kristine1975, 2022-02-04:
gcc tries to link against `libusb.{a,dylib}`, i.e. without version in the filename, but it looks like you only have `/usr/local/lib/libusb-1.0.0.dylib`, i.e. with a version in the filename? If that is correct, have you tried creating a symlink `/usr/local/lib/libusb.dylib` to `/usr/local/lib/libusb-1.0.0.dylib`?

lurch, 2022-02-04:
> P.S: Is there a particular reason you've installed libusb manually instead of via Homebrew?

Indeed, using Homebrew is what we recommend in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

kpishere, 2022-02-04:
@lurch Thanks for the input.  Actually, I'd added an update but it looks like I didn't save it as I don't see it here. :(  

Anyhow, hes homebrew was first choice.  My mac has been upgraded since 10.6 days so, it appears to have some legacy stuff.  From this trouble I'd noticed I still had power pc binaries in my library collection.  So, did some cleanup.  The hangup was with the USB library though.  The default homebrew didn't appear to have what I'd needed.

I'd messed with the links a lot to summarize.

I ended up installing with this option from brew: `brew --HEAD --build-from-source  openocd`.  With that option, I'd noticed it downloaded and configured the libusb differently (and correctly) this time so it could be found by cmake/configure.

Then, I'd built this latest branch and installed over top of it to get this variant with the raspbian additions.  

Also, I'd had to give up on using the clone STLink SWD debugger, it just wasn't working and reverted to my JLink one which also does SWD and it works a treat now at 10Mbs.

closed 2022-02-04 by kpishere

lurch, 2022-02-04:
As a non-Mac-user most of that went over my `--HEAD`, but I'm glad to hear that you've got things working now :+1: