raspberrypi/openocd #56: Send changes upstream

rkta, 2022-02-14:
Any chance this is going upstream? I don't really want to maintain a
separate install just for the RPi pico.

JamesH65, 2022-02-14:
AIUI, sending upstream is not a problem (we may already have done so), getting upstream to acknowledge and merge them has proven more difficult.

aallan, 2022-02-14:
See for instance #39.

stevemarple, 2022-08-16:
+1 for sending the changes upstream.

Is there any possibility of sending the picoprobe interface code upstream? It would be great to have the option of using a Pico as generic SWD and JTAG programmer in the official OpenOCD release.

vulpes2, 2022-11-13:
picoprobe has been updated recently to include a CMSIS compatible implementation: raspberrypi/picoprobe#31, so perhaps a separate openocd fork is no longer necessary.

P33M, 2022-11-14:
Work is actively ongoing to reduce the delta between our downstream patches and upstream. openocd 0.12-rc2 is useable, albeit without SMP support. There will be a brief interlude in adding new features as the release candidate cycle needs to complete, but 0.13 should end up at feature parity with downstream.

closed 2022-11-14 by P33M