raspberrypi/openocd #58: Flashing Pico w/ Jlink

Machine-Hum, 2022-03-08:
I'm not sure if this is the right place to ask this question, if not I apologise. I'm trying to flash my Pico with a Jlink.

```
[machinehum@whitebox rp2040-test]$ ls
blink  blink.elf  flash.sh  rp2040.cfg
[machinehum@whitebox rp2040-test]$ cat flash.sh 
#!/bin/bash

# openocd -f interface/jlink.cfg -f rp2040.cfg -c "program blink/blink.elf verify reset exit"
openocd -f interface/jlink.cfg -f target/rp2040.cfg -c "program blink/blink.elf verify reset exit"
[machinehum@whitebox rp2040-test]$ ./flash.sh 
Open On-Chip Debugger 0.11.0-g610f137d2 (2022-03-07-19:06)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : auto-selecting first available session transport "jtag". To override use 'transport select <transport>'.
/usr/local/bin/../share/openocd/scripts/target/swj-dp.tcl:28: Error: Unknown param: 0x01002927, try one of: -irlen, -irmask, -ircapture, -enable, -disable, -expected-id, -ignore-version, -dp-id, or -instance-id
in procedure 'script' 
at file "embedded:startup.tcl", line 26
in procedure 'swj_newdap' called at file "/usr/local/bin/../share/openocd/scripts/target/rp2040.cfg", line 8
at file "/usr/local/bin/../share/openocd/scripts/target/swj-dp.tcl", line 28
```

Any help would be appreciated.

PetteriAimonen, 2022-03-08:
It seems that openocd is selecting `jtag` as the default transport, while the `rp2040.cfg` has SWD-only parameters.
Putting `transport select swd` somewhere might help.

lurch, 2022-03-08:
I've not used JLink myself (and IIRC we don't "officially" support it), but it's mentioned in [other issues](https://github.com/raspberrypi/openocd/search?q=jlink&type=issues), which might give you some useful pointers?

Machine-Hum, 2022-03-08:
Yeah I got this working with something like this...

```
openocd -f interface/jlink.cfg -c "transport select swd" -c "adapter_khz 6000" -f target/rp2040.cfg  -c "program blink.elf reset exit"
```

closed 2022-03-08 by Machine-Hum