raspberrypi/openocd #59: Cannot flash nrf51822 using Picoprobe and windows build of OpenOCD

Python380, 2022-06-02:
I cannot flash an nrf51822 using a Picoprobe and OpenOCD on windows (I know this is a copy that supports the picoprobe). 

Error message from openocd, using the same arguments as Issue #37:
```
Open On-Chip Debugger 0.10.0+dev-geb22ace (2021-04-12-11:02)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
Warn : Transport "swd" was already selected
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
target halted due to debug-request, current mode: Thread
xPSR: 0x61000000 pc: 0x00002fa0 msp: 0x20003d10
Info : nRF51822-QFAA(build code: H1) 256kB Flash, 16kB RAM
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
Info : Flash write discontinued at 0x000007c0, next section at 0x00001000
Error: flash write algorithm aborted by target
Error: timed out while waiting for target halted
```

Output from ```arm-none-eabi-gdb firmware.hex```
```
GNU gdb (GNU Arm Embedded Toolchain 10-2020-q4-major) 10.1.90.20201028-git
Copyright (C) 2020 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "--host=i686-w64-mingw32 --target=arm-none-eabi".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word"...
Reading symbols from firmware.hex...
(No debugging symbols found in firmware.hex)
(gdb) target remote localhost:3333
Remote debugging using localhost:3333
0x00002fa0 in ?? ()
(gdb) load
Loading section .sec1, size 0x7c0 lma 0x0
Loading section .sec2, size 0xf000 lma 0x1000
Loading section .sec3, size 0xb4d8 lma 0x10000
Loading section .sec4, size 0x2c6c lma 0x1c000
Error finishing flash operation
(gdb) Quit
(gdb) reset
Undefined command: "reset".  Try "help".
(gdb) monitor reset
(gdb)
```

I have already tried the fix that #37 implemented, but I am unable to get a working result...

Python380, 2022-06-02:
Will try latest (Just noticed the version I got wasn't latest)

Python380, 2022-06-03:
Updating versions got me closer, and also made so that I don't need to implement the workaround in #37.

logs from OpenOCD now:
```
Open On-Chip Debugger 0.11.0-g4f2ae61 (2022-06-02-16:52)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Error: nrf51.cpu -- clearing lockup after double fault
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : nRF51822-QFAA(build code: H1) 256kB Flash, 16kB RAM
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
Info : Flash write discontinued at 0x000007c0, next section at 0x00001000
Info : Flash write discontinued at 0x0001b4d8, next section at 0x0001c000
Info : SWD DPIDR 0x0bb11477
Error: Failed to write memory at 0x20000034
Error: Failed to write to nrf5 flash
Error: error writing to flash at address 0x00000000 at offset 0x00001000
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
```

Python380, 2022-06-03:
Could not write to 0x00000000 due to memory protection. ```monitor nrf51 mass_erase``` in gdb terminal fixes this.

New errors:
```
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 1000 kHz
Info : SWD DPIDR 0x0bb11477
Info : nrf51.cpu: hardware has 4 breakpoints, 2 watchpoints
Error: nrf51.cpu -- clearing lockup after double fault
Info : starting gdb server for nrf51.cpu on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : nRF51822-QFAA(build code: H1) 256kB Flash, 16kB RAM
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0xfffffffe msp: 0xfffffffc
Info : Flash write discontinued at 0x000007c0, next section at 0x00001000
Info : Flash write discontinued at 0x0001b4d8, next section at 0x0001c000
Info : SWD DPIDR 0x0bb11477
Error: Failed to write memory at 0x20000044
Error: Failed to write to nrf5 flash
Error: error writing to flash at address 0x00000000 at offset 0x00001000
target halted due to debug-request, current mode: Thread
xPSR: 0xc1000000 pc: 0x000006d0 msp: 0x000007c0
```

Python380, 2022-06-03:
May have found a fix... can't remember what I did though.
Will close for now.

Python380, 2022-06-03:
For those of you finding this issue later, I did a bit of searching and found the workaround I used. Run openocd with this command (May need tweaking so that it works for your setup): ```-f interface/picoprobe.cfg -c "transport select swd; set WORKAREASIZE 0" -f target/nrf51.cfg```

```WORKAREASIZE``` forces openocd to use the slow writing procedure, which I found to be more reliable.

lurch, 2022-06-04:
This is a `raspberrypi` repository, so I doubt we'll be providing help ourselves with getting a `nrf51822` working :wink: