raspberrypi/openocd #60: ./bootstrap fails ...  repo.or.cz is down?

gbin, 2022-06-14:
when I try to compile the project it fails at the first ./bootstrap stage:
```
sal ➜  openocd (rp2040) ✗ ./bootstrap
+ aclocal --warnings=all
+ libtoolize --automake --copy
+ autoconf --warnings=all
+ autoheader --warnings=all
+ automake --warnings=all --gnu --add-missing --copy
Setting up submodules
Cloning into '/home/gbin/projects/openocd/src/jtag/drivers/libjaylink'...
fatal: unable to access 'https://repo.or.cz/libjaylink.git/': Failed to connect to repo.or.cz port 443 after 128844 ms: Connection timed out
fatal: clone of 'https://repo.or.cz/libjaylink.git' into submodule path '/home/gbin/projects/openocd/src/jtag/drivers/libjaylink' failed
Failed to clone 'src/jtag/drivers/libjaylink'. Retry scheduled
Cloning into '/home/gbin/projects/openocd/tools/git2cl'...
```
it looks like repo.or.cz as a host is not responsive.

lurch, 2022-06-14:
"Duplicate" of #49 ? ping @liamfraser

pfletch101, 2022-06-15:
repo.or.cz is still down, and it looks as if there have been (different) problems with it before. libjaylink appears to have a repo here on github ([https://github.com/syntacore/libjaylink](url), but I don't know whether it is up to date. If it is, could the links be updated?

pfletch101, 2022-06-15:
There have been messages on the RPi forum about this, and there seems to be good reason to believe that repo.or.cz is down for good (or at least for a long and indefinite time).

bikerglen, 2022-06-15:
It's not just libjaylink. It's also something else called git2cl:

`fatal: clone of 'https://repo.or.cz/libjaylink.git' into submodule path '/home/glen/pico/openocd/src/jtag/drivers/libjaylink' failed`

and

`fatal: clone of 'https://repo.or.cz/git2cl.git' into submodule path '/home/glen/pico/openocd/tools/git2cl' failed`

I messaged the server's maintainers on Twitter who stated the server has a failed PSU and there is no word on when or if it will be fixed.

bikerglen, 2022-06-16:
Looks like there's an open pull request in #49 from January 13, 2022, that points the git submodules at new repos. I'm not the greatest with git but tomorrow I'm going to clone the raspberrypi/openocd:rp2040 branch then attempt to apply the changes from #49 then run the bootstrap script again and see if I can at least get my local copy working.

dwhitsed, 2022-06-16:
I don't know git well enough to work out how to do this. Please post an update if you manage to get it working.

dwhitsed, 2022-06-16:
It's working properly now thank you

closed 2022-06-20 by liamfraser