raspberrypi/openocd #61: bcm2835gpio not available after install

nouun, 2022-06-18:
Following the steps from [getting-started-with-pico.pdf](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf):
```sh
$ git clone https://github.com/raspberrypi/openocd.git --recursive --branch rp2040 --depth=1
$ cd openocd
$ ./bootstrap
$ ./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio
$ make -j4
$ sudo make install
```

I get the following output after configuring:
```
OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes
Raspberry Pi Pico Probe                 yes (auto)
ST-Link Programmer                      yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes (auto)
Altera USB-Blaster II Compatible        yes (auto)
Bitbang mode of FT232R based devices    yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Andes JTAG Programmer                   yes (auto)
USBProg JTAG Programmer                 yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
CMSIS-DAP Compliant Debugger            no
Nu-Link Programmer                      no
Cypress KitProg Programmer              no
Altera USB-Blaster Compatible           yes (auto)
ASIX Presto Adapter                     yes (auto)
OpenJTAG Adapter                        yes (auto)
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                yes (auto)
Use Capstone disassembly framework      no
```

Then after trying to run `openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg -c "program blink/blink.elf
verify reset exit"` I get:
```
Open On-Chip Debugger 0.11.0-g228ede4 (2022-06-19-03:39)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (bcm2835gpio)
The following debug adapters are available:
1: ftdi
2: usb_blaster
3: ft232r
4: presto
5: usbprog
6: openjtag
7: jlink
8: vsllink
9: rlink
10: ulink
11: arm-jtag-ew
12: hla
13: osbdm
14: opendous
15: sysfsgpio
16: aice
17: picoprobe
18: cmsis-dap
19: xds110
20: st-link
```

Running on Ubuntu 22.04 if that makes a difference.

mcuee, 2022-07-13:
> Linux GPIO bitbang through libgpiod     no

You probably want to install libgpiod development package.

mcuee, 2022-07-17:
Duplicate of 
* #23.

Solution in #23 as well.
> > This is not a bug, you just forgot to add `--enable-bcm2835gpio` to your `./configure` command, you should always check what flags can be enabled by doing `./configure --help`.

mcuee, 2022-07-17:
So this can be closed.

nouun, 2022-07-17:
> Duplicate of
> 
> * [Linux compile error #23](https://github.com/raspberrypi/openocd/issues/23).
> 
> Solution in #23 as well.
> 
> > This is not a bug, you just forgot to add `--enable-bcm2835gpio` to your `./configure` command, you should always check what flags can be enabled by doing `./configure --help`.

If you look at the configure command I used, it includes the flag you mentioned was missing. Will close anyway as I don’t require this anymore.

closed 2022-07-17 by nouun