raspberrypi/openocd #62: RPi and aduc7020

thijn1234, 2022-06-20:
I want to do the following, I use the RPi as a chip programmer via JTAG / SWD, I have managed to program schellinde chips. now i would like to program the aduc7020. but that does not work. Anyone have a fix or can help me? I tried alot

openocd.cfg:

#log_output testlog 

source [find interface/raspberrypi2-native.cfg]
transport select jtag

bcm2835gpio trst_num 8
bcm2835gpio tdi_num 7
bcm2835gpio tms_num 1
bcm2835gpio tck_num 12
bcm2835gpio tdo_num 16
 
adapter speed 400

set CHIPNAME aduc7020
source [find target/aduc702x.cfg]
 
# reset config
reset_config trst_only

#jtag_ntrst_assert_width 200
jtag_ntrst_delay 200

#init
echo "target try halted"
#halt

soft_reset_halt

sleep 2000

#flash erase_address 0x80000 0x0F800

flash write_bank 0 testradar.hex 
#flash write_image testradar.hex 

#verify_image testradar.bin 0x80000

exit

Return from openocd: 

/Radarchip $ openocd
Open On-Chip Debugger 0.11.0+dev-00590-g2586fec92 (2022-03-10-10:20)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
/usr/local/bin/../share/openocd/scripts/target/aduc702x.cfg:52: Error: -event: Unknown: reset-halt-post, try one of: gdb-halt, halted, resumed, resume-start, resume-end, step-start, step-end, gdb-start, gdb-end, reset-start, reset-assert-pre, reset-assert, reset-assert-post, reset-deassert-pre, reset-deassert-post, reset-init, reset-end, examine-start, examine-fail, examine-end, debug-halted, debug-resumed, gdb-attach, gdb-detach, gdb-flash-write-start, gdb-flash-write-end, gdb-flash-erase-start, gdb-flash-erase-end, trace-config, semihosting-user-cmd-0x100, semihosting-user-cmd-0x101, semihosting-user-cmd-0x102, semihosting-user-cmd-0x103, semihosting-user-cmd-0x104, semihosting-user-cmd-0x105, semihosting-user-cmd-0x106, or semihosting-user-cmd-0x107
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "openocd.cfg", line 16
at file "/usr/local/bin/../share/openocd/scripts/target/aduc702x.cfg", line 52
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 400 kHz
Info : JTAG tap: aduc7020.cpu tap/device found: 0x3f0f0f0f (mfg: 0x787 (<unknown>), part: 0xf0f0, ver: 0x3)
Info : Embedded ICE version 1
Info : aduc7020.cpu: hardware has 2 breakpoint/watchpoint units
Info : starting gdb server for aduc7020.cpu on 3333
Info : Listening on port 3333 for gdb connections

lurch, 2022-06-20:
This sounds like a question for the [forums](https://forums.raspberrypi.com/) ? Raspberry Pi won't be providing support for third-party products.

thijn1234, 2022-06-20:
it looks like an error into de default configuration fila from openocd aduc7020x 

/usr/local/bin/../share/openocd/scripts/target/aduc702x.cfg:52: Error: -event: Unknown: reset-halt-post, try one of: gdb-halt, halted, resumed, resume-start, resume-end, step-start, step-end, gdb-start, gdb-end, reset-start, reset-assert-pre, reset-assert, reset-assert-post, reset-deassert-pre, reset-deassert-post, reset-init, reset-end, examine-start, examine-fail, examine-end, debug-halted, debug-resumed, gdb-attach, gdb-detach, gdb-flash-write-start, gdb-flash-write-end, gdb-flash-erase-start, gdb-flash-erase-end, trace-config, semihosting-user-cmd-0x100, semihosting-user-cmd-0x101, semihosting-user-cmd-0x102, semihosting-user-cmd-0x103, semihosting-user-cmd-0x104, semihosting-user-cmd-0x105, semihosting-user-cmd-0x106, or semihosting-user-cmd-0x107
in procedure 'script'
at file "embedded:startup.tcl", line 26
at file "openocd.cfg", line 16
at file "/usr/local/bin/../share/openocd/scripts/target/aduc702x.cfg", line 52

mcuee, 2022-07-17:
This is not relevant for this repo and can be closed.

closed 2023-07-07 by P33M