raspberrypi/openocd #65: bcm2835gpio interface not found even thought --enable-bcm2835gpio

mistrjirka, 2022-07-29:
My system is Manjaro Cinnamon kernel 5.17.5-1, Ryzen 5 1600, RX480
I did follow the installation instructions for openocd in getting started, apart from some Debian-only steps (installing libraries with apt, I made sure I had those).
I made sure that there is no other openocd in the system.

Error when trying to run " openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg" in pico-examples folder:

```
Open On-Chip Debugger 0.11.0-g4f2ae6197 (2022-07-29-17:23)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Error: The specified debug interface was not found (bcm2835gpio)
The following debug adapters are available:
1: parport
2: dummy
3: ftdi
4: usb_blaster
5: jtag_vpi
6: ft232r
7: amt_jtagaccel
8: gw16012
9: presto
10: usbprog
11: openjtag
12: jlink
13: vsllink
14: rlink
15: ulink
16: arm-jtag-ew
17: buspirate
18: remote_bitbang
19: hla
20: osbdm
21: opendous
22: sysfsgpio
23: xlnx_pcie_xvc
24: aice
25: picoprobe
26: cmsis-dap
27: kitprog
28: xds110
29: st-link

```
step to reproduce:
```
git clone https://github.com/raspberrypi/openocd.git --recursive --branch rp2040 --depth=1
$ cd openocd
$ ./bootstrap
$ ./configure --enable-ftdi --enable-sysfsgpio --enable-bcm2835gpio
$ make -j4
$ sudo make install
```
What I tried?
doing it more times :D didn't work obviously.
Trying PAMAC package: openocd-picoprobe - no difference

output of git:

[git.log](https://github.com/raspberrypi/openocd/files/9221539/git.log)

output of bootstrap:

[bootstrap.log](https://github.com/raspberrypi/openocd/files/9221534/bootstrap.log)

output of configure: 

[configureouput.log](https://github.com/raspberrypi/openocd/files/9221531/configureouput.log)

make output:
[ouput.make.log](https://github.com/raspberrypi/openocd/files/9221521/ouput.make.log)
sudo make install output:

[sudomakeinstall.log](https://github.com/raspberrypi/openocd/files/9221528/sudomakeinstall.log)

lurch, 2022-07-29:
> My system is ... Ryzen 5 1600

`bcm2835gpio` is only for use with the GPIO pins on a Raspberry Pi, so it's not surprising that it doesn't work on a desktop PC?

mistrjirka, 2022-07-29:
> > My system is ... Ryzen 5 1600
> 
> `bcm2835gpio` is only for use with the GPIO pins on a Raspberry Pi, so it's not surprising that it doesn't work on a desktop PC?

ohh, my bad, thank you I misread/misunderstanded the documentation.
Thank you for the quick response

closed 2022-07-29 by mistrjirka