raspberrypi/openocd #67: Make this program portable

Waaflee, 2022-08-07:
is there any possibility to add a statically linked build option to the make file? This way the program could be "portable" as in no external dependencies needed. This is useful for publishing as a stand alone program for platformio or other platforms for example.

josuah, 2022-12-15:
This would likely be something to ask to the upstream OpenOCD, which would affect all forks of OpenOCD.

It is also possible to only install static libraries on your system so that they get picked instead of the shared version by the linker, which would produce a static binary in the end.

See also: https://github.com/raspberrypi/openocd/issues/56#issuecomment-1313326392

P33M, 2023-07-07:
The Pico Windows Installer ships a precompiled version of OpenOCD. We won't be shipping a static Linux binary.

closed 2023-07-07 by P33M