raspberrypi/openocd #68: libusb errors seen on both RPiOS and macOS

smittytone, 2022-08-11:
**Note** Previously posted in `raspberrypi/picoprobe`, but this is a more appropriate place.

With a fresh builds of openocd and picoprobe (using Pico Getting Started Guide instructions) on a Pi 400, I'm seeing:

```
sudo openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl                                    [ pico ]
Open On-Chip Debugger 0.11.0-g228ede4-dirty (2022-08-10-09:29)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : clock speed 5000 kHz
Error: libusb_bulk_read error: LIBUSB_ERROR_TIMEOUT
Error: libusb_bulk_read error: LIBUSB_ERROR_OVERFLOW
Info : DAP init failed

Error: libusb_bulk_read error: LIBUSB_ERROR_OVERFLOW
openocd: src/jtag/drivers/picoprobe.c:101: picoprobe_bulk_write: Assertion `pkt_hdr->total_packet_length <= PICOPROBE_MAX_PACKET_LENGTH' failed.
```

I did this on a Pi because I was seeing exactly the same issue under macOS (12.5 on an M1 Mac Mini) and wanted to see if I could get it working a more 'official' way.

Picoprobe is running on a Pico, as is the target.

Let me know if there's more info you need.

`libusb` is at 1.0.26

forlfs, 2022-08-18:
I encountered a similar problem under win7 + mingw64, and find out it was caused by memory alignment.
src/jtag/drivers/picoprobe.c
```
struct __attribute__((__packed__)) probe_cmd_hdr {
	uint8_t id;
	uint8_t cmd;
	uint32_t bits;
};

struct __attribute__((__packed__)) probe_pkt_hdr {
	uint32_t total_packet_length;
};
```
After I add #pragma pack(push, 1) and #pragma pack(pop), everthing is ok.
But I think it may not be a problem under RPiOS.

lurch, 2022-08-18:
ping @P33M

P33M, 2022-08-18:
This will go away once https://github.com/raspberrypi/picoprobe/pull/31 is merged - the picoprobe driver becomes redundant.

closed 2023-02-22 by smittytone