raspberrypi/openocd #71: Why can't I find picoprobe?; on macOS Monterey

TheABC1025, 2022-11-13:
# What problem
I wanted to use picoprobe for debuging Raspberry Pi pico as mentioned in Getting Started document.
So I tried build and use openocd as usual mentioned in Internet (and document):
`$ openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl`
but, i got these error logs:
```
Open On-Chip Debugger 0.11.0-g228ede43d (2022-11-11-01:03)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and permissions.
```
# What I tried
I will write what i tried:
## Hardware problem?
I think it is NOT hardware problems
I checked the connections and change USB-cable to another, but it doesnt got better.
(And I tried on other computer, but the tool couldn't go well)
## Software problem?
For solving permission, I tried abobe-mentioned command with `sudo` like this:
```sudo openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl -c "bindto 0.0.0.0"```
but I got same error.
the environment is here:
* macOS Monterey 12.6 (macbook pro 2017, Intel core-i7)
* OpenOCD (cloned on rp2040 branch, commit on [228ede4])
* Picoprobe(cloned on master, commit on [b1799bf]])
# Conclusion
So I think the tool cannot run well on macOS.
Is there anyone who had same problem like me?

thanks,

aallan, 2022-11-13:
You did upload the Picoprobe UF2 from here, https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html#debugging-using-another-raspberry-pi-pico, to the Pico you were using as a debugger?

TheABC1025, 2022-11-14:
No, I only tried clone picoprobe repos, and build and generates uf2 file by myself.
Does it prefer to upload uf2 file which is downloaded from this site?

lurch, 2022-11-14:
There's recently been some changes to Picoprobe https://github.com/raspberrypi/picoprobe/releases , so whilst we're getting all our ducks back in a row again, you might want to try the precompiled UF2 that @aallan has pointed to? :man_shrugging:

TheABC1025, 2022-11-14:
Thanks for your advices!
I downloaded uf2 file from web-site, and tried to load it to pico.
Then it works well, and I can debug normally.
Good luck for your fixing.

closed 2023-07-07 by P33M