raspberrypi/openocd #73: FreeRTOS awareness + SMP

maxgerhardt, 2023-01-15:
See discussion at https://github.com/earlephilhower/arduino-pico/discussions/1122.

Does this OpenOCD version have the capability to recognize FreeRTOS on top the symmetric multiprocessing awareness (dual-core Cortex-M0+)? 

The FreeRTOS port used is https://github.com/FreeRTOS/FreeRTOS-Kernel/tree/main/portable/ThirdParty/GCC/RP2040 as is e.g. used in the popular Arduino core https://github.com/earlephilhower/arduino-pico.

Trying to mess around with the rp2040.cfg file with `-rtos FreeRTOS` has not gotten me anywhere since this FreeRTOS version does not use `pxCurrentTCB` but the "multicore" version of with `pxCurrentTCBs` that OpenOCD version does not recognize, unlike the e.g. [ESP32 version.](https://github.com/espressif/openocd-esp32/blob/master/src/rtos/FreeRTOS.c#L283-L301)

It would be really nice to have GDB be able to decode the FreeRTOS tasks running on both Cortex-M0+s.

P33M, 2023-01-15:
Do you get the same results with the branch https://github.com/raspberrypi/openocd/tree/rp2040-0.12-rc2 ? If so, this isn't a downstream bug.

maxgerhardt, 2023-01-15:
Sure I can compile that branch and test. Do you suggest any changes to the `rp2040.cfg` too in regards to `-rtos FreeRTOS`?

earlephilhower, 2023-01-18:
@maxgerhardt that branch also doesn't understand `pxCurrentTCBs` so I guess this needs to be bumped upstream.

P33M, 2023-07-07:
Downstream is now the same major version release as upstream, so if there's a bug in freertos task context handling then raise the issue upstream.

closed 2023-07-07 by P33M