raspberrypi/openocd #75: OpenOCD can't find the CMSIS-DAP device, but pyocd can.

tmb, 2023-01-23:
New to hardware and MCUs in general; apologies in advance if there's something obvious going on here.

I have a Picoprobe with the latest firmware which is ostensibly connected correctly to another Pico. I built openocd out of this repo, which fails with the following log when running `src/openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -s tcl -d`: https://hastebin.com/nahocejomu.apache

That line `Debug: 162 16 cmsis_dap_usb_bulk.c:177 cmsis_dap_usb_open(): enumerating interfaces of 0x2e8a:0x0004` is intriguing to me because the Picoprobe is connected to my machine with the vendor ID `0x2e8a`.

I tried `pyocd` after getting this error repeatedly, which has been working well; I'd like to be using `openocd`, though.

P33M, 2023-01-23:
Which instructions are you following?

tmb, 2023-01-26:
Sorry for the delay in response. I'm following the PDF here: https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf

Specifically, I got the error after I hit page 66.

P33M, 2023-01-26:
Are you using the release uf2 from this repository?

fex95, 2024-01-09:
I had the same problem. The command to install the dependencies in `getting-started-with-pico.pdf` has a line break in it, which means that `mingw-w64-x86_64-libusb` will not be installed and therefore the `cmsis-dap` adapter will not be included in the OpenOCD build.

You can install the package later using `pacman -S mingw-w64-x86_64-libusb` and then perform the configuration and build again.

After the `./configure --disable-werror` command, check that the CMSIS-DAP v2 Compliant Debugger is enabled. My output looks like this:
```
OpenOCD configuration summary
--------------------------------------------------
MPSSE mode of FTDI based devices        yes (auto)
ST-Link Programmer                      yes (auto)
TI ICDI JTAG Programmer                 yes (auto)
Keil ULINK JTAG Programmer              yes (auto)
Altera USB-Blaster II Compatible        yes (auto)
Bitbang mode of FT232R based devices    yes (auto)
Versaloon-Link JTAG Programmer          yes (auto)
TI XDS110 Debug Probe                   yes (auto)
CMSIS-DAP v2 Compliant Debugger         yes (auto)
OSBDM (JTAG only) Programmer            yes (auto)
eStick/opendous JTAG Programmer         yes (auto)
Olimex ARM-JTAG-EW Programmer           yes (auto)
Raisonance RLink JTAG Programmer        yes (auto)
USBProg JTAG Programmer                 yes (auto)
Espressif JTAG Programmer               yes (auto)
Andes JTAG Programmer (deprecated)      no
CMSIS-DAP Compliant Debugger            no
Nu-Link Programmer                      no
Cypress KitProg Programmer              no
Altera USB-Blaster Compatible           no
ASIX Presto Adapter                     no
OpenJTAG Adapter                        no
Linux GPIO bitbang through libgpiod     no
SEGGER J-Link Programmer                yes (auto)
Bus Pirate                              no
Use Capstone disassembly framework      no
```

lurch, 2024-01-09:
> The command to install the dependencies in `getting-started-with-pico.pdf` has a line break in it, which means that `mingw-w64-x86_64-libusb` will not be installed

I believe the style used in the Pico / RP2040 databooks is that if a line starts with a `$` then it means that it's the start of a new command-line, but if it _doesn't_ start with a `$` then it's a continuation of the previous command-line (i.e. it all needs to be typed on a single line). See also https://github.com/raspberrypi/pico-feedback/issues/331

drankinatty, 2025-07-18:
On Linux, this mean you need to install `libusb-1_0` for cmsis-dap to be automatically enabled by `./configure`. You can explicitly enable the configure option(s)  for cmsis-dap with ` --enable-cmsis-dap[=yes/no/auto]` and `--enable-cmsis-dap-v2[=yes/no/auto]` which will then cause `./configure`  to fail if the necessary libraries are not installed.

On may distributions, the needed headers are broken out into a subpackage, e.g. `libusb-1_0-devel` so check your distributions available packages to determine what is needed.

Similarly, the TI XDS110 adapter requires `libhidapi` be installed before it will be automatically enabled by `./configure`.

You can check what can be enabled with `./configure --help` to obtain a list of supported adapters, and then provide the option to `./configure` to find out what additional library is needed.