raspberrypi/openocd #76: Using a Pico W for a picoprobe not working

d-adamson, 2023-01-30:
I have been naively trying to use a **PICO W** for my picoprobe. However, it does not appear to work properly.
See below for detailed error messages.

I did some investigating and found the following:
In the driver file picoprobe.c (line 33), the PID is hardcoded for a regular pico.

  #define **PID 0x0004** /* Picoprobe */

However, the PID for the **PICO W** is **0x0000**. I believe this is what is causing the device not to be found.

Command & Response (also tested using sudo and got the same output)
$ openocd -s ~/pico/openocd/tcl -f interface/picoprobe.cfg -f target/rp2040.cfg

Open On-Chip Debugger 0.11.0-g8e3c38f (2023-01-24-14:28)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to open or find the device
**Error: Can't find a picoprobe device! Please check device connections and permissions.**

P33M, 2023-01-30:
Which instruction document are you following?

d-adamson, 2023-01-31:
Getting started w Raspberry Pi Pico C/C++ development ...
 downloaded from the website:
 v2.0 01 Dec 2022.

Below is the section of code. The PID is hard-coded for the standard pico (0x0004), and not the pico w (0x0000).

~/pico/openocd/src/jtag/drivers/picoprobe.c

...
#include "libusb_helper.h"

#define VID 0x2E8A /* Raspberry Pi */
#define PID 0x0004 /* Picoprobe */
...

P33M, 2023-01-31:
Which website? Please provide a URL.

lurch, 2023-01-31:
https://github.com/raspberrypi/usb-pid says that PID `0x0004` is "Raspberry Pi PicoProbe" and PID `0x000C` is "Raspberry Pi RP2040 CMSIS-DAP debug adapter", so I think your comments about Pico W are just a red herring.

I see in your original post you have ` -f interface/picoprobe.cfg -f target/rp2040.cfg` but page 66 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf (which is the same v2.0 that you claim you're using) says to use ` -f interface/cmsis-dap.cfg -f target/rp2040.cfg`.

d-adamson, 2023-01-31:
@P33M 
Here is the URL.
[(https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf)](url)

@lurch 
Thanks. I didn't know about this document
[https://github.com/raspberrypi/usb-pid ](url)

Yes, I've tried using the instructions on page 66, but I had to add the adapter speed command as follows
$ openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000"

With this I can get GDB running from the command line, but not from VS Code. When VS Code launches the debugger, gdb-multiarch times out waiting for a connection to openocd. Inspecting the terminal window shows that VS Code sends the following command to launch openocd

$/usr/local/bin/openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f ~/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg

**This command specifies gdb_port 50000, but the debugger is expecting port 3333**. I cannot find where this command is being generated. There is nothing obvious in settings.json & launch.json. By the way, if I paste the above command into a terminal and modify the line to gdb_port 3333 then openocd and gdb_multiarch are happy to communicate again in the command line interface.

In any case, I will receive a regular pico (non WiFi) in a couple of days to use as the picoprobe, and I think most of these issues will go away. Originally, I purchased a 3-pack of pico W devices, and I just assumed everything would work the same as for the regular pico (except, of course, for the added wifi functionality).

Assuming using a regular pico this corrects the issues, it would be nice to either 
 * update the getting started doc to specify using standard pico for the picoprobe or
 * update the the picoprobe code to be compatible with both devices

In any case, I'll report back what I learn.

P33M, 2023-01-31:
The interface protocol selected must match what's on the picoprobe (both the setup instructions and published picoprobe firmware match), but gdb->openocd timeouts are merely a result of configuration differences.

The command is generated internally by the cortex-debug plugin unless config options are overridden in .vscode/launch.json. Please post your launch.json file.

d-adamson, 2023-01-31:
I've tried several. However, this is the launch.json file that seems to have gotten the furthest using cmsis-dap.cfg with some commands to set the search directory, the adapter speed and the port number.

{
    "version": "0.2.0",
    "configurations": [
        {
            "name": "Pico Debug",
            "cwd": "${workspaceRoot}",
            "executable": "${command:cmake.launchTargetPath}",
            "request": "launch",
            "type": "cortex-debug",
            "servertype": "openocd",
            // This may need to be arm-none-eabi-gdb depending on your system
            "gdbPath" : "gdb-multiarch",
            "device": "RP2040",
            "searchDir":["~/pico/openocd/tcl"],
            "openOCDLaunchCommands": ["adapter speed 5000", "gdb_port 3333"],
            "configFiles": [
                "interface/cmsis-dap.cfg",
                "target/rp2040.cfg"
            ],
            "svdFile": "${env:PICO_SDK_PATH}/src/rp2040/hardware_regs/rp2040.svd",
            "runToMain": true,
            // Work around for stopping at main on restart
            "postRestartCommands": [
                "break main",
                "continue"
            ]
        }
    ]
}

**Terminal Output**
openocd -c "gdb_port 50000" -c "tcl_port 50001" -c "telnet_port 50002" -s ~/pico/openocd/tcl -f /home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/support/openocd-helpers.tcl -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -c "gdb_port 3333"
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-01-24-14:28)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
CDRTOSConfigure
Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
adapter speed: 5000 kHz

Info : Listening on port 50001 for tcl connections
Info : Listening on port 50002 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E6614103E77D7B37
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
Info : dropped 'gdb' connection
shutdown command invoked
[... 38 lines cut ...]
-> ~"This GDB was configured as \"x86_64-linux-gnu\".\n"
-> ~"Type \"show configuration\" for configuration details.\n"
-> ~"For bug reporting instructions, please see:\n"
-> ~"<https://www.gnu.org/software/gdb/bugs/>.\n"
-> ~"Find the GDB manual and other documentation resources online at:\n    <http://www.gnu.org/software/gdb/documentation/>."
-> ~"\n\n"
-> ~"For help, type \"help\".\n"
-> ~"Type \"apropos word\" to search for commands related to \"word\".\n"
-> 1^done
0x100011b4 in ?? ()
2-gdb-set mi-async on
-> 2^error,msg="Cannot change this setting while the inferior is running."
3-interpreter-exec console "set print demangle on"
Failed to generate gdb commands: Cannot change this setting while the inferior is running. (from gdb-set mi-async on)
Cannot change this setting while the inferior is running.
    at i (/home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/dist/debugadapter.js:2:151139)
    at handlers.<computed> (/home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/dist/debugadapter.js:2:151633)
    at /home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/dist/debugadapter.js:2:138971
    at Array.forEach (<anonymous>)
    at b.onOutput (/home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/dist/debugadapter.js:2:138404)
    at b.stdout (/home/dadamson/.vscode/extensions/marus25.cortex-debug-1.6.10/dist/debugadapter.js:2:137821)
    at Socket.emit (node:events:526:28)
    at addChunk (node:internal/streams/readable:315:12)
    at readableAddChunk (node:internal/streams/readable:289:9)
    at Readable.push (node:internal/streams/readable:228:10)

d-adamson, 2023-02-01:
**SOLUTION: Only use a regular pico (non-wifi version) to build the picoprobe.**

There is joy today! I received a regular pico (non-wifi) and connected it as the picoprobe. Used the picoprobe.uf2, launch.json & settings.json as specified in the getting started doc. All is working now both at the command line as well as in VS Code.

Should update appendix A in the getting started document to specify the picoprobe is currently only compatible with a regular pico (non-wifi). Alternatively, update the software, config files, etc. to be compatible with the pico w.

aallan, 2023-02-03:
I don't believe there should be any difference between a Pico and Pico W when it comes to using Pico Probe, none of the pins used are pulled elsewhere on Pico W. They should work the same.

aallan, 2023-02-03:
That is not an issue, see @lurch's reply above.

> https://github.com/raspberrypi/usb-pid says that PID `0x0004` is "Raspberry Pi PicoProbe" and PID `0x000C` is "Raspberry Pi RP2040 CMSIS-DAP debug adapter", so I think your comments about Pico W are [...]

lurch, 2023-02-03:
To clarify (and I'm sure @P33M will correct me if I'm wrong) I believe the PID of `0x0004` was for the "old" version of the picoprobe UF2 using the original (custom?) picoprobe interface (hence the `-f interface/picoprobe.cfg` argument to openocd), but the PID of 0x000C` is for the "new" version of the picoprobe UF2 using the CMSIS-DAP protocol (hence the `-f interface/cmsis-dap.cfg` argument to openocd). All of this is **entirely** orthogonal to the differences between Pico and Pico W.

d-adamson, 2023-02-03:
Interesting. So the PID follows the firmware (*.uf2) and not the actual hardware? I missed that detail in the earlier post.

Any ideas what the solution may be? Has anyone successfully tested a pico w as the picoprobe?

d-adamson, 2023-02-03:
I'll do a test later today. I'll have a project that is running successfully with the picoprobe (regular pico) and then swap in a pico w. I won't change any other parameters, config files, etc., and then I'll post the output of both.

Maybe I started down a false path and got myself confused (it happens), but hopefully this test should help clarify things.

lurch, 2023-02-03:
> So the PID follows the firmware (*.uf2) and not the actual hardware?

Correct. There's no "settings" or "parameters" inside the RP2040, so when the chip first boots up it doesn't even *know* if it's on a Pico board, a Pico W board, or some other 3rd-party RP2040 board.

d-adamson, 2023-02-03:
**SOLUTION**

Well, I just ran the test and I confirm that picoprobe (from within vs code) **is indeed working with both a pico and a pico w.**

You guy are right, I went down a false path somewhere. I should have run this test earlier to confirm what I thought was the solution.

My apologies. I'll amend my comments.

[pico_output.txt](https://github.com/raspberrypi/openocd/files/10579915/pico_output.txt)
[picow_output.txt](https://github.com/raspberrypi/openocd/files/10579920/picow_output.txt)

closed 2023-02-03 by aallan