raspberrypi/openocd #77: Any chance to update to openocd 0.12?

rgrr, 2023-02-02:
openocd 0.11 has a bug which prevents CMSIS-DAP connection to probes with version > 2.0.0, i.e. CMSIS > 5.7.0 which is about 3 years old.

If one wants to use Arduino/PlatformIO with Earles enhancements (https://github.com/earlephilhower/arduino-pico) this openocd is not usable with newer probes.

P33M, 2023-02-22:
https://github.com/raspberrypi/openocd/tree/rp2040-v0.12.0

This will become the default branch in the near future.

aallan, 2023-02-22:
Should I be changing the

````
$ git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --depth=1 --no-single-branch
````

line in the [documentation](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#installing-openocd) to use the `rp2040-v0.12.0` branch?

P33M, 2023-02-22:
Not yet, as there are several (as yet untested) upstream commits on the development branch to do with GPIO SWD that need pulling in.

aallan, 2023-02-22:
> Not yet, as there are several (as yet untested) upstream commits on the development branch to do with GPIO SWD that need pulling in.

Got it. Can you let me know when I should flip that over?

ilg-ul, 2023-06-29:
Any progress on this?

An update would also solve https://github.com/raspberrypi/openocd/issues/82.

P33M, 2023-07-07:
Branch has now been switched to v0.12. There should be no functional difference as the downstream swd config file has been replicated in the new upstream environment.

closed 2023-07-07 by P33M