raspberrypi/openocd #79: New install of OpenOCD and Msys2 - bootstrap and configure errors

nyh-workshop, 2023-03-25:
Hello there,

I have an issue with a fresh install of Msys2 and PicoProbe OpenOCD compilation.  Basically, it could not even get past the bootstrap and configure - these complaining about AC_PROG_CC_C99 or the m4_warn.

Here is the error when I do the bootstrap:
```
$ ./bootstrap
+ aclocal --warnings=all
+ libtoolize --automake --copy
+ autoconf --warnings=all
configure.ac:33: warning: ac_ext=c
configure.ac:33: ac_cpp='$CPP $CPPFLAGS'
configure.ac:33: ac_compile='$CC -c $CFLAGS $CPPFLAGS conftest.$ac_ext >&5'
configure.ac:33: ac_link='$CC -o conftest$ac_exeext $CFLAGS $CPPFLAGS $LDFLAGS conftest.$ac_ext $LIBS >&5'
configure.ac:33: ac_compiler_gnu=$ac_cv_c_compiler_gnu
configure.ac:33: if test -n "$ac_tool_prefix"; then
configure.ac:33:   # Extract the first word of "${ac_tool_prefix}gcc", so it can be a program name with args.
configure.ac:33: set dummy ${ac_tool_prefix}gcc; ac_word=$2
configure.ac:33: { printf "%s\n" "$as_me:${as_lineno-$LINENO}: checking for $ac_word" >&5
configure.ac:33: printf %s "checking for $ac_word... " >&6; }
configure.ac:33: if test ${ac_cv_prog_CC+y}
configure.ac:33: then :
configure.ac:33:   printf %s "(cached) " >&6
configure.ac:33: else $as_nop
configure.ac:33:   if test -n "$CC"; then
configure.ac:33:   ac_cv_prog_CC="$CC" # Let the user override the test.
configure.ac:33: else
configure.ac:33: as_save_IFS=$IFS; IFS=$PATH_SEPARATOR
configure.ac:33: for as_dir in $PATH
configure.ac:33: do
configure.ac:33:   IFS=$as_save_IFS
configure.ac:33:   case $as_dir in #(((
configure.ac:33:     '' is m4_require'd but not m4_defun'd
../autoconf-2.71/lib/m4sugar/m4sh.m4:692: _AS_IF_ELSE is expanded from...
../autoconf-2.71/lib/m4sugar/m4sh.m4:699: AS_IF is expanded from...
../autoconf-2.71/lib/autoconf/general.m4:2249: AC_CACHE_VAL is expanded from...
../autoconf-2.71/lib/autoconf/programs.m4:41: _AC_CHECK_PROG is expanded from...
../autoconf-2.71/lib/autoconf/programs.m4:101: AC_CHECK_PROG is expanded from...
../autoconf-2.71/lib/autoconf/programs.m4:221: AC_CHECK_TOOL is expanded from...
../autoconf-2.71/lib/autoconf/c.m4:458: AC_PROG_CC is expanded from...
configure.ac:33: the top level
configure.ac:6: error: possibly undefined macro: dnl
      If this token and others are legitimate, please use m4_pattern_allow.
      See the Autoconf documentation.
configure.ac:26: error: possibly undefined macro: AC_PROG_CC
configure.ac:28: error: possibly undefined macro: m4_warn
configure.ac:29: error: possibly undefined macro: AC_PROG_CC_C99
configure.ac:31: error: possibly undefined macro: AC_REQUIRE
```

config:
```
username@DESKTOP-XXXXX MSYS /h/picoprobe/openocd
$ ./configure --disable-werror
configure: loading site script /etc/config.site
checking for makeinfo... makeinfo
checking for a BSD-compatible install... /usr/bin/install -c
checking whether build environment is sane... yes
checking for a race-free mkdir -p... /usr/bin/mkdir -p
checking for gawk... gawk
checking whether make sets $(MAKE)... yes
checking whether make supports nested variables... yes
checking for gcc... gcc
checking whether the C compiler works... yes
checking for C compiler default output file name... a.exe
checking for suffix of executables... .exe
checking whether we are cross compiling... no
checking for suffix of object files... o
checking whether the compiler supports GNU C... yes
checking whether gcc accepts -g... yes
checking for gcc option to enable C11 features... none needed
checking whether gcc understands -c and -o together... yes
checking whether make supports the include directive... yes (GNU style)
checking dependency style of gcc... gcc3
./configure: line 4903: syntax error near unexpected token `[obsolete],'
./configure: line 4903: `m4_warn([obsolete],'
```

I'm not sure what's the m4_warn thing there - I'm suspecting it has something to do with the updated tools that are not compatible with the OpenOCD.

Here's the list of what I did earlier:
- PicoProbe Git Commit: 8e3c38f78730ce878ff81448bc3f6c1e6bb06e13 
- Msys2 version: [msys2-x86_64-20230318.exe](https://github.com/msys2/msys2-installer/releases/download/2023-03-18/msys2-x86_64-20230318.exe) 
- Following the install instructions from: https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html

P33M, 2023-07-07:
We no longer recommend installing openocd from source on windows. We supply precompiled versions as part of our Pico on Windows installer. https://github.com/raspberrypi/pico-setup-windows

closed 2023-07-07 by P33M

lurch, 2023-07-13:
In https://github.com/raspberrypi/documentation/pull/3010 we've updated the branch of openocd that you ought to be building from, if you don't want to use the precompiled version from the Pico Windows Installer.