raspberrypi/openocd #80: Picoprobe with not wokring openOCD mac osx

tonycarbon, 2023-03-27:
Hi 

Following the gettings started guide Apendix A for Picoprobe
I cannot get Openocd to recognise the connected picoprobe.

When following the install instructions everything works fine except
**brew install libtool automake libusb wget pkg-config gcc texinfo** 

Where I get the following errors

```
==> Installing dependencies for wget: openssl@3
==> Installing wget dependency: openssl@3
==> perl ./Configure --prefix=/usr/local/Cellar/openssl@3/3.1.0 --openssldir=/us
==> make
Last 15 lines from /Users/jake/Library/Logs/Homebrew/openssl@3/02.make:
           ^~~~~~
crypto/bn/rsaz-3k-avx512.s:1285:12: error: register %ymm22 is only available with AVX512
 vmovdqu64 %ymm22,384(%rdi)
           ^~~~~~
crypto/bn/rsaz-3k-avx512.s:1286:12: error: register %ymm23 is only available with AVX512
 vmovdqu64 %ymm23,416(%rdi)
           ^~~~~~
crypto/bn/rsaz-3k-avx512.s:1287:12: error: register %ymm24 is only available with AVX512
 vmovdqu64 %ymm24,448(%rdi)
           ^~~~~~
crypto/bn/rsaz-3k-avx512.s:1288:12: error: register %ymm25 is only available with AVX512
 vmovdqu64 %ymm25,480(%rdi)
           ^~~~~~
make[1]: *** [crypto/bn/libcrypto-shlib-rsaz-3k-avx512.o] Error 1
make: *** [build_sw] Error 2
``` 

When starting Openocd I get this error

```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.11.0-g4f2ae61-dirty (2022-11-13-14:20)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : only one transport option; autoselect 'swd'
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Error: Failed to open or find the device
Error: Can't find a picoprobe device! Please check device connections and `permissions.
``` 

any help would be greatly appreciated

lobotony, 2023-05-18:
@tonycarbon  Try this: 
* use latest picoprobe build https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.02
* in your local openocd clone, in file tcl/interface/picoprobe.cfg, in line 2, change "picoprobe" to "cmsis-dap"
* if you installed openocd: in your local openocd clone, do a "sudo make install" just to make sure the config files are written to the appropriate locations
Now, when calling ```openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl```, openocd seems to connect:
```
openocd -f interface/picoprobe.cfg -f target/rp2040.cfg -s tcl
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-05-19-00:15)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
adapter speed: 5000 kHz

Info : auto-selecting first available session transport "swd". To override use 'transport select <transport>'.
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : Using CMSIS-DAPv2 interface with VID:PID=0x2e8a:0x000c, serial=E66118C4E3497327
Info : CMSIS-DAP: SWD  Supported
Info : CMSIS-DAP: FW Version = 2.0.0
Info : CMSIS-DAP: Interface Initialised (SWD)
Info : SWCLK/TCK = 0 SWDIO/TMS = 0 TDI = 0 TDO = 0 nTRST = 0 nRESET = 0
Info : CMSIS-DAP: Interface ready
Info : clock speed 5000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```
This is as far as I've come after a few hours of permutating. It's the first time I'm trying to setup the toolchain, and I haven't yet deployed and debugged a binary. 
The config file change is from a rejected pull request: https://github.com/raspberrypi/openocd/pull/72
I'm on an M1 Mac Mini, Mac OS Ventura 13.3.1
Hope this helps.

P33M, 2023-07-07:
>When starting Openocd I get this error
... `Error: Can't find a picoprobe device! Please check device connections and `permissions.`
The instructions no longer tell you to use `interface/picoprobe.cfg`.

closed 2023-07-07 by P33M