raspberrypi/openocd #83: Packaging raspberrypi/openocd on nixpkgs

LU15W1R7H, 2023-07-29:
I'm trying to package raspberrypi/openocd on nixpkgs, but I'm running into some problems.

This is the related PR NixOS/nixpkgs#179713

First off all I'm not quite sure which branch I should use. So far I've been trying to base the package off the most recent commit
of branch the `rp2040`. But in the meantime you've chosen `rp2040-v0.12.0` as the default branch, so should I use this?

It seems something strange is happening (maybe you keep force pushing) but the commit seems to be changing every once in a while. The sha256 hash (not the revision hash, but a hash of the contents of the commit) keeps changing. This doesn't allow me to package this repository.

Thanks for any help!

lurch, 2023-08-03:
I've never used Nix myself, but just out of curiosity I did some google searching...

> But in the meantime you've chosen `rp2040-v0.12.0` as the default branch, so should I use this?

Yes, `rp2040-v0.12.0` is now the branch we're building from, see e.g. https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#linux-and-raspberry-pi

> It seems something strange is happening (maybe you keep force pushing) but the commit seems to be changing every once in a while. The sha256 hash (not the revision hash, but a hash of the contents of the commit) keeps changing.

Looks like you're running into https://github.com/NixOS/nixpkgs/issues/8567 which is a problem with nixpkgs, rather than anything to do with https://github.com/raspberrypi/openocd ?

LU15W1R7H, 2023-08-07:
@lurch I believe you are right. Thanks for your help!!

closed 2023-08-07 by LU15W1R7H