raspberrypi/openocd #85: Make errors out on -Werror

LBrinston, 2023-08-08:
I am trying to build openocd on Pop-OS (22.04 LTS) to use with a pi pico. 
Initially I was trying to build the mainline openocd (apt is stuck at 0.11.0) since 0.12.0 now supports the RP2040 but encountered different errors and ended up here.

## Process:
```bash
git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1 --no-single-branch
cd openocd/
./bootstrap
./configure --enable-picoprobe --enable-cmsis-dap
make -j2
```

I enabled both pico-probe and cmsis-dap as I have vague recollections of not being able to get picoprobe to work but cmsis-dap would when I did this three months ago. 

## Expected result
openocd compiles.

## Actual result
make crashes into a `-Werror`  over some indentation in `src/flash/nor/numicro.c:1244:17`
If I comment out `-Werror` in the Makefile it does compile but that's probably not ideal?

make's output:
```
parablennius@pop-os:~/Downloads/openocd$ make
make  all-recursive
make[1]: Entering directory '/home/parablennius/Downloads/openocd'
Making all in jimtcl
make[2]: Entering directory '/home/parablennius/Downloads/openocd/jimtcl'
make[2]: Nothing to be done for 'all'.
make[2]: Leaving directory '/home/parablennius/Downloads/openocd/jimtcl'
Making all in src/jtag/drivers/libjaylink
make[2]: Entering directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
make  all-recursive
make[3]: Entering directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
Making all in libjaylink
make[4]: Entering directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink/libjaylink'
make[4]: Nothing to be done for 'all'.
make[4]: Leaving directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink/libjaylink'
make[4]: Entering directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
make[4]: Leaving directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
make[3]: Leaving directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
make[2]: Leaving directory '/home/parablennius/Downloads/openocd/src/jtag/drivers/libjaylink'
make[2]: Entering directory '/home/parablennius/Downloads/openocd'
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/xsvf/libxsvf.la  src/xsvf/xsvf.lo  
libtool: link: ar cr src/xsvf/.libs/libxsvf.a  src/xsvf/xsvf.o
libtool: link: ranlib src/xsvf/.libs/libxsvf.a
libtool: link: ( cd "src/xsvf/.libs" && rm -f "libxsvf.la" && ln -s "../libxsvf.la" "libxsvf.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/svf/libsvf.la  src/svf/svf.lo  
libtool: link: ar cr src/svf/.libs/libsvf.a  src/svf/svf.o
libtool: link: ranlib src/svf/.libs/libsvf.a
libtool: link: ( cd "src/svf/.libs" && rm -f "libsvf.la" && ln -s "../libsvf.la" "libsvf.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/pld/libpld.la  src/pld/pld.lo src/pld/xilinx_bit.lo src/pld/virtex2.lo  
libtool: link: ar cr src/pld/.libs/libpld.a  src/pld/pld.o src/pld/xilinx_bit.o src/pld/virtex2.o
libtool: link: ranlib src/pld/.libs/libpld.a
libtool: link: ( cd "src/pld/.libs" && rm -f "libpld.la" && ln -s "../libpld.la" "libpld.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/jtag/hla/libocdhla.la  src/jtag/hla/hla_transport.lo src/jtag/hla/hla_tcl.lo src/jtag/hla/hla_interface.lo src/jtag/hla/hla_layout.lo  
libtool: link: ar cr src/jtag/hla/.libs/libocdhla.a  src/jtag/hla/hla_transport.o src/jtag/hla/hla_tcl.o src/jtag/hla/hla_interface.o src/jtag/hla/hla_layout.o
libtool: link: ranlib src/jtag/hla/.libs/libocdhla.a
libtool: link: ( cd "src/jtag/hla/.libs" && rm -f "libocdhla.la" && ln -s "../libocdhla.la" "libocdhla.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/jtag/aice/libocdaice.la  src/jtag/aice/libocdaice_la-aice_transport.lo src/jtag/aice/libocdaice_la-aice_interface.lo src/jtag/aice/libocdaice_la-aice_port.lo src/jtag/aice/libocdaice_la-aice_usb.lo src/jtag/aice/libocdaice_la-aice_pipe.lo  
libtool: link: ar cr src/jtag/aice/.libs/libocdaice.a  src/jtag/aice/libocdaice_la-aice_transport.o src/jtag/aice/libocdaice_la-aice_interface.o src/jtag/aice/libocdaice_la-aice_port.o src/jtag/aice/libocdaice_la-aice_usb.o src/jtag/aice/libocdaice_la-aice_pipe.o
libtool: link: ranlib src/jtag/aice/.libs/libocdaice.a
libtool: link: ( cd "src/jtag/aice/.libs" && rm -f "libocdaice.la" && ln -s "../libocdaice.la" "libocdaice.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/jtag/drivers/usb_blaster/libocdusbblaster.la  src/jtag/drivers/usb_blaster/libocdusbblaster_la-usb_blaster.lo src/jtag/drivers/usb_blaster/libocdusbblaster_la-ublast_access_ftdi.lo src/jtag/drivers/usb_blaster/libocdusbblaster_la-ublast2_access_libusb.lo  
libtool: link: ar cr src/jtag/drivers/usb_blaster/.libs/libocdusbblaster.a  src/jtag/drivers/usb_blaster/libocdusbblaster_la-usb_blaster.o src/jtag/drivers/usb_blaster/libocdusbblaster_la-ublast_access_ftdi.o src/jtag/drivers/usb_blaster/libocdusbblaster_la-ublast2_access_libusb.o
libtool: link: ranlib src/jtag/drivers/usb_blaster/.libs/libocdusbblaster.a
libtool: link: ( cd "src/jtag/drivers/usb_blaster/.libs" && rm -f "libocdusbblaster.la" && ln -s "../libocdusbblaster.la" "libocdusbblaster.la" )
/bin/bash ./libtool  --tag=CC   --mode=link gcc -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Werror -g -O2   -o src/jtag/drivers/libocdjtagdrivers.la  src/jtag/drivers/libocdjtagdrivers_la-driver.lo src/jtag/drivers/libocdjtagdrivers_la-jtag_usb_common.lo src/jtag/drivers/libocdjtagdrivers_la-libusb_helper.lo src/jtag/drivers/libocdjtagdrivers_la-usb_common.lo src/jtag/drivers/libocdjtagdrivers_la-jlink.lo    src/jtag/drivers/libocdjtagdrivers_la-ftdi.lo src/jtag/drivers/libocdjtagdrivers_la-mpsse.lo    src/jtag/drivers/libocdjtagdrivers_la-ft232r.lo     src/jtag/drivers/libocdjtagdrivers_la-bitq.lo src/jtag/drivers/libocdjtagdrivers_la-presto.lo src/jtag/drivers/libocdjtagdrivers_la-usbprog.lo src/jtag/drivers/libocdjtagdrivers_la-rlink.lo src/jtag/drivers/libocdjtagdrivers_la-rlink_speed_table.lo src/jtag/drivers/libocdjtagdrivers_la-ulink.lo src/jtag/drivers/versaloon/usbtoxxx/libocdjtagdrivers_la-usbtogpio.lo src/jtag/drivers/versaloon/usbtoxxx/libocdjtagdrivers_la-usbtojtagraw.lo src/jtag/drivers/versaloon/usbtoxxx/libocdjtagdrivers_la-usbtoswd.lo src/jtag/drivers/versaloon/usbtoxxx/libocdjtagdrivers_la-usbtopwr.lo src/jtag/drivers/versaloon/usbtoxxx/libocdjtagdrivers_la-usbtoxxx.lo src/jtag/drivers/versaloon/libocdjtagdrivers_la-versaloon.lo src/jtag/drivers/libocdjtagdrivers_la-vsllink.lo src/jtag/drivers/libocdjtagdrivers_la-arm-jtag-ew.lo   src/jtag/drivers/libocdjtagdrivers_la-stlink_usb.lo src/jtag/drivers/libocdjtagdrivers_la-ti_icdi_usb.lo src/jtag/drivers/libocdjtagdrivers_la-nulink_usb.lo  src/jtag/drivers/libocdjtagdrivers_la-osbdm.lo src/jtag/drivers/libocdjtagdrivers_la-opendous.lo    src/jtag/drivers/libocdjtagdrivers_la-openjtag.lo src/jtag/drivers/libocdjtagdrivers_la-cmsis_dap_usb_hid.lo src/jtag/drivers/libocdjtagdrivers_la-cmsis_dap.lo src/jtag/drivers/libocdjtagdrivers_la-cmsis_dap_usb_bulk.lo   src/jtag/drivers/libocdjtagdrivers_la-kitprog.lo src/jtag/drivers/libocdjtagdrivers_la-picoprobe.lo src/jtag/drivers/libocdjtagdrivers_la-xds110.lo  -lusb-1.0 -lusb -lftdi  -lhidapi-hidraw  src/jtag/drivers/libjaylink/libjaylink/libjaylink.la src/jtag/drivers/usb_blaster/libocdusbblaster.la -lm 
[... cut ...]
src/flash/nor/numicro.c:1604:41: note: in expansion of macro ‘LOG_DEBUG’
 1604 |                                         LOG_DEBUG("status: 0x%" PRIx32 "", status);
      |                                         ^~~~~~~~~
cc1: all warnings being treated as errors
make[2]: *** [Makefile:3629: src/flash/nor/numicro.lo] Error 1
make[2]: Leaving directory '/home/parablennius/Downloads/openocd'
make[1]: *** [Makefile:5318: all-recursive] Error 1
make[1]: Leaving directory '/home/parablennius/Downloads/openocd'
```

lurch, 2023-08-08:
I don't know if it'll fix your particular problem, but FYI the default branch for this repo is now `rp2040-v0.12.0`.

LBrinston, 2023-08-08:
> I don't know if it'll fix your particular problem, but FYI the default branch for this repo is now `rp2040-v0.12.0`.

I'm going to have to double check what version I compiled. I thought I cloned the most recent branch but I think `checkinstall` identified it as 0.11.0.

lurch, 2023-08-08:
Your original message above says ` --branch picoprobe`, which is why I pointed out that's not the latest. :slightly_smiling_face:

josuah, 2023-08-10:
Note that this fork is no longer required for using the latest picoprobe firmware, which uses the CMSIS-DAP firmware, which is supported in upstream OpenOCD.

Related:
https://github.com/raspberrypi/openocd/issues/39#issuecomment-1625438550

You might still need a recent [OpenOCD](https://github.com/openocd-org/openocd/) depending on what is packaged on your distribution.
You might also need to make sure you use the [latest release of the picoprobe firmware](https://github.com/raspberrypi/picoprobe/releases).

And then you might forget about this fork as all the work went upstream. Yay! :)

lurch, 2023-08-11:
> And then you might forget about this fork as all the work went upstream.

Looking at https://github.com/raspberrypi/openocd/commits/rp2040-v0.12.0 there's still a couple of downstream commits on top of the upstream code.

josuah, 2023-08-11:
Thank you! I forgot about this part...
I understand better why I saw more commits going.

LBrinston, 2023-08-12:
> Your original message above says ` --branch picoprobe`, which is why I pointed out that's not the latest. 🙂

🤦 yup, sure does. My bad, thank you for catching that.

closed 2023-08-12 by LBrinston

Caraffa-git, 2023-09-19:
I encountered the same problem and it is worth noting that in "getting started with pico" the latest branch is still not included (`$ git clone https://github.com/raspberrypi/openocd.git --branch rp2040 --depth=1` page no. 60)