raspberrypi/openocd #88: Cannot build on M1 macOS Sonoma

adigandhi, 2023-10-10:
I followed steps from Getting Started with Pi Pico, and[https://github.com/raspberrypi/openocd/issues/7]( https://github.com/raspberrypi/openocd/issues/7). Also [https://hackaday.io/page/13821-building-openocd-for-picoprobe-on-m1-mac](https://hackaday.io/page/13821-building-openocd-for-picoprobe-on-m1-mac)

MacOS Sonoma M1 Max (Mac Studio)

```brew install open-ocd --only-dependencies
brew install automake
brew install autoconf pkg-config libtool texinfo wget gcc

cd ~/projects/pico
git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1 --no-single-branch
cd openocd

export PATH="$(brew --prefix)/opt/texinfo/bin:$PATH"
./bootstrap

CAPSTONE_CFLAGS="-I$(brew --prefix)/include" \
  ./configure --prefix="$(brew --prefix)"  \
  --enable-picoprobe --disable-presto --disable-openjtag
  
```

I see this error:

```
make -j4
cat src/helper/startup.tcl src/jtag/startup.tcl src/target/startup.tcl src/server/startup.tcl src/flash/startup.tcl | ./src/helper/bin2char.sh > src/startup_tcl.inc || { rm -f src/startup_tcl.inc; false; }
cp src/jtag/drivers/minidriver_imp.h src/jtag/minidriver_imp.h
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-recursive
Making all in jimtcl
        MKLDEXT _load-static-exts.c
        CC      jim-subcmd.o
        CC      jim-interactive.o
        CC      jim-format.o
        CC      jim.o
        CC      jimregexp.o
        CC      utf8.o
        CC      jimiocompat.o
        CC      linenoise.o
        CC      jim-tty.o
        CC      jim-aio.o
        CC      jim-array.o
        CC      jim-clock.o
        CC      jim-eventloop.o
        CC      jim-exec.o
jim-exec.c:1046:15: warning: 'vfork' is deprecated: Use posix_spawn or fork [-Wdeprecated-declarations]
        pid = vfork();
              ^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/unistd.h:604:1: note: 'vfork' has been explicitly marked deprecated here
__deprecated_msg("Use posix_spawn or fork")
^
/Applications/Xcode.app/Contents/Developer/Platforms/MacOSX.platform/Developer/SDKs/MacOSX.sdk/usr/include/sys/cdefs.h:215:48: note: expanded from macro '__deprecated_msg'
        #define __deprecated_msg(_msg) __attribute__((__deprecated__(_msg)))
                                                      ^
jim-exec.c:729:12: warning: variable 'child_environ' set but not used [-Wunused-but-set-variable]
    char **child_environ;
           ^
2 warnings generated.
        CC      jim-file.o
        CC      jim-history.o
        CC      jim-interp.o
        CC      jim-load.o
        CC      jim-namespace.o
        CC      jim-pack.o
        CC      jim-package.o
        CC      jim-posix.o
        CC      jim-readdir.o
        CC      jim-regexp.o
        CC      jim-signal.o
        CC      jim-syslog.o
        TCLEXT  _glob.c
        TCLEXT  _nshelper.c
        TCLEXT  _oo.c
        TCLEXT  _stdlib.c
        CC      nshelper.o
        CC      oo.o
        CC      stdlib.o
        CC      glob.o
        TCLEXT  _tclcompat.c
        TCLEXT  _tree.c
        CC      jimsh.o
        TCLEXT  _initjimsh.c
        CC      _load-static-exts.o
        CC      tree.o
        CC      initjimsh.o
        CC      tclcompat.o
        AR      libjim.a
        LINK    jimsh
ld: archive member '/' not a mach-o file in '/Users/adityagandhi/Workspace/pico-probe/openocd/openocd/jimtcl/libjim.a'
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [jimsh] Error 1
make[1]: *** [all-recursive] Error 1
make: *** [all] Error 2
```

aallan, 2023-10-10:
I'm not sure anyone here is running Sonoma yet. I'll see if I can get something upgraded and take a look.

rybalkoss, 2023-11-07:
I have exactly same error when building another small project after update to sonoma. Maybe it's something in macos

rybalkoss, 2023-11-11:
reproduced issue on minimal build example, https://stackoverflow.com/q/77444892/11335766

simonjwright, 2024-08-23:
I don't see this issue on Sonoma 14.6.1, Command Line Tools (i.e. Xcode) 15.3 (building with CC=clang). Unfortunately this brings another issue, CLT15.3 doesn't include m4. I tried `export M4=gm4` but Apple's gm4 doesn't accept `--gnu`, which Homebrew's automake uses. Fortunately, HB's m4 (`M4=/opt/homebrew/opt/m4/bin/m4`) works fine.

simonjwright, 2024-08-23:
I have a further issue, I guess from upstream, with `CC=clang`. Doesn't happen with GCC 14.1.0 (aarch64), though I guess it should!

```
libtool: compile:  clang -DHAVE_CONFIG_H -I. -I./src -I./src -DPKGDATADIR=\"/Users/simon/local/share/openocd\" -DBINDIR=\"/Users/simon/local/bin\" -I./jimtcl -I./jimtcl -Wall -Wstrict-prototypes -Wformat-security -Wshadow -Wextra -Wno-unused-parameter -Wbad-function-cast -Wcast-align -Wredundant-decls -Wpointer-arith -Wundef -Werror -g -O2 -MT src/target/riscv/batch.lo -MD -MP -MF src/target/riscv/.deps/batch.Tpo -c src/target/riscv/batch.c -o src/target/riscv/batch.o
src/target/riscv/batch.c:194:24: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Werror,-Wcast-align]
                        uint32_t *data_in = (uint32_t *)batch->data_in;
                                            ^~~~~~~~~~~~~~~~~~~~~~~~~~
src/target/riscv/batch.c:243:23: error: cast from 'uint8_t *' (aka 'unsigned char *') to 'uint32_t *' (aka 'unsigned int *') increases required alignment from 1 to 4 [-Werror,-Wcast-align]
                uint32_t *data_in = (uint32_t *)batch->data_in;
                                    ^~~~~~~~~~~~~~~~~~~~~~~~~~
2 errors generated.
make[2]: *** [src/target/riscv/batch.lo] Error 1
```

lurch, 2024-08-25:
> I have a further issue...

@simonjwright See #107