raspberrypi/openocd #93: Not able to run Openocd for rp2040 on rpi5

pranjalchanda08, 2023-12-23:
I am stuck with the following screen state on rpi5:
``` sh
sudo openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.11.0-g8e3c38f (2023-12-23-19:52)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
adapter speed: 1000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 1001 kHz

**** Stuck after this ****
```

lurch, 2023-12-23:
I've not tried it myself yet, but does https://github.com/raspberrypi/pico-setup/pull/28 help?

pranjalchanda08, 2023-12-23:
I will have to try and get back to you

pranjalchanda08, 2023-12-23:
Now I have got this issue:
```sh
sudo openocd -f interface/raspberrypi-swd.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.12.0-g4d87f6d (2023-12-23-23:46)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
WARNING: Unknown type of the host SoC. Expect JTAG/SWD clock slower than requested.
Warn : TMS/SWDIO moved to GPIO 8 (pin 24). Check the wiring please!
adapter speed: 5000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : BCM2835 GPIO JTAG/SWD bitbang driver
Info : clock speed 5000 kHz
Error: Failed to connect multidrop rp2040.dap0

```
Keeping the following warning in mind: **Warn : TMS/SWDIO moved to GPIO 8 (pin 24). Check the wiring please!**

**SWCLK -> GPIO25
GND -> Pin20
SWDIO -> GPIO8**

pranjalchanda08, 2023-12-28:
@lurch any updates on this. As I need this info to create some CI/CD Setups

lurch, 2023-12-28:
I'm afraid you'll have to wait until all the RPi engineers are back from holiday in the New Year.

pranjalchanda08, 2023-12-28:
Oh yeah Happy new year!

cleverca22, 2024-01-05:
i got it working fairly easily on my pi5:
```
clever@raspberrypi:~ $ cat test.cfg
adapter driver linuxgpiod
adapter gpio swclk 20 -chip 4
adapter gpio swdio 21 -chip 4
clever@raspberrypi:~ $ openocd -f test.cfg -f target/rp2040.cfg
Open On-Chip Debugger 0.12.0-g4d87f6dca (2024-01-05-00:08)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Warn : An adapter speed is not selected in the init scripts. OpenOCD will try to run the adapter at the low speed (100 kHz)
Warn : To remove this warnings and achieve reasonable communication speed with the target, set "adapter speed" or "jtag_rclk" in the init scripts.
Info : Linux GPIOD JTAG/SWD bitbang driver
Info : This adapter doesn't support configurable speed
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
```

all it needs now, is to drop that into the interface folder and name it well, and maybe change the pins to match the other examples

pranjalchanda08, 2024-01-05:
Please share the test.cfg file

> i got it working fairly easily on my pi5:
>
> ```
> clever@raspberrypi:~ $ cat test.cfg
> adapter driver linuxgpiod
> adapter gpio swclk 20 -chip 4
> adapter gpio swdio 21 -chip 4
> clever@raspberrypi:~ $ [...]

pranjalchanda08, 2024-01-05:
Ok I get now. My config was using the old rpi bitbanging. Need to switch to Linuxgpiod adapter

pranjalchanda08, 2024-01-05:
I think this info needs a released documentation somewhere from rpi team.

pranjalchanda08, 2024-01-05:
Few questions:
1. Is 100KHz limit of linuxgpiod.
2. How do we determine on rpi5 that the script is being run on rp5 or older version. This info is needed to maintain cross platform debug scripts.
3. And what does -chip 4 means here?

cleverca22, 2024-01-05:
> And what does -chip 4 means here?

```
clever@raspberrypi:~ $ gpiodetect
gpiochip0 [gpio-brcmstb@107d508500] (32 lines)
gpiochip1 [gpio-brcmstb@107d508520] (4 lines)
gpiochip2 [gpio-brcmstb@107d517c00] (17 lines)
gpiochip3 [gpio-brcmstb@107d517c20] (6 lines)
gpiochip4 [pinctrl-rp1] (54 lines)
```
the pi5 has 5 gpio controllers
the 40 pin header is on the 4th one

the debug port on the pi5 is on another, and a simple 3 pin JST-SH cable could link a pi5 directly to the pico with no fuss

> How do we determine on rpi5 that the script
`tcl/interface/raspberrypi-native.cfg` has code to read `/proc/device-tree/compatible` and check for strings like bcm2712

i'm not sure where the speed would be set

CamDavidsonPilon, 2024-01-06:
FYI this works on older Pis:
```
adapter driver linuxgpiod
adapter gpio swclk 20 -chip 0
adapter gpio swdio 21 -chip 0
```

cleverca22, 2024-01-06:
ah good catch, so we could just use gpiod on every model
and then all you need to do is change the chip#

pranjalchanda08, 2024-01-06:
> ```
> clever@raspberrypi:~ $ gpiodetect
> gpiochip0 [gpio-brcmstb@107d508500] (32 lines)
> gpiochip1 [gpio-brcmstb@107d508520] (4 lines)
> gpiochip2 [gpio-brcmstb@107d517c00] (17 lines)
> gpiochip3 [...]

I guess "adapter_speed xxxx" sets of the adapter speed in KHz. Currently as nothing is given it is setting to 100KHz by default

cleverca22, 2024-01-07:
> But 100KHz and that too it says "Info : This adapter doesn't support configurable" is kinda bothering me. Since 100KHz is very low specially when it comes to over network remote debugging.

```
static int adapter_set_speed(int speed)
{
        /* this command can be called during CONFIG,
         * in which case adapter isn't initialized */
        return is_adapter_initialized() ? adapter_driver->speed(speed) : ERROR_OK;
}
```
at the end, its just calling a speed function within a function pointer table
```
struct adapter_driver linuxgpiod_adapter_driver = {
        .name = "linuxgpiod",
        .transports = linuxgpiod_transport,

        .init = linuxgpiod_init,
        .quit = linuxgpiod_quit,
        .reset = linuxgpiod_reset,
 
        .jtag_ops = &linuxgpiod_interface,
        .swd_ops = &bitbang_swd,
};
```
and the linuxgpiod driver lacks a speed function
```
struct adapter_driver bcm2835gpio_adapter_driver = {
        .name = "bcm2835gpio",
        .transports = bcm2835_transports,
        .commands = bcm2835gpio_command_handlers,

        .init = bcm2835gpio_init,
        .quit = bcm2835gpio_quit,
        .reset = bcm2835gpio_reset,
        .speed = bcm2835gpio_speed,
        .khz = bcm2835gpio_khz,
        .speed_div = bcm2835gpio_speed_div,
 
        .jtag_ops = &bcm2835gpio_interface,
        .swd_ops = &bitbang_swd,
};
```
while the bcm2835 one does have one
so somebody will need to port that over

g3grau, 2024-04-07:
Took a while to figure it out... Had to read the configure script to realize that there is no enable switch for linuxgpiod but it's looking for an environment variable. So here's what I did to run it on a pi5 with bookworm. Hope it helps someone:

1. install "linuxgpiod" (why isn't it called libgpiod in the cfg?)

```
sudo apt install python3-libgpiod libgpiod-dev
git clone https://github.com/openocd-org/openocd.git
cd openocd
./bootstrap
export enable_linuxgpiod=yes
./configure
make -j 4
sudo make install

```
2. make a config file for raspberry pi 5. I used emacs, kate doesn't want to run as root?

```
cd /usr/local/share/openocd/scripts/interface
sudo -i
emacs raspberrypi5-swd.cfg&

```
3. edit the file. I have no idea why the example code use different pins in the manual ... there is no secret pin mapping, the sample file content is simply using other pins as described! This one matches the sketch in the manual:

```
# match the pins used in the Pi Pico C/C++ manual
# GPIO25=swclk, GPIO24=swdio
# NOTE: trying to use GPIO8 for swdio failed
adapter driver linuxgpiod
adapter gpio swclk 25 -chip 4
adapter gpio swdio 24 -chip 4

```
Note: when you tried the existing raspberrypi-swd.cfg before, you may have noticed a message regarding a pinout change from GPIO24 to GPIO8. However, my attempt to use this pin in the above file failed with "error requesting gpio line swdio"

4. Connect the rp2040 Pico with pins 18 (GPIO24=swdio),20 (gnd),22 (GPIO25=swclk) as shown in the C/C++ SDK manual (and provide 5V e.g. from the USB port).
  
5. Now openocd should work with raspi 5 gpio.
Go to the build folder and install the .elf file:

```
cd pico/pico-examples/build
openocd -f interface/raspberrypi5-swd.cfg -f target/rp2040.cfg -c "program blink/blink.elf verify reset exit"

```
6. The expected output looks like this:

```
Open On-Chip Debugger 0.12.0+dev-01562-g47d983a77 (2024-04-07-22:04)
Licensed under GNU GPL v2
For bug reports, read
	http://openocd.org/doc/doxygen/bugs.html
Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : Linux GPIOD JTAG/SWD bitbang driver
Info : Note: The adapter "linuxgpiod" doesn't support configurable speed
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477, DLPIDR 0x10000001
Info : [rp2040.core0] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core0] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core0] Examination succeed
Info : [rp2040.core1] Cortex-M0+ r0p1 processor detected
Info : [rp2040.core1] target has 4 breakpoints, 2 watchpoints
Info : [rp2040.core1] Examination succeed
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
[rp2040.core0] halted due to breakpoint, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
[rp2040.core1] halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ea msp: 0x20041f00
** Programming Started **
Info : Found flash device 'win w25q16jv' (ID 0x001540ef)
Info : RP2040 B0 Flash Probe: 2097152 bytes @0x10000000, in 32 sectors

Info : Padding image section 1 at 0x100051fc with 4 bytes (bank write end alignment)
Warn : Adding extra erase range, 0x10005200 .. 0x1000ffff
** Programming Finished **
** Verify Started **
** Verified OK **
** Resetting Target **
shutdown command invoked

```

pranjalchanda08, 2024-04-08:
Hi @g3grau,

Thanks for the detailed explanation!

aronneGrinton, 2024-05-28:
> Took a while to figure it out... Had to read the configure script to realize that there is no enable switch for linuxgpiod but it's looking for an environment variable. So here's what I did to run [...]

This works perfectly for me

ragazenta, 2024-09-10:
CMIIW, newer kernel `6.6.47+rpt-rpi-2712` will need to use `-chip 0` again. See https://github.com/raspberrypi/linux/pull/6144.

P33M, 2024-09-10:
Upstream have added support for Pi 5 (raspberrypi5-gpiod.cfg) but picked up the legacy gpiochip 4. Downstream has diverged with the release of RP2350, but the gpiod changes should be orthogonal.

P33M, 2024-09-10:
Once https://review.openocd.org/c/openocd/+/8363 goes in, then I can look at rebasing downstream.