raspberrypi/openocd #94: Raspberry Pi Debug Probe documentation needs clarification

seamusdemora, 2024-01-23:
I'm trying to connect my *new* Debug Probe to an (apparently) older Pico.  IOW, my Pico **does not** have the **'JST'** connector at the SWD terminals; it has only 3 holes, and I've soldered a 3-pin header in that location for making connections. 

I am trying to follow the [online documentation](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html) for making the required connections. However, the documentation assumes the existence of the JST connector at SWD, and does not indicate how the female headers are to be connected to the SWD. The middle GND connection seems clear enough, but which wire goes to `SWCLK` and which one goes to `SWDIO` is unclear. I've been unable to find any markings that make these connections clear.  

Can someone clear this up for me?

P33M, 2024-01-23:
SWCLK/SWDIO are marked on silkscreen on the underside of the board.

lurch, 2024-01-23:
ping @aallan in case the documentation needs clarifying.

@seamusdemora See also Chapter 5 of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf (which unfortunately doesn't cover the Debug Probe yet). There's also pinout diagrams at https://www.raspberrypi.com/documentation/microcontrollers/raspberry-pi-pico.html

seamusdemora, 2024-01-23:
@P33M , @lurch 

Thanks, but the problem is actually in the documentation. Nowhere is it pointed out which pin from the debug probe goes to SWDIO, and which goes to SWCLK. I finally surmised from the box that the **C** in `TX/SC` was for SW**C**LK, but that seems a bit of a reach. And I guess there are still a lot of non-JST Picos out there!

lurch, 2024-01-23:
Yeah, perhaps this NOTE block needs a bit more clarification?
[image]

But do note that https://github.com/raspberrypi/documentation/pull/3454 changes the "SWDCLK" wording to "SWCLK", so hopefully that makes it a bit more obvious that `SC` from the Debug Probe connects to `SWCLK` on the Pico, and `SD` from the Debug Probe connects to `SWDIO` on the Pico? :shrug: 
See also https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#the-debug-probe

seamusdemora, 2024-01-23:
Yes, the NOTE block would be a good place to put some clarification in the "Getting Started Guide", and IMHO it needs a picture to support the point. It should unequivocally establish the point that the Wire labeled `SC` is to be connected to the board terminal `SWCLK`, and the wire labeled `SD` is to be connected to the board terminal `SWDIO`.  

With 100's of thousands of graphics like the one below still floating around on the Internet, it seems to me the users/customers deserve clarity. It will be years before the _non-JST-equipped_ boards disappear.  

Just my 2 pence worth...

[image]

lurch, 2024-01-23:
> It will be years before the _non-JST-equipped_ boards disappear.

Correct. The "non-JST boards" are Pico and Pico W, and the boards _with_ the JST debug headers are Pico H and Pico WH. All 4 boards serve different purposes, and as it says at https://www.raspberrypi.com/products/raspberry-pi-pico/ 
"Raspberry Pi Pico series will remain in production until at least January 2034".

seamusdemora, 2024-01-24:
@lurch 
Ah! Did not know that...  I thought the "older/original" Picos had been phased out in favor of the new. 
All the more reason to fix the docs, eh?  :)

lurch, 2024-01-25:
https://shop.pimoroni.com/collections/pico-w-aboard is an example of where the Picos-without-headers are still useful. See Chapter 3 of https://datasheets.raspberrypi.com/rp2040/hardware-design-with-rp2040.pdf for more information.

closed 2024-02-19 by aallan