raspberrypi/openocd #95: In rp2040.cfg getting error "Unknown param: -dp-id"

jt000, 2024-02-02:
I'm attempting to upload a program using the debug probe following the directions at [Uploading new programs to your Pico](https://www.raspberrypi.com/documentation/microcontrollers/debug-probe.html#uploading-new-programs-to-your-pico) and am encountering a couple of errors.

`sudo openocd -f interface/cmsis-dap.cfg -f target/rp2040.cfg -c "adapter speed 5000" -c "program blink.elf verify reset exit"`

**Problem 1:** `Error: Can't find target/rp2040.cfg`
I assume this is because the openocd/tcl folder isn't in my path, so I've resolved this by passing a relative path which seems to have averted this issue. Not sure if the docs need to be updated to reflect.

`sudo openocd -f interface/cmsis-dap.cfg -f ../openocd/tcl/target/rp2040.cfg -c "adapter speed 5000" -c "program blink.elf verify reset exit"`

**Problem 2:** `../openocd/tcl/target/rp2040.cfg:72: Error: Unknown param: -dp-id, try one of: -chain-position, or -ignore-syspwrupack
in procedure 'script'`
Line 72 does indeed contain a parameter to -dp-id, but this appears to be valid according to the [TAP Declarations 10.8](https://openocd.org/doc/html/TAP-Declaration.html). Not sure why I'm getting this error here or what to do about it...

`dap create $_CHIPNAME.dap0 -chain-position $_CHIPNAME.cpu -dp-id $_CPUTAPID -instance-id 0`

lurch, 2024-02-03:
Have you installed openocd following the instructions in https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf or are you using some older version?

jt000, 2024-02-03:
@lurch Yeah, I followed the instructions on the debug-probe page. I actually did both the `apt install` and locally cloned\built\installed, which I wonder if that might be causing the issue...

jt000, 2024-02-03:
Figured it out... Appears I need to read instructions better.

For anyone else who might've did the same thing I did:
`sudo apt install openocd` is **only** if you're running on Raspberry Pi OS. If you're running on Debian, then do not install this. It will not work even if you also follow the clone\build\install directions as well.

closed 2024-02-03 by jt000

lurch, 2024-02-04:
Usually, if you've installed a program through `apt` (which typically installs things into `/usr/bin`), and also compile and install the same program from source (which normally installs things into `/usr/local/bin`), then the version that you've installed from source takes precedence over the apt-installed version, because `/usr/local/bin` normally appears before `/usr/bin` in your `$PATH`. But I guess running things as `sudo` might interfere with that?

If you want to ensure that you're using the installed-from-source version rather than the apt-installed version, you could run
```
sudo /usr/local/bin/openocd ...
```
instead of
```
sudo openocd ...
```