raspberrypi/openocd #96: Unable to build openocd

maaakit, 2024-02-11:
I'm trying to build openocd on fresh repository clone. Bootstrap script fails as follows: 

```
marcin@marcin-VirtualBox:~/pico$ sudo apt install make libtool pkg-config autoconf automake texinfo 
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
make is already the newest version (4.3-4.1build1).
libtool is already the newest version (2.4.7-7).
pkg-config is already the newest version (1.8.1-2).
autoconf is already the newest version (2.71-3).
automake is already the newest version (1:1.16.5-1.3).
texinfo is already the newest version (7.0.3-2).
0 upgraded, 0 newly installed, 0 to remove and 0 not upgraded.
marcin@marcin-VirtualBox:~/pico$ git clone https://github.com/raspberrypi/openocd.git --branch picoprobe --depth=1 --no-single-branch
Cloning into 'openocd'...
remote: Enumerating objects: 6009, done.
remote: Counting objects: 100% (6009/6009), done.
remote: Compressing objects: 100% (2891/2891), done.
remote: Total 6009 (delta 3988), reused 4606 (delta 3049), pack-reused 0
Receiving objects: 100% (6009/6009), 5.41 MiB | 13.58 MiB/s, done.
Resolving deltas: 100% (3988/3988), done.
marcin@marcin-VirtualBox:~/pico$ cd openocd
marcin@marcin-VirtualBox:~/pico/openocd$ ./bootstrap 
bash: ./bootstrap: cannot execute: required file not found
```

lurch, 2024-02-11:
Your usage of `--branch picoprobe` indicates that you must be following very out-of-date instructions? You might want to try the instructions in the latest version of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf (latest version of that PDF currently says `build-date: 2024-02-02` on page 1).

stimpoff, 2025-03-27:
I have slightly different problem to build OPENOCD on raspberry PI OS 64bit

make

```
...
/usr/bin/ld: /usr/local/lib/libjim.a(jim.o): in function `JimExprOpDoubleUnary':
jim.c:(.text+0x15038): undefined reference to `sqrt'
/usr/bin/ld: jim.c:(.text+0x150f0): undefined reference to `log10'
/usr/bin/ld: jim.c:(.text+0x15100): undefined reference to `log'
/usr/bin/ld: jim.c:(.text+0x15110): undefined reference to `exp'
/usr/bin/ld: jim.c:(.text+0x15138): undefined reference to `tanh'
/usr/bin/ld: jim.c:(.text+0x15148): undefined reference to `cosh'
/usr/bin/ld: jim.c:(.text+0x15158): undefined reference to `sinh'
/usr/bin/ld: jim.c:(.text+0x15168): undefined reference to `atan'
/usr/bin/ld: jim.c:(.text+0x15178): undefined reference to `acos'
/usr/bin/ld: jim.c:(.text+0x15188): undefined reference to `asin'
/usr/bin/ld: jim.c:(.text+0x15198): undefined reference to `tan'
/usr/bin/ld: jim.c:(.text+0x151a8): undefined reference to `cos'
/usr/bin/ld: jim.c:(.text+0x151b8): undefined reference to `sin'
/usr/bin/ld: /usr/local/lib/libjim.a(jim.o): in function `JimExprOpBin':
jim.c:(.text+0x15db0): undefined reference to `pow'
/usr/bin/ld: jim.c:(.text+0x15e9c): undefined reference to `hypot'
/usr/bin/ld: jim.c:(.text+0x15eac): undefined reference to `atan2'
/usr/bin/ld: jim.c:(.text+0x15ebc): undefined reference to `fmod'
collect2: error: ld returned 1 exit status
make[2]: *** [Makefile:3239: src/openocd] Error 1
make[2]: Leaving directory '/home/pi/openocd'
make[1]: *** [Makefile:5010: all-recursive] Error 1
make[1]: Leaving directory '/home/pi/openocd'
make: *** [Makefile:2148: all] Error 2
```

lurch, 2025-03-27:
@stimpoff I just tried the "Building OpenOCD" instructions from Appendix A of https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf on Raspberry Pi OS 64-bit and they worked totally fine for me.
Can you see if you're still able to reproduce the problem on a fresh install of Raspberry Pi OS 64-bit please?

stimpoff, 2025-04-01:
Hmm, it works on clean install. Thank you I will investigate what broke my installation

lurch, 2025-04-01:
Closing issue, as the OP hasn't responded for over a year.
@stimpoff If you run into any other problems, please open a new issue.

closed 2025-04-01 by lurch