raspberrypi/openocd #97: Unable to flash using openocd

sionik, 2024-02-20:
Hi,

first of all, I'm not sure if it's really a problem with openocd. The toolchain could also be responsible for the actual problem. Anyway, I think openocd is either misbehaving or at least not behaving well.
I am happy to open an issue with the relevant project if I know where the root of the problem lies.

About the problem:
I somehow created a .elf file that openocd can't flash.
After much searching, I found that the problem can be solved either by changing the code or by moving the project to another location.
Through further trial and error, I found that moving the project does not _always_ fix the problem. 

If the path length (`pwd | wc -c`) is exactly 48, the problem is reproducible no matter where the project is stored.
The paths I have tried are:

    /home/simon/devel/src/beerpong/fw_control_board # the original path
    /home/simon/devel/src/beerpong/fw_control_12345
    /home/simon/devel/src/beerpong/abc123/123456789
    /home/simon/12345678901234567890123456789012345

The problem with openocd occurs with all of them.

Here is a faulty file: [faulty.zip](https://github.com/raspberrypi/openocd/files/14351764/faulty.zip)
I have also described below how I can reproduce the error on my end. The project is under development, so I created a tag for the status I used to run the tests.
Every change to the code makes the problem disappear.

Unfortunately, at this point I can't find out exactly where the problem lies.

### openocd output
```
❯ openocd -f interface/jlink.cfg -c "transport select swd" -c "adapter speed 10000" -f target/rp2040.cfg
Open On-Chip Debugger 0.11.0-g8e3c38f78 (2024-02-10-09:32)
Licensed under GNU GPL v2
For bug reports, read
        http://openocd.org/doc/doxygen/bugs.html
swd
adapter speed: 10000 kHz

Info : Hardware thread awareness created
Info : Hardware thread awareness created
Info : RP2040 Flash Bank Command
Info : Listening on port 6666 for tcl connections
Info : Listening on port 4444 for telnet connections
Info : J-Trace PRO V2 Cortex compiled Jan 17 2024 14:52:43
Info : Hardware version: 2.00
Info : VTarget = 3.280 V
Info : clock speed 10000 kHz
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x00000001
Info : SWD DPIDR 0x0bc12477
Info : SWD DLPIDR 0x10000001
Info : rp2040.core0: hardware has 4 breakpoints, 2 watchpoints
Info : rp2040.core1: hardware has 4 breakpoints, 2 watchpoints
Info : starting gdb server for rp2040.core0 on 3333
Info : Listening on port 3333 for gdb connections
Info : accepting 'gdb' connection on tcp/3333
Info : RP2040 B0 Flash Probe: 2097152 bytes @10000000, in 512 sectors

Info : New GDB Connection: 1, Target rp2040.core0, state: halted
undefined debug reason 8 - target needs reset
Warn : Prefer GDB command "target extended-remote 3333" instead of "target remote 3333"
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0xf1000000 pc: 0x000000ee msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
Info : Writing 39788 bytes starting at 0x0
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
target halted due to debug-request, current mode: Thread 
xPSR: 0x01000000 pc: 0x00000178 msp: 0x20041f00
Warn : Failed to receive data from device: LIBUSB_ERROR_TIMEOUT.
Warn : Failed to receive data from device: LIBUSB_ERROR_TIMEOUT.
Error: Receiving data from device timed out.
Error: transport_read() failed: timeout occurred.
Error: jaylink_swd_io() failed: timeout occurred.
Warn : Last read operation left 1535 bytes.
Error: SWD I/O operation failed: 0xc7.
Error: jaylink_swd_io() failed: device: unspecified error.
Error: RP2040 BOOT ROM not found
Error: Debug trampoline not found in RP2040 ROM.
Error: Failed to invoke flash programming code on target
Error: error writing to flash at address 0x10000000 at offset 0x00000000
Warn : keep_alive() was not invoked in the 1000 ms timelimit. GDB alive packet not sent! (2196 ms). Workaround: increase "set remotetimeout" in GDB
Warn : Last read operation left 1512 bytes in the buffer.
Error: SWD I/O operation failed: 0xe.
Error: jaylink_swd_io() failed: device: unspecified error.
Warn : Last read operation left 22 bytes in the buffer.
Polling target rp2040.core1 failed, trying to reexamine
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 100ms
Error: Could not find MEM-AP to control the core
Error: Could not find MEM-AP to control the core
Info : AP write error, reset will not halt
Error: DP initialisation failed
Error executing event gdb-flash-write-end on target rp2040.core0:

Polling target rp2040.core0 failed, trying to reexamine
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 100ms
Polling target rp2040.core0 failed, trying to reexamine
Error: Could not find MEM-AP to control the core
Examination failed, GDB will be halted. Polling again in 300ms
[...]
^C
```

### My Setup
A RP Pico is powered via USB and connected to a Segger JLink.
openocd is started on the commandline via `openocd -f interface/jlink.cfg -c "transport select swd" -c "adapter speed 10000" -f target/rp2040.cfg`

Flashing is initiated via gdb using `target remote :3333` and `load`

    ❯ arm-none-eabi-ld --version
    GNU ld (GNU Binutils) 2.41

    ❯ arm-none-eabi-gcc --version
    arm-none-eabi-gcc (Arch Repository) 13.2.0

    ❯ arm-none-eabi-g++ --version
    arm-none-eabi-g++ (Arch Repository) 13.2.0

    ❯ pacman -Qi arm-none-eabi-newlib | rg Version 
    Version                  : 4.3.0.20230120-1

    ❯ pacman -Qi arm-none-eabi-binutils | rg Version
    Version                  : 2.41-1

    ❯ pacman -Qi arm-none-eabi-gcc | rg Version
    Version                  : 13.2.0-2

### Reproduce
    git clone --recurse-submodules --branch=issue_branchlen_48 https://git.sinic.eu/beerpong/fw_control_board.git
    cd fw_control_board
    pwd | wc -c                            # must return `48`
    mkdir build ; cd build
    cmake -DCMAKE_BUILD_TYPE=Debug ..      # only reproducible with "Debug"
    cmake --build . -- -j$(nproc)          # multi core does not matter
    arm-none-eabi-gdb fw_control_unit.elf
    > target remote :3333
    > load

P33M, 2024-02-21:
As per the link in the 4th line of OpenOCD output, information on bug reporting can be found here: https://openocd.org/doc/doxygen/html/bugs.html

Please report this issue to the openocd mailing list.

closed 2024-02-21 by P33M

sionik, 2024-02-21:
Sorry for my confusion. 
I didn't think the bug report belonged there as I'm not using mainline openocd as I couldn't get this to work.
The bug occurs with the version from the `rp2040` branch of this repo.

I have now installed the version of `rp2040-v0.12.0`. With this the problem does not seem to occur.