raspberrypi/openocd #103: Submodule jimtcl refers to "unadvertised" commits, if an https-URL for this repo is used

slimhazard, 2024-08-23:
I wanted to build the new version of openocd in branch sdk-2.0.0, and I kept getting puzzling errors when I tried to update submodules:
```
$ git submodule update
error: Server does not allow request for unadvertised object 1933e5457b9512d39ebbe11ed32578aada149f49
fatal: Fetched in submodule path 'jimtcl', but it did not contain 1933e5457b9512d39ebbe11ed32578aada149f49. Direct fetching of that commit failed.
```
Various incantations of `git submodule update`, for example with `--init --recursive --remote --force` and permutations thereof, didn't improve the situation.

`bootstrap` doesn't run by default unless `git submodule update` succeeds. I ran it with `SKIP_SUBMODULE`, but then `configure` failed. There was just no way to build the software.

So I cloned the openocd repo all over again, and realized that my previous version had been cloned with the https-URL (`https://github.com/raspberrypi/openocd.git`). The new clone used the git@-URL, and then everything worked.

I've never seen anything like that before. My interpretation is that with Github, it's apparently possible that certain commits are not accessible unless you're a logged-in user, via the git user. Or maybe that's not the explanation, but any rate I could build and install openocd after cloning with the git-URL, and it seemed to be impossible with the https-URL.

lurch, 2024-08-24:
> I've never seen anything like that before.

Hmmm, me neither. So I just checked:
```console
andrew@newny:~$ cd /tmp
andrew@newny:/tmp$ git clone git@github.com:raspberrypi/openocd.git openocd-via-ssh
Cloning into 'openocd-via-ssh'...
remote: Enumerating objects: 78432, done.
remote: Counting objects: 100% (7509/7509), done.
remote: Compressing objects: 100% (697/697), done.
remote: Total 78432 (delta 6894), reused 7013 (delta 6803), pack-reused 70923 (from 1)
Receiving objects: 100% (78432/78432), 22.18 MiB | 16.72 MiB/s, done.
Resolving deltas: 100% (64958/64958), done.
andrew@newny:/tmp$ git clone https://github.com/raspberrypi/openocd.git openocd-via-https
Cloning into 'openocd-via-https'...
remote: Enumerating objects: 78432, done.
remote: Counting objects: 100% (7509/7509), done.
remote: Compressing objects: 100% (697/697), done.
remote: Total 78432 (delta 6894), reused 7013 (delta 6803), pack-reused 70923 (from 1)
Receiving objects: 100% (78432/78432), 22.18 MiB | 29.12 MiB/s, done.
Resolving deltas: 100% (64958/64958), done.
andrew@newny:/tmp$ cd openocd-via-ssh
andrew@newny:/tmp/openocd-via-ssh$ git submodule update --init
Submodule 'jimtcl' (https://github.com/msteveb/jimtcl.git) registered for path 'jimtcl'
Submodule 'src/jtag/drivers/libjaylink' (https://gitlab.zapb.de/libjaylink/libjaylink.git) registered for path 'src/jtag/drivers/libjaylink'
Submodule 'tools/git2cl' (https://git.savannah.nongnu.org/git/git2cl.git) registered for path 'tools/git2cl'
Cloning into '/tmp/openocd-via-ssh/jimtcl'...
Cloning into '/tmp/openocd-via-ssh/src/jtag/drivers/libjaylink'...
Cloning into '/tmp/openocd-via-ssh/tools/git2cl'...
Submodule path 'jimtcl': checked out '1933e5457b9512d39ebbe11ed32578aada149f49'
Submodule path 'src/jtag/drivers/libjaylink': checked out '0d23921a05d5d427332a142d154c213d0c306eb1'
Submodule path 'tools/git2cl': checked out '8373c9f74993e218a08819cbcdbab3f3564bbeba'
andrew@newny:/tmp/openocd-via-ssh$ cd ../openocd-via-https
andrew@newny:/tmp/openocd-via-https$ git submodule update --init
Submodule 'jimtcl' (https://github.com/msteveb/jimtcl.git) registered for path 'jimtcl'
Submodule 'src/jtag/drivers/libjaylink' (https://gitlab.zapb.de/libjaylink/libjaylink.git) registered for path 'src/jtag/drivers/libjaylink'
Submodule 'tools/git2cl' (https://git.savannah.nongnu.org/git/git2cl.git) registered for path 'tools/git2cl'
Cloning into '/tmp/openocd-via-https/jimtcl'...
Cloning into '/tmp/openocd-via-https/src/jtag/drivers/libjaylink'...
Cloning into '/tmp/openocd-via-https/tools/git2cl'...
Submodule path 'jimtcl': checked out '1933e5457b9512d39ebbe11ed32578aada149f49'
Submodule path 'src/jtag/drivers/libjaylink': checked out '0d23921a05d5d427332a142d154c213d0c306eb1'
Submodule path 'tools/git2cl': checked out '8373c9f74993e218a08819cbcdbab3f3564bbeba'
```

All looks fine, so I guess "something" in your old repo had got out of sync or something? Sometimes a `git submodule sync; git submodule update` can fix problems with wonky submodules.

slimhazard, 2024-08-24:
Thanks @lurch, if it can't be reproduced, then IMO the issue can be closed. I for one am OK now with the newly cloned repo using the git-URL -- I could test the new version of openocd successfully with a picoprobe and a Pico2.

I thought I tried everything, but didn't think of `git submodule sync`, it might have done the trick.

closed 2024-08-24 by lurch